open62541/open62541 #332: how to add a method call with UA_Int32 array as input and output argument?

martina-felber, 2015-08-17:
Hello!

I would like to set arrays as input and output arguments for method calls. But the client call always fails.
At the moment my callback function at the server just prints a message and returns with UA_STATUSCODE_GOOD. 

I added the method node to the address space of the server:
```C
	// add method node "IncInt32ArrayValues"
	// define input arguments
	UA_Argument_init(&inputArguments);
	inputArguments.arrayDimensionsSize = 5; // one dim array with 5 elements
	UA_UInt32 * pInputDimensions = UA_UInt32_new();
	*pInputDimensions = 1;
	inputArguments.arrayDimensions = pInputDimensions; // NULL; // UA_Array_new(&UA_TYPES[UA_TYPES_INT32], 5);
	inputArguments.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	inputArguments.description = UA_LOCALIZEDTEXT("en_US", "input an array with 5 elements, type int32");
	inputArguments.name = UA_STRING("int32 value");
	inputArguments.valueRank = -1;

	// define output arguments
	UA_Argument_init(&outputArguments);
	outputArguments.arrayDimensionsSize = 5;
	UA_UInt32 * pOutputDimensions = UA_UInt32_new();
	*pOutputDimensions = 1;
	outputArguments.arrayDimensions = pOutputDimensions; //NULL; // UA_Array_new(&UA_TYPES[UA_TYPES_INT32], 5);
	outputArguments.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	outputArguments.description = UA_LOCALIZEDTEXT("en_US", "increment each array index");
	outputArguments.name = UA_STRING("output is the array, each index is incremented");
	outputArguments.valueRank = -1;

	UA_Server_addMethodNode(server, UA_QUALIFIEDNAME(1,"IncInt32ArrayValues"), UA_NODEID_STRING(1, "IncInt32ArrayValues"),
			UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&IncInt32ArrayValues, 1, &inputArguments, 1, &outputArguments, NULL);
```

client code:
```C
    UA_Int32 InputArray[5] = {0, 0, 0, 0, 0};
    UA_Variant_init(&input);
    UA_Variant_setArray(&input, InputArray, 5, &UA_TYPES[UA_TYPES_INT32]);
    retval = UA_Client_CallServerMethod(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
    		UA_NODEID_STRING(1, "IncInt32ArrayValues"), 1, &input, &outputSize, &output);

    if(retval == UA_STATUSCODE_GOOD) {
        printf("Method call was successfull!\n");
        // ....
    } else {
        printf("Method call was unsuccessfull, error code: %x\n\n", retval);
    }
    UA_Variant_deleteMembers(&input);
```

the return value at the client is UA_STATUSCODE_BADINVALIDARGUMENT. 
I tried with UaExpert too, but the call fails with the same error. I think that I misused inputArguments.arrayDimensionsSize and inputArguments.arrayDimensions when adding the method node to the server address space.

Thank you for your help.
Martina

Stasik0, 2015-08-17:
Without haveing tested it and without looking into the spec, I'd guess that arrayDimensionsSize has to be 1, since you specify an array and not a matrix, the second argument has to be an array of size 1 containing a 5 i.e.:

'''
inputArguments.arrayDimensionsSize = 1; // one dim array
UA_UInt32 * pInputDimensions = UA_UInt32_new();
*pInputDimensions = 5;
inputArguments.arrayDimensions = pInputDimensions;
'''

Stasik0, 2015-08-17:
@martina-felber would it be possile to ask you to protocol such things in a wiki page? Maybe just as QA, or as a list of links to the relative issue (since there trend to get lost).

martina-felber, 2015-08-18:
Good morning!
Many thanks for your answer! I tried your suggestion but unfortunately it didn't work.

I tried the following:
server.c
```C
	////////////////////////////////////////////////////////////////////////////////////////////////////////////
	// add method node "IncInt32ArrayValues"
	// define input arguments
	UA_Argument_init(&inputArguments);
	inputArguments.arrayDimensionsSize = 1;
	UA_UInt32 * pInputDimensions = UA_UInt32_new();
	*pInputDimensions = 5;
	inputArguments.arrayDimensions = pInputDimensions;
	inputArguments.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	inputArguments.description = UA_LOCALIZEDTEXT("en_US", "input an array with 5 elements, type int32");
	inputArguments.name = UA_STRING("int32 value");
	inputArguments.valueRank = -1;

	// define output arguments
	UA_Argument_init(&outputArguments);
	outputArguments.arrayDimensionsSize = 1;
	UA_UInt32 * pOutputDimensions = UA_UInt32_new();
	*pOutputDimensions = 5;
	outputArguments.arrayDimensions = pOutputDimensions;
	outputArguments.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	outputArguments.description = UA_LOCALIZEDTEXT("en_US", "increment each array index");
	outputArguments.name = UA_STRING("output is the array, each index is incremented");
	outputArguments.valueRank = -1;

	UA_Server_addMethodNode(server, UA_QUALIFIEDNAME(1,"IncInt32ArrayValues"), UA_NODEID_STRING(1, "IncInt32ArrayValues"),
			UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&IncInt32ArrayValues, 1, &inputArguments, 1, &outputArguments, NULL);

	// if I use inputArgumentsSize = 1 and outputArgumentsSize = 1 - client return value is UA_STATUSCODE_BADINVALIDARGUMENT
	// if I use inputArgumentsSize = 5 and outputArgumentsSize = 5 - server crashes at ->
	// UA_Server_addMethodNode():
	// open62541.c 7600:	UA_Variant_setArrayCopy(&inputArgumentsVariableNode->value.variant, inputArguments,
    //        	inputArgumentsSize, &UA_TYPES[UA_TYPES_ARGUMENT]);
```

client.c:
```C
    ////////////////////////////////////////////////////////////////////////////////////////////////
    // call method "IncInt32ArrayValues"
    // input: Int32 array with 5 elements
    // output: the given array, each index position is incremented

    // first try:
    UA_Int32 InputArray[5] = {0, 0, 0, 0, 0};
    // second try:
    UA_Int32 * InputArray2 = UA_Array_new(&UA_TYPES[UA_TYPES_INT32], 5);
    for(int i = 0; i < 5; i++) {
    	InputArray2[i] = i;
    }

    UA_Variant_init(&input);
    UA_Variant_setArray(&input, InputArray2, 5, &UA_TYPES[UA_TYPES_INT32]);
    retval = UA_Client_CallServerMethod(client, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
    		UA_NODEID_STRING(1, "IncInt32ArrayValues"), 1, &input, &outputSize, &output);

    // possible mem leak -> InputArray2

    if(retval == UA_STATUSCODE_GOOD) {
        printf("Method call was successfull!\n");
        // ....
    } else {
        printf("Method call was unsuccessfull, error code: %x\n\n", retval);
    }
    UA_Variant_deleteMembers(&input);
```

Of course I can protocol such questions with examples in your wiki page. Just show me how =)

best regards

FlorianPalm, 2015-08-18:
Hi Martina,
first of all you need to set the valueRank of your inputArguments and outputArguments variable to 1
```c
inputArguments.valueRank = 1;
outputArguments.valueRank = 1;
```
This means that you are using an array type as argument:
```c
rank = -3 ->  scalar or 1d array
rank = -2 -> scalar or array with any number of dimensions
rank = -1 -> scalar
rank = 0 -> array with one or more dimensions
rank = 1 -> array with one dimensions
rank > 1 -> array with rank dimensions
```
inputArgumentsSize needs to be 1 as you just pass 1 parameter (array type) to the method.

I have looked into the source code and it seems there is a problem with array types at the moment. I will see if there is a quick fix

martina-felber, 2015-08-20:
hi florian!

thank you very much for your fix!
I am very sorry but it seems that I'm still misusing the addMethodNode function. May I ask for your help?

I wrote following code at the server:
```C 
	////////////////////////////////////////////////////////////////////////////////////////////////////////////
	// add method node "IncInt32ArrayValues"
	// define input arguments
	UA_Argument_init(&inputArguments);
	inputArguments.arrayDimensionsSize = 1;				// one dim array
	UA_UInt32 * pInputDimensions = UA_UInt32_new();
	*pInputDimensions = 5;								// with 5 indices
	inputArguments.arrayDimensions = pInputDimensions;
	inputArguments.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	inputArguments.description = UA_LOCALIZEDTEXT("en_US", "input an array with 5 elements, type int32");
	inputArguments.name = UA_STRING("int32 value");
	inputArguments.valueRank = 1;						// valueRank 1 = array with one dimension

	// define output arguments
	UA_Argument_init(&outputArguments);
	outputArguments.arrayDimensionsSize = 1;			// one dim array
	UA_UInt32 * pOutputDimensions = UA_UInt32_new();
	*pOutputDimensions = 5;								// with 5 indices
	outputArguments.arrayDimensions = pOutputDimensions;
	outputArguments.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	outputArguments.description = UA_LOCALIZEDTEXT("en_US", "increment each array index");
	outputArguments.name = UA_STRING("output is the array, each index is incremented");
	outputArguments.valueRank = 1;						// valueRank 1 = array with one dimension

	UA_Server_addMethodNode(server, UA_QUALIFIEDNAME(1,"IncInt32ArrayValues"), UA_NODEID_STRING(1, "IncInt32ArrayValues"),
			UA_EXPANDEDNODEID_NUMERIC(1, 4711), UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&IncInt32ArrayValues, 1, &inputArguments, 1, &outputArguments, NULL);
```

the call in UaExpert looks better than before, the first InputArgument is the array with 5 indices, I can enter values, but the call fails.
[image]

I debugged and in the function 
callMethod()
    if(inputArguments) {
    argConformsToDefinition()
        statisfySignature()
            if(arg.valueRank >= 1 && var->arrayDimensionsSize != arg.arrayDimensionsSize)
            return UA_STATUSCODE_BADINVALIDARGUMENT;

there the function returns with UA_STATUSCODE_BADINVALIDARGUMENT.
arg.valueRank = 1
var->arrayDimensionsSize = -1
arg.arrayDimensionsSize = 1

The arrayDimensionsSize of the var doesn't tally with the arg arrayDimensionsSize. So the request doesn't tally with the InputArgument? I don't know which param I should change, but it nearly works.

Many thanks!

FlorianPalm, 2015-08-20:
Hi Martina,
your code is completely fine. It should work now

martina-felber, 2015-08-20:
Thank you very much! It works now!
I didn't see that you changed the master branch 40 minutes ago, so I was still using the code from this morning!
Many thanks

@Stasik0 
If you like I can add the examples to your wiki page. I try to make some examples with different methods (different input- and output argument sizes, arrays etc. )

Stasik0, 2015-08-20:
@martina-felber you are welcome to, or maybe event better into "server_method.c" if it is not too big?

martina-felber, 2015-08-21:
Hi!

I tried to add a method IncInt32Value and IncOrDecInt32Value (2 input and 2 output arguments).
But I have a memory leak problem with IncOrDecInt32Value. 
I allocated an UA_Argument array for the input and output arguments. If I don't free it, I get memory leaks, but if I try to free it valgrind states invalid frees. 
```C
UA_Argument * inputArgumentArray = UA_Array_new(&UA_TYPES[UA_TYPES_ARGUMENT], 2);
UA_Argument * outputArgumentArray = UA_Array_new(&UA_TYPES[UA_TYPES_ARGUMENT], 2);

....
	// TODO: invalid free
	UA_Array_delete(inputArgumentArray, &UA_TYPES[UA_TYPES_ARGUMENT], 2);
	UA_Array_delete(outputArgumentArray, &UA_TYPES[UA_TYPES_ARGUMENT], 2);

	// UA_Argument_deleteMembers(&inputArgumentArray[0]);
	// UA_Argument_deleteMembers(&inputArgumentArray[1]);
	// UA_Argument_deleteMembers(&outputArgumentArray[0]);
	// UA_Argument_deleteMembers(&outputArgumentArray[1]);
	// UA_Argument_deleteMembers(inputArgumentArray);

```

here is the complete server_method.c file (with the memory leak problem)
It's probably very clumsy, so you can take or discard everything you like or don't use anything of it.

```C
/*
 * This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information.
 */

#include <signal.h>
#include <stdlib.h>

#ifdef UA_NO_AMALGAMATION
# include "ua_types.h"
# include "ua_server.h"
# include "logger_stdout.h"
# include "networklayer_tcp.h"
#else
# include "open62541.h"
#endif

UA_Boolean running = UA_TRUE;
UA_Logger logger;

static UA_StatusCode helloWorldMethod(const UA_NodeId objectId,
		const UA_Variant *input, UA_Variant *output) {
	UA_String *inputStr = (UA_String*) input->data;
	UA_String tmp = UA_STRING_ALLOC("Hello ");
	if (inputStr->length > 0) {
		tmp.data = realloc(tmp.data, tmp.length + inputStr->length);
		memcpy(&tmp.data[tmp.length], inputStr->data, inputStr->length);
		tmp.length += inputStr->length;
	}
	UA_Variant_setScalarCopy(output, &tmp, &UA_TYPES[UA_TYPES_STRING]);
	UA_String_deleteMembers(&tmp);
	UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "Hello World was called");
	return UA_STATUSCODE_GOOD;
}

static UA_StatusCode MethodNoArguments(const UA_NodeId objectId, const UA_Variant *input, UA_Variant *output) {
	// client invoked method with no arguments (no input, no output)
	// do something ...
	UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "MethodNoArguments() method call\n");
	return UA_STATUSCODE_GOOD;
}

static UA_StatusCode IncInt32Value(const UA_NodeId objectId, const UA_Variant *input, UA_Variant *output) {
	UA_LOG_INFO(logger, UA_LOGCATEGORY_SERVER, "IncInt32Value() method call\n");
[... 232 lines cut ...]
	UA_Array_delete(outputArgumentArray, &UA_TYPES[UA_TYPES_ARGUMENT], 2);

	// UA_Argument_deleteMembers(&inputArgumentArray[0]);
	// UA_Argument_deleteMembers(&inputArgumentArray[1]);
	// UA_Argument_deleteMembers(&outputArgumentArray[0]);
	// UA_Argument_deleteMembers(&outputArgumentArray[1]);
	// UA_Argument_deleteMembers(inputArgumentArray);

	/* start server */
	UA_StatusCode retval = UA_Server_run(server, 1, &running); //blocks until running=false

	/* ctrl-c received -> clean up */

	UA_UInt32_delete(pInputDimensions);
	UA_UInt32_delete(pOutputDimensions);
	UA_Server_delete(server);

	return retval;
}

```

Btw: I was a bit confused that your IncInt32ArrayValues method increments by 2. 

Best regards

FlorianPalm, 2015-08-21:
Hi Martina,
it's quite confusing: UA_STRING("") just allocates memory on the stack, but for dynamic structures 
you need to allocate memory on the heap. You could use this function for example:

```C
inputArgumentArray[0].name = UA_String_fromChars("Input1");
```
Same goes for the localizedText field
```C
inputArgumentArray0].description.locale = UA_String_fromChars("en_US");
inputArgumentArray[0].description.text = UA_String_fromChars("A Int32");
```
Perhaps it would be good to add an entry into the wiki page for this issue

martina-felber, 2015-08-21:
@FlorianPalm 
May I ask you how you freed the inputArgumentArray?
I changed every name and description attribute and I am using UA_String_fromChars as you have suggested. 
I have no invalid frees anymore, but memory leaks.

I tried:
```C
	UA_Array_delete(inputArgumentArray, &UA_TYPES[UA_TYPES_ARGUMENT], 2);
	UA_Array_delete(outputArgumentArray, &UA_TYPES[UA_TYPES_ARGUMENT], 2);
```

and 
```C 
	UA_Argument_deleteMembers(&inputArgumentArray[0]);
	UA_Argument_deleteMembers(&inputArgumentArray[1]);
	UA_Argument_deleteMembers(&outputArgumentArray[0]);
	UA_Argument_deleteMembers(&outputArgumentArray[1]);
```

and (this causes invalid frees)
```C
	UA_Argument_delete(&inputArgumentArray[0]);
	UA_Argument_delete(&inputArgumentArray[1]);
	UA_Argument_delete(&outputArgumentArray[0]);
	UA_Argument_delete(&outputArgumentArray[1]);
```

Thank you very much!

FlorianPalm, 2015-08-21:
Hey,
this works for me fine:
```C
		/* add another method node: 2 different argument types*/
	UA_Argument *inputArgumentArray = UA_Array_new(&UA_TYPES[UA_TYPES_ARGUMENT],
			2);

	for(int i=0;i<2;i++)
		UA_Argument_init(&inputArgumentArray[i]);

	inputArgumentArray[0].arrayDimensionsSize = -1;
	inputArgumentArray[0].arrayDimensions = NULL;
	inputArgumentArray[0].dataType = UA_TYPES[UA_TYPES_STRING].typeId;
	inputArgumentArray[0].description.locale = UA_String_fromChars("en_US");
	inputArgumentArray[0].description.text = UA_String_fromChars("A String");
	inputArgumentArray[0].name =  UA_String_fromChars("Input1");
	inputArgumentArray[0].valueRank = -1;

	inputArgumentArray[1].arrayDimensionsSize = -1;
	inputArgumentArray[1].arrayDimensions = NULL;
	inputArgumentArray[1].dataType = UA_TYPES[UA_TYPES_INT32].typeId;

	inputArgumentArray[1].description.locale = UA_String_fromChars("en_US");
	inputArgumentArray[1].description.text = UA_String_fromChars("A Int32");

	inputArgumentArray[1].name =  UA_String_fromChars("Input2");
	inputArgumentArray[1].valueRank = -1;

	UA_Argument *outputArgumentArray = UA_Array_new(
			&UA_TYPES[UA_TYPES_ARGUMENT], 2);
	for(int i=0;i<2;i++)
		UA_Argument_init(&outputArgumentArray[i]);

	outputArgumentArray[0].arrayDimensionsSize = -1;
	outputArgumentArray[0].arrayDimensions = NULL;
	outputArgumentArray[0].dataType = UA_TYPES[UA_TYPES_STRING].typeId;
	outputArgumentArray[0].description.locale = UA_String_fromChars("en_US");
	outputArgumentArray[0].description.text = UA_String_fromChars("A String");

	outputArgumentArray[0].name = UA_String_fromChars("Output1");
	outputArgumentArray[0].valueRank = -1;

	outputArgumentArray[1].arrayDimensionsSize = -1;
	outputArgumentArray[1].arrayDimensions = NULL;
	outputArgumentArray[1].dataType = UA_TYPES[UA_TYPES_DOUBLE].typeId;
	outputArgumentArray[1].description.locale = UA_String_fromChars("en_US");
	outputArgumentArray[1].description.text = UA_String_fromChars("A Double");

	outputArgumentArray[1].name = UA_String_fromChars("Output2");
	outputArgumentArray[1].valueRank = -1;

	UA_Server_addMethodNode(server, UA_QUALIFIEDNAME(1, "Two different Args"),
			UA_NODEID_NUMERIC(1, 62543), UA_EXPANDEDNODEID_NUMERIC(1, 31415),
			UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&callWithDifferentInputArgTypes, 2, inputArgumentArray, 2,
			outputArgumentArray, NULL);

	/* start server */

    UA_Array_delete(inputArgumentArray,&UA_TYPES[UA_TYPES_ARGUMENT],2);
    UA_Array_delete(outputArgumentArray,&UA_TYPES[UA_TYPES_ARGUMENT],2);
    UA_UInt32_delete(pInputDimensions);
    UA_UInt32_delete(pOutputDimensions);

	UA_StatusCode retval = UA_Server_run(server, 1, &running); //blocks until running=false

	/* ctrl-c received -> clean up */

	UA_Server_delete(server);
```

martina-felber, 2015-08-21:
thank you very much!

martina-felber, 2015-08-24:
good morning @FlorianPalm !
Many thanks for your help. I'm very sorry to bother you again but I always have memory leaks and I don't understand why.

I copied your code and it works fine and I think that mine is nearly the same, except I'm using different data types, but I get a memory leak every time I call UA_String_fromChars(). But at your code it works. I don't get it.

This is my main, at first are my defined input and output argument arrays and then I copied yours (just changed inputArgumentArray to inputArgumentArray2 etc. )

```C
int main(int argc, char** argv) {
	signal(SIGINT, stopHandler); /* catches ctrl-c */

	UA_Server *server = UA_Server_new(UA_ServerConfig_standard);
	logger = Logger_Stdout_new();
	UA_Server_setLogger(server, logger);
	UA_Server_addNetworkLayer(server, ServerNetworkLayerTCP_new(UA_ConnectionConfig_standard, 16664));

	////////////////////////////////////////////////////////////////////////////////////////////////////////////
	// add method node: "IncOrDecValue"
	// two input arguments:
	// bool SelectOperation		true = inc; false = dec;
	// Int32 Value

	// two output arguments:
	// Int32 Value				result value
	// String Message			a result message - just to provide a second argument

	// define input arguments
	UA_Argument * inputArgumentArray = UA_Array_new(&UA_TYPES[UA_TYPES_ARGUMENT], 2);

	// first input argument:
	// boolean value indicates the operation
	// true = inc; false = dec
	UA_Argument_init(&inputArgumentArray[0]);
	inputArgumentArray[0].arrayDimensions = NULL;
	inputArgumentArray[0].arrayDimensionsSize = -1;
	inputArgumentArray[0].dataType = UA_TYPES[UA_TYPES_BOOLEAN].typeId;
	//	inputArgumentArray[0].description = UA_LOCALIZEDTEXT("en_US", "true = inc; false = dec");
	//	inputArgumentArray[0].name = UA_STRING("operation: inc or dec");
	inputArgumentArray[0].description.locale = UA_String_fromChars("en_US");
	inputArgumentArray[0].description.text = UA_String_fromChars("true = inc; false = dec");
	inputArgumentArray[0].name = UA_String_fromChars("operation: inc or dec");
	inputArgumentArray[0].valueRank = -1;

	// second input argument:
	// the int32 input value
	UA_Argument_init(&inputArgumentArray[1]);
	inputArgumentArray[1].arrayDimensions = NULL;
	inputArgumentArray[1].arrayDimensionsSize = -1;
	inputArgumentArray[1].dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	//	inputArgumentArray[1].description = UA_LOCALIZEDTEXT("en_US", "input value");
	//	inputArgumentArray[1].name = UA_STRING("value");
	inputArgumentArray[0].description.locale = UA_String_fromChars("en_US");
	inputArgumentArray[0].description.text = UA_String_fromChars("input value");
	inputArgumentArray[0].name = UA_String_fromChars("value");
	inputArgumentArray[1].valueRank = -1;

	// define output arguments
	UA_Argument * outputArgumentArray = UA_Array_new(&UA_TYPES[UA_TYPES_ARGUMENT], 2);

	// first output argument:
	// the int32 result value
	UA_Argument_init(&outputArgumentArray[0]);
	outputArgumentArray[0].arrayDimensions = NULL;
	outputArgumentArray[0].arrayDimensionsSize = -1;
	outputArgumentArray[0].dataType = UA_TYPES[UA_TYPES_INT32].typeId;
	//	outputArgumentArray[0].description = UA_LOCALIZEDTEXT("en_US", "result");
	//	outputArgumentArray[0].name = UA_STRING("result int32 value");
	inputArgumentArray[0].description.locale = UA_String_fromChars("en_US");
	inputArgumentArray[0].description.text = UA_String_fromChars("result");
	inputArgumentArray[0].name = UA_String_fromChars("result int32 value");
[... 80 lines cut ...]
			UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
			&IncOrDecInt32Value, 2, inputArgumentArray2, 2,
			outputArgumentArray2, NULL);

	/* start server */

	UA_Array_delete(inputArgumentArray2,&UA_TYPES[UA_TYPES_ARGUMENT],2);
	UA_Array_delete(outputArgumentArray2,&UA_TYPES[UA_TYPES_ARGUMENT],2);

	// *************************************************************************** //

	//start server
	UA_StatusCode retval = UA_Server_run(server, 1, &running); //blocks until running=false
	//ctrl-c received -> clean up
	UA_Server_delete(server);
	return retval;
}

```

I don't call the function IncOrDecInt32Value, I just execute the server with valgrind (no client connection).

Many thanks for your help
best regards
Martina

FlorianPalm, 2015-08-24:
Hi Martina,
some of the array indices seem to be not correct:
```C
inputArgumentArray[0], outputArgumentArray[1] 
```
(0 and 1 are mixed). I am not sure if this fixes all the leaks. I will have some time to debug it later.
best regards
Florian

martina-felber, 2015-08-24:
Many thanks for your answer!
I guess the indices were the problem, I typed everything new and now it works!
Thank you very much!

closed 2015-08-24 by martina-felber

awmeer, 2017-06-28:
Anyone know how I can create an array input argument with an arbitrary size. I tried the following:
	        inputArgument.arrayDimensionsSize = 1;
	        UA_UInt32 * pInputDimensions = UA_UInt32_new();
	        *pInputDimensions = 0; //Makes the array an arbitrary size ? 
	        inputArgument.arrayDimensions = pInputDimensions;
	        inputArgument.valueRank = 1;
UAExpert seems to recognize that the argument can take an array of arbitrary size. However, when the server gets the messsage, it responds with bad arguments. It looks like the server is comparing inputArgument.arrayDimensions to they array size in the variant. See static UA_StatusCode
satisfySignature() method. Does this mean I can't specifiy an array of arbitrary size as an input argument?

jpfr, 2017-06-28:
The code called by `satisfySignature` accepts 0 as a wildcard for the dimension size:
https://github.com/open62541/open62541/blob/0.2/src/server/ua_services_attribute.c#L86

Can you set breakpoints with a debugger to see where the error code you see comes from?
Possibly you need to set a different valuerank for the argument.
https://open62541.org/doc/0.2/information_modelling.html#value-rank

awmeer, 2017-06-29:
jpfr, I was using v.2 RC-1 version of the library. I just updated to v0.2 and things seem to be working.