open62541/open62541 #336: how to use UaModeler with cmake - generate namespace0 file

martina-felber, 2015-08-18:
Hello!
I would very appreciate it if someone could give me information about the UaModeler and cmake.

I would like to define ObjectTypes like:
```C
MyObject:
	double Temperature;
	double Humidity;
```

In your API there's an UA_Client_createObjectTypeNode function but I wasn't able to find any equal function on the server side like UA_Server_addObjectTypeNode. I don't understand why the client can add object types or delete nodes and the server can't do any of this (maybe you are just working on the server code, but I really don't understand why the client can add or delete nodes, why is the client capable of doing this?).

Anyway, as I wasn't able to create an object type with source code I tried to create an object type with the UaModeler. The model looked fine, the consistency check was successful. I exported the model as XML and renamed it to Opc.Ua.NodeSet2.Minimal.xml, because I could only choose between Opc.Ua.NodeSet2.xml and  Opc.Ua.NodeSet2.Minimal.xml in the cmake-gui, although other xml files were present in the open62541-master\tools\schema directory. I enabled  ENABLE_GENERATE_NAMESPACE0 and chose the file. But the server didn't work, because the UaModeler exported only the user-defined nodes in the xml and not the predefined server object or any other base type nodes. 
After that I copied the content from the UaModeler xml file and pasted it above the end of the Opc.Ua.NodeSet2.Minimal.xml file above </UANodeSet>. It didn't work either, I think it's because of the namespace and extension definitions of UaModeler.

Then I tried something easier and just added a variable to the /Root/Objects folder in the UaModeler and copied only the xml code for the variable to Opc.Ua.NodeSet2.Minimal.xml above </UANodeSet>.

```XML
    <UAVariable DataType="Double" NodeId="ns=1;s=MyVariable" BrowseName="1:MyVariable" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>MyVariable</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
        </References>
        <Value>
            <uax:Double>4711</uax:Double>
        </Value>
    </UAVariable>
```

The strange thing is that UaModeler never defines a parentNodeId. I tried to add the Id manually but it didn't work either. 

Does anybody know how to use the UaModeler and cmake to generate variables/objects/object types/... in the namespace?

Many thanks!
Best regards
Martina

Stasik0, 2015-08-18:
hi Martina,

please look into the "api_hl_abstractions" branch. There a demo server which is able to create nodeset is shown https://github.com/acplt/open62541/blob/api_hl_abstractions/examples/server_nodeset.xml and https://github.com/acplt/open62541/blob/api_hl_abstractions/examples/server_nodeset.c. The example is compilable with BUILD_EXAMPLES option.

hopefuly it will be integrated into the master soon

martina-felber, 2015-08-19:
Hi @Stasik0!

Many thanks for your answer! :)
I tried to stick to your master branch. I used your server_nodeset.xml file and copied everything beneath the aliases (the object type definition and instantiation) to the Opc.Ua.NodeSet2.Minimal.xml file above the end \UANodeSet. And it worked!!!
The strange thing is that I'm using the minimal nodeset, which does not contain the folder ObjectTypes or the BaseObjectType (ID = 58) and I have a reference to it. But it works without problems, the only disadvantage is that I don't see the new defined type in the UaExpert, because the folder ObjectTypes and the BaseObjectType does not exist. I wonder why it works. 

The only thing I had to change manually is the value field of a variable. The UaModeler writes "uax:" before any data type.
```XML
        <Value>
            <uax:Double>44.4</uax:Double>
        </Value>
```

If I delete "uax" it works, maybe there's a setting in the UaModeler to change this syntax. 
Anyway, I can define object types, instantiate them or create variables with UaModeler and your API. That's great! Many thanks!
Martina

Stasik0, 2015-08-19:
all credits go to @ichrispa 

regarding "uax" could you post the "<UANodeSet" tag of your file including all the namesapce definitions? I think uax is just an alias to "http://opcfoundation.org/UA/2008/02/Types.xsd" as seen in http://forum.unified-automation.com/topic1530.html

@ichrispa do you evaluate the XML namespace definitions?

martina-felber, 2015-08-19:
```C
// I created an object type
MySimpleObjectType:
	double Temperature;
	uint32 Humidity;
	
// and instantiated it once in /Root/Objects/MySimpleObjectInstance
// and once in /Root/Objects/MyFolder/MySimpleObjectInstance2
```
this is the generated xml file from UaModeler:
```XML
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd" xmlns:s1="http://yourorganisation.org/opc_complex_datatype_server/Types.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema">
    <NamespaceUris>
        <Uri>http://yourorganisation.org/opc_complex_datatype_server/</Uri>
    </NamespaceUris>
    <Aliases>
        <Alias Alias="UInt32">i=7</Alias>
        <Alias Alias="Double">i=11</Alias>
        <Alias Alias="Organizes">i=35</Alias>
        <Alias Alias="HasModellingRule">i=37</Alias>
        <Alias Alias="HasTypeDefinition">i=40</Alias>
        <Alias Alias="HasSubtype">i=45</Alias>
        <Alias Alias="HasComponent">i=47</Alias>
    </Aliases>
    <Extensions>
        <Extension>
            <ModelInfo Tool="UaModeler" Hash="cD39uvwUoho8lVA1PIg40A==" Version="1.4.0"/>
        </Extension>
    </Extensions>
    <UAObjectType NodeId="ns=1;i=1002" BrowseName="1:MySimpleObjectType">
        <DisplayName>MySimpleObjectType</DisplayName>
        <Description>a simple object with fields temperature and humidity</Description>
        <References>
            <Reference ReferenceType="HasComponent">ns=1;i=6002</Reference>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
            <Reference ReferenceType="HasComponent">ns=1;i=6001</Reference>
        </References>
    </UAObjectType>
    <UAVariable DataType="UInt32" ParentNodeId="ns=1;i=1002" NodeId="ns=1;i=6002" BrowseName="1:Humidity" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Humidity</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
        </References>
        <Value>
            <uax:UInt32>60</uax:UInt32>
        </Value>
    </UAVariable>
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=1002" NodeId="ns=1;i=6001" BrowseName="1:Temperature" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Temperature</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasModellingRule">i=78</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1002</Reference>
        </References>
        <Value>
            <uax:Double>21</uax:Double>
        </Value>
    </UAVariable>
    <UAObject NodeId="ns=1;i=5003" BrowseName="1:MyFolder">
        <DisplayName>MyFolder</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=61</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
            <Reference ReferenceType="Organizes">ns=1;i=5004</Reference>
        </References>
    </UAObject>
    <UAObject NodeId="ns=1;i=5004" BrowseName="1:MySimpleObjectInstance2">
        <DisplayName>MySimpleObjectInstance2</DisplayName>
[... 35 lines cut ...]
    </UAObject>
    <UAVariable DataType="UInt32" ParentNodeId="ns=1;i=5002" NodeId="ns=1;i=6003" BrowseName="1:Humidity" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Humidity</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5002</Reference>
        </References>
        <Value>
            <uax:UInt32>70</uax:UInt32>
        </Value>
    </UAVariable>
    <UAVariable DataType="Double" ParentNodeId="ns=1;i=5002" NodeId="ns=1;i=6004" BrowseName="1:Temperature" UserAccessLevel="3" AccessLevel="3">
        <DisplayName>Temperature</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5002</Reference>
        </References>
        <Value>
            <uax:Double>21</uax:Double>
        </Value>
    </UAVariable>
</UANodeSet>
```

I still have one problem. Before I tried the built example server and it worked. Now I tried to build a project in Eclipse to use the defined object type in the source code but it seems that all generated nodes with the UaModeler are missing. I used cmake with amalgamation and checked the open62541.c file. The defined nodes are at the end of ua_namespaceinit_generated.c but UaExpert doesn't show any of them.

martina-felber, 2015-08-19:
I made a mistake - I forgot to add the symbol ENABLE_GENERATE_NAMESPACE0 to the project build settings. Now it's working with Eclipse as in the built example server.

ichrispa, 2015-08-19:
Hi @martina-felber,

The compiler cannot handle uax:<type> messages, because as far as I can see its usage is illegal in this context. The value should use aliases to fields as defined in the dataType, but uamodeller simply inserts "uax:" to all field aliases. As I do not consider that to be in compliance with an XML namespace description, I am a bit unwilling to fix it on the compiler side.

I always remove the uax: things with awk, but simply replacing them with any text editor will do (replace them with nothing).

Regards,

ichrispa

martina-felber, 2015-08-19:
Many thanks, replacing "uax:" works fine.
I wondered if it's possible to add method nodes too. I added an Object with a Method (int32 as input and output argument). I can see the method node in UaExpert, but a call generates a "BadInternalError" from UaExpert. I defined the method in server.c with the same name as the BrowseName, but it does not work.

ichrispa, 2015-08-19:
The following must apply:
  * The object passed to the call must have a forward hasComponent reference to the method
  * The In|OutputArguments Variablenodes of the method must be referenced with hasProperty references by the method
  * The Browsename of both argument variable nodes must be in namespace 0 (qualifiedname), no matter what namespace the actual nodes are in.
  * Both In|OutputArgument nodes must exist for a method; they may contain no UA_Argument struct in their variant if there are no arguments (UAModeller does adhere do this either).
  * UA_Argument structs in the variable nodes must always be arrays, even if there is only one argument.

Stasik0, 2015-08-19:
@ichrispa the usage is not illegal since "uax:" has been declared as an XML-namespace here ```<UANodeSet [...] xmlns:uax="http://opcfoundation.org/UA/2008/02/Types.xsd" [...]>``` which reference the types definitions. Can Pyhon's parser cope with namespace? Or can we detect this particular namespace manually (as kinda pre-processor in Pyhton)?

ichrispa, 2015-08-19:
it could, but the namespace name is not valid in that context. A double is defined in namespace 0 with the alias "Double". The type <uax:Double> would suggest that uamodeller has defined something called a double in his own namespace context, which it has not. Is that a general convention (in which case it should be integrated into the compiler) or is this just uamodeller misbehaving?

martina-felber, 2015-08-19:
Hi @ichrispa!
Many thanks for your answer! I'm sorry but I didn't understand everything. 
What do you mean with:
> The object passed to the call must have a forward hasComponent reference to the method.

I checked your other points and I think that they are fulfilled, I only had to change the ValueRank of the Arguments to Array. 

this is the generated xml code after I have replaced every uax:
```XML
   <UAMethod NodeId="ns=1;i=7001" BrowseName="1:MyMethod">
        <DisplayName>MyMethod</DisplayName>
        <Description>method returns the incremented input</Description>
        <References>
            <Reference ReferenceType="HasProperty">ns=1;i=6009</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5005</Reference>
            <Reference ReferenceType="HasProperty">ns=1;i=6010</Reference>
        </References>
    </UAMethod>
    <UAVariable DataType="Argument" ParentNodeId="ns=1;i=7001" ValueRank="1" NodeId="ns=1;i=6009" ArrayDimensions="1" BrowseName="InputArguments">
        <DisplayName>InputArguments</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=7001</Reference>
        </References>
        <Value>
            <ListOfExtensionObject>
                <ExtensionObject>
                    <TypeId>
                        <Identifier>i=297</Identifier>
                    </TypeId>
                    <Body>
                        <Argument>
                            <Name>Input_to_increment</Name>
                            <DataType>
                                <Identifier>i=6</Identifier>
                            </DataType>
                            <ValueRank>1</ValueRank>
                            <ArrayDimensions></ArrayDimensions>
                            <Description/>
                        </Argument>
                    </Body>
                </ExtensionObject>
            </ListOfExtensionObject>
        </Value>
    </UAVariable>
    <UAVariable DataType="Argument" ParentNodeId="ns=1;i=7001" ValueRank="1" NodeId="ns=1;i=6010" ArrayDimensions="1" BrowseName="OutputArguments">
        <DisplayName>OutputArguments</DisplayName>
        <References>
            <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
            <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=7001</Reference>
        </References>
        <Value>
            <ListOfExtensionObject>
                <ExtensionObject>
                    <TypeId>
                        <Identifier>i=297</Identifier>
                    </TypeId>
                    <Body>
                        <Argument>
                            <Name>Output_incremented_input</Name>
                            <DataType>
                                <Identifier>i=7</Identifier>
                            </DataType>
                            <ValueRank>1</ValueRank>
                            <ArrayDimensions></ArrayDimensions>
                            <Description/>
                        </Argument>
                    </Body>
                </ExtensionObject>
            </ListOfExtensionObject>
        </Value>
    </UAVariable>
```

UaExpert displays the method node but the call does not work and there are no fields for the input|output arguments. I think it's because I defined the method incorrectly in the source code. 

[image]

I added this method to the server.c file, but I never referenced to it.

```C
static UA_StatusCode MyMethod(const UA_NodeId objectId, const UA_Variant *input, UA_Variant *output) {
	if(input != 0 && output != 0) {
		UA_Int32 * In = (UA_Int32*) input->data;
		(*In)++;
		UA_Variant_setScalarCopy(output, In, &UA_TYPES[UA_TYPES_INT32]);
		return UA_STATUSCODE_GOOD;
	}
	return UA_STATUSCODE_BADARGUMENTSMISSING;
}
```

Many thanks for your help!

ichrispa, 2015-08-19:
Hi @martina-felber,

try expanding the method node in ua expert. You should see an inputArguments and an outputArguments node. Make sure those two are actually arrays (they end with an index such as [0]). 

The method being displayed by uaexpert is purely based on the namespace model (the XML file) and not on what methods you connect. If you have not connected a method, the call will always fail, but the arguments are still visible in uaexpert. Given your error description, something is wrong with how the methods are defined.

Please also take a close look at the browseName of InputArguments. It _must_ be  ```0,"InputArguments"```, not "1, InputArguments".

Can you post a screenshot showing the arguments variable node in uaexpert?

Regards,

ichrispa

Stasik0, 2015-08-19:
@ichrispa wait, but "uax" references to a scheme hosted by opcfoundation (https://opcfoundation.org/UA/2008/02/Types.xsd). I think this is something that maps UA types to XML types? E.g. uax:Double is mapped to xs:double which is a basic XML type hmmmm.

martina-felber, 2015-08-20:
Hi @ichrispa!
Many thanks for your help!
You are right, the definition of the Input|Output Arguments is wrong.
I tried to get the InputArguments working, but I failed. 
The namespace index of the BrowseName is always 0, but the Arguments are Boolean Arrays[-1].
I set the data type to int32 but the datatype is always boolean.

My first try:
InputArgument is an Int32 scalar with ArrayDimension = 0.
[image]

the output of UaExpert is:
[image]

The strange thing is that UaExpert displays the ValueRank = 1, but in the XML the ValueRank = -1.

My second try:
InputArgument is an Int32 array with ArrayDimension = 1
[image]
the output of UaExpert is:
[image]
here at least the ValueRank tallies with the defined ValueRank in the XML

My third try:
InputArgument is an Int32 array with ArrayDimension = 5
[image]
the output of UaExpert is:
[image]

All in all the datatype is always Boolean instead of Int32 and the value is Boolean Array[-1]. I don't know why, because in the XML file the datatype ID = 6 and this should be an Int32. 
```XML
                            <DataType>
                                <Identifier>i=6</Identifier>
                            </DataType>
```

Many thanks for your help.
And I would like to ask how I can connect the method with a function in the source code. 
Thank you very much!

ichrispa, 2015-08-20:
Dear @martina-felber,

> The strange thing is that UaExpert displays the ValueRank = 1, but in the XML the ValueRank = -1.

That is normal. My compiler adds the real valueRank used to the variable to the nodes and removes wildcards. Only types are permitted to use wildcard /negative valueranks (ie meaning "one dimension or less" or "any dimensions").

The problem appears to originate from the namespace compiler. Apparently it does not know how to encode a UA_Argument - which means that the Type definition is not present in any namespace XML parsed during compilation. That is due to the fact that you are using the NodeSet.minimal; please either switch to the full nodeset or make sure the definition of a UA_Argument from namespace 0 is present in the Nodeset.minimal.

The namespace compiler really parses the namespace, it does not just transform XML to C code. If a definition/datatype does not exist in the namespaces known to the compiler, it cannot compile them into C.

Regards,

ichrispa

martina-felber, 2015-08-21:
Hi!
I tried to include the UA_Argument definition to the NodeSet.minimal, but there are many other references to many different nodes. I tried to include everything but it didn't work.

So I used the full nodeset and copied my method definition at the end of it, it compiles but the result in UaExpert is still the same, the method call fails with BadInternalError and the Input and OutputArguments are still BooleanArray[-1].

ichrispa, 2015-08-21:
Hi martina,

there is an example by @Stasik0  in the hl-branch on how to compile your own nodeset along with namespace 0. You do not need to copy anything, just remove the uax: from the XML and take a look at the cmake parameters.

If you don't mind, you can paste your xml on pastebin or something similar and I'll have a look.

Best regards,

ichrispa

martina-felber, 2015-08-24:
Good morning!

Many thanks for your help!
I use the master branch from 17. August. I don't want to use the high level branch, so I tried to copy the generated XML from UaExpert to Nodeset.minimal (and the full nodeset) and compile it with cmake. But if methods doesn't work this way it's no problem. I'm very happy with object types and variables, I just wanted to try methods too. 

I created a simple Objecttype with 2 values (humidity: UInt32 and temperature: double) and instantiated it once in /Root/Objects/MySimpleObjectInstance and /Root/Objects/MyFolder/MySimpleObjectInstance2 this works fine. So I tried to create a complex Object with a method and a variable /Root/Objects/MyFolder/MyComplexObject but this doesn't work with cmake. 

[image]

I copied the XML from UaModeler and tried to compile the full nodeset:

[full nodeset part 1: ](http://pastebin.com/JX26vUaJ)
[full nodeset part 2: ](http://pastebin.com/UbzyQPRe)

and the minimal nodeset:
[minimal nodeset: ](http://pastebin.com/QrwerkgU)

at the minimal nodeset I tried to add the definitions for InputArguments, OutputArguments and Argument, but as they reference to other definitions I added a lot more nodes until I arrived at 
<UAVariable NodeId="i=8252" BrowseName="Opc.Ua" SymbolicName="OpcUa_XmlSchema" DataType="ByteString"> which references a dozen other nodes and I stopped at this point, tried to compile and it didn't work.

Many thanks for your help, but if methods doesn't work with the master branch and cmake it's no big problem.
Thank you very much.
Best regards
Martina

ichrispa, 2015-08-24:
Hi @martina-felber,

the master branch cannot encode the extensionObjects necessary to store UA_Arguments (inputArguments/outputArguments). Unless you use the hl-abstractions branch, you cannot create functions from XML definitions this way. If you do not want to use the hl-abstractions branch, you must use UA_Server_addMethodNode().

The graph of UA Namespace 0 Nodes is very shallow - following the nodes for about 5 levels of links will include the entire namespace 0 due to interdependencies. Use the namespace 0 XML instead.

Best regards,

ichrispa

martina-felber, 2015-08-24:
Hello!

Thank you very much for your help!

I'm sorry but what do you mean with:
> The graph of UA Namespace 0 Nodes is very shallow - following the nodes for about 5 levels of links will include the entire namespace 0 due to interdependencies. Use the namespace 0 XML instead.

I am not using methods anymore but I have another problem with the generated namespace. Valgrind states more than 100 invalid read errors and after server clean up a lot of memory leaks.

I used the Opc.Ua.NodeSet2.Minimal.xml I stated in my previous post. Each generated node from UaModeler creates a lot of invalid reads and memory leaks.

```C
==3710== Invalid read of size 4
==3710==    at 0x8049C5F: UA_NodeId_copy (open62541.c:2414)
==3710==    by 0x80555A7: UA_Server_addNodeWithSession (open62541.c:7539)
==3710==    by 0x80553D4: UA_Server_addNode (open62541.c:7487)
==3710==    by 0x8066CBE: ua_namespaceinit_generated (open62541.c:13993)
==3710==    by 0x8051033: UA_Server_new (open62541.c:6904)
==3710==    by 0x807BBC4: main (server.c:62)
==3710==  Address 0x42064dc is 4 bytes inside a block of size 108 free'd
==3710==    at 0x402B1DD: free (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3710==    by 0x807B8DD: UA_NodeStore_insert (open62541.c:15582)
==3710==    by 0x80555D6: UA_Server_addNodeWithSession (open62541.c:7544)
==3710==    by 0x80553D4: UA_Server_addNode (open62541.c:7487)
==3710==    by 0x8066BB4: ua_namespaceinit_generated (open62541.c:13991)
==3710==    by 0x8051033: UA_Server_new (open62541.c:6904)
==3710==    by 0x807BBC4: main (server.c:62)
==3710== 
==3710== 
==3710== More than 100 errors detected.  Subsequent errors
==3710== will still be recorded, but in less detail than before.
[08/24/2015 09:46:45.973.372] info/communication        Listening on opc.tcp://linux.site:16664
^C[08/24/2015 09:46:50.275.180] info/server     Received Ctrl-C

==3710== 
==3710== HEAP SUMMARY:
==3710==     in use at exit: 144 bytes in 6 blocks
==3710==   total heap usage: 974 allocs, 968 frees, 189,010 bytes allocated
==3710== 
==3710== 24 bytes in 1 blocks are definitely lost in loss record 1 of 6
==3710==    at 0x4029EAD: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3710==    by 0x8079D6B: ua_namespaceinit_generated (open62541.c:15060)
==3710==    by 0x8051033: UA_Server_new (open62541.c:6904)
==3710==    by 0x807BBC4: main (server.c:62)
==3710== 
==3710== 24 bytes in 1 blocks are definitely lost in loss record 2 of 6
==3710==    at 0x4029EAD: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==3710==    by 0x807A3CF: ua_namespaceinit_generated (open62541.c:15086)
==3710==    by 0x8051033: UA_Server_new (open62541.c:6904)
==3710==    by 0x807BBC4: main (server.c:62)
```

my server is empty
[server.c](http://pastebin.com/5xEHr61v)

generated open62541.h file
[open62541.h](http://pastebin.com/WFnd7CcG)

generated open62541.c file
[open62541.c part 1] (http://pastebin.com/3H1866Ca)
[open62541.c part 2](http://pastebin.com/sjmjRKHt)

It works fine but creates a lot of memory errors. I'm still using the master branch, maybe that's the problem. 
Best regards

ichrispa, 2015-08-25:
ua_namespace_init is a one pass function. It may well be that some variable contents are not deallocated after creating them, but the effects on the running server are neglible - in particular, these leaks are not cumulative. 

I don't know where the read errors in the nodestore come from, but I could assume that these are slot- or hash-reads that result in a "not in the nodestore" assertion. The nodestore magicians @Stasik0 or @jpfr might know more about it.

Stasik0, 2015-08-25:
the access errors do not appear in the "origin/api_hl_abstractions" branch, @martina-felber would you please give it a try? Just checkout and use your XML file for namespace generation. It it works - i'd consider it fixed since a merge will *hopefully* not break it

Stasik0, 2015-08-25:
@ichrispa offtopic: can we add "<uax:Double>4711</uax:Double>" replacing "uax" to the python script in order to make it work?

martina-felber, 2015-08-26:
Hello!

Many thanks for your help!

@ichrispa I thought that my last question fits to the issue title, because I still tried to use UaModeler with cmake, generating the namespace 0 with the nodes from the UaModeler XML file. Which title would you suggest to open a new issue?

@Stasik0 I tried to use the high level branch, with your example server_nodeset.xml but I am not able to compile it.
I checked out the high level branch, created the build output directory and selected:
BUILD_EXAMPLECLIENT, BUILD_EXAMPLES, BUILD_EXAMPLESERVER, ENABLE_AMALGAMATION

the build process stops at the server_nodeset example. the file server_nodeset.c tries to include server/ua_server_internal.h, but the file does not exist. 

this is the output from make: (I tried with windows 7 and MinGW)
[mingw make output](http://pastebin.com/)

Afterwards I tried to compile again, I disabled ENABLE_AMALGAMATION and the make process was successful. But I could not run the server_nodeset.exe - Execution creates a system error message:
"Das Programm kann nicht gestartet werden, da libopen62541.dll auf dem Computer fehlt. Installieren Sie das Programm erneut, um das Problem zu beheben."

I tried to build an eclipse project with the the single files, but I got one linker error after the other. I wasn't able to build the project without the amalgamated open62541 header and c file. 

Thank you very much for your support!

Stasik0, 2015-08-26:
@martina-felber the link does not work, for the time being just disable BUILD_EXAMPLES, and go for GENERATE_NAMESPACE

martina-felber, 2015-08-26:
hello!
sorry for the link, now it should work [make output](http://pastebin.com/btJ9cppu)

I tried to compile with ENABLE_AMALGAMATION and ENABLE_EXTERNAL_NAMESPACES but the build failed with a different error.
[make output](http://pastebin.com/Q0B22gPb)
(the name of the file is wrong -> ENABLE_EXTERNAL_NAMESPACES)

if I compile with ENABLE_EXTERNAL_NAMESPACES only it fails too.
[make output](http://pastebin.com/RmAJqq8f)

closed 2015-09-04 by Stasik0