open62541/open62541 #337: 0.2 release meta issue

Stasik0, 2015-08-18:
Hey folks,

i'd like to evaluate how 0.2 can evolve. From my point of view we have more-or-less enough feature to create a new stable release. Afaik these are:

    methods
    subscriptions
    xml-nodeset generation
    new user-api
    MSVS compatibililty
    performance improvements.

I'd like to have an alpha tag due to our workshop on the 7.9.15. Imho only, the api_hl_abstractions branch needs to be reintegrated in order to create an 0.2-alpha. Is it correct, in this case @ichrispa: would you please estimate the deadline or the blockers for it?

Is there anything else we should/can bring 0.2 until it is feature-frozen?

ichrispa, 2015-08-18:
@Stasik0: The single major blocker right now is the serverside api and services support. Some isolated services like the nodemanagement and attributes are not fully matures (ex: you can create 4 node types and delete none using a client). Also, many of the hl-features use the server's nodestore instead of reusing the service apis. The two major blockers I currently see are:

  * Nodemanagement Service does allow creation of all node types
  * Nodemanagement Service can not delete nodes yet
  * Attributes Service only allows the value attribute to be written to

These three are majorly counter-intuitiv. I will fix them asap, as I want the hl-branch to be merged into master before they deviate too much.

I would like to add a series of tutorials to the list of 0.2 features-freezes. We are getting many new users and a lot of "how can I do this/that" questions. Many of these questions have answers hidden in  the examples, but a conclusive tutorial will propably be better.

Stasik0, 2015-08-19:
Okay, so the commitment is to do the merge before 7.9.? So we can present the 0.2-alpha version there? Would you prefer to write tutorials in the doc-system or via a wiki in github, or both?

jpfr, 2015-08-20:
>  Nodemanagement Service does allow creation of all node types
    Nodemanagement Service can not delete nodes yet
    Attributes Service only allows the value attribute to be written to

We probably won't achieve complete coverage of this for 0.2.
Creation of all node types requires that we handle object modelling rules, i.e. required and optional components and properties (which we don't consider so far).
The same applies to the deletion of objects and *type*-nodes.

Another blocker is that subscriptions (to my knowledge) have not been verfied with the official testing tools and have no unit tests. For the 0.1 release it was a lot of work to get all tests go green.

> Would you prefer to write tutorials in the doc-system or via a wiki in github

Maybe we can switch to the sphinx+doxygen documentation between 0.2-pre and 0.2...
Those tutorials in the hi branch are great!

jpfr, 2015-08-20:
btw, I'm making progress in generalizing the async event processing loop so that it can be reused in the client. But that will not be ready before 0.2.

Stasik0, 2015-08-20:
@jpfr imho *not* passing CTT this is not a blocker for a 0.2-review, :+1: for going for sphinx

jpfr, 2015-08-20:
> not passing CTT this is not a blocker for a 0.2-review

Right. For 0.1 we passed the last test only days before the release.

But since we still get reports about memleaks, etc. We surely still have some bugs/edge cases that have not been found.
That's no critique! We all have caused our fair share of bugs.
So we know that truly **stable** code quality is best achieved by process/infrastructure/tool support.

Can we try to have at least *some* unit tests for the new features?

ichrispa, 2015-08-24:
@Stasik0: The tutorials are currently in the sphinx syntax, though they are not very elaborate in terms of sphinx capabilities.

The HL-Branch does have support for writing any node attribute now. I will work on deleting nodes next.

I am having a problem with user generated namespaces, which I will post in another issue.

Stasik0, 2015-08-24:
@ichrispa did we lose this capability? I did not notice it :/

ichrispa, 2015-08-25:
@Stasik0: See #344 for the actual problem. We still have the capability to encode any userspace stuff when the stack gets generated, but I want users to have their own dataTypes and be able to load their namespace using init_my_namespace(server) from userspace. We are not there yet.

ichrispa, 2015-08-28:
@Stasik0: I have completed the Attributes writing issue; but we do now have addInstanceOf(), user namespaces and several other functions in the hl-branch. There are also numerous bugfixes, fixed the unit tests and 4 new tutorials.

I would suggest deferring the creation of all node types from the client side as well as node deletion (additional requirements requested in #345) in favor for merging the current hl-api with the master branch. What do you guys think?

Stasik0, 2015-08-28:
+1 for merging asap to stabilize it a bit before the workshop

jpfr, 2015-08-28:
+1 for thin-slicing feature branches. so we don't need to review the many improvements together with the hl abstraction
+1 for getting the changes into the eventually-to-be 0.2 branch.

But dunno if this should be master.

Many endusers take master for their development instead of a release (and we currently encourage that). The question is how we see the role of master. 

- master -> currently best available release
- master -> dev code. we let the *banana ripe at the customers site*

Relevant post: http://nvie.com/posts/a-successful-git-branching-model/
Let's discuss this at the event!

Stasik0, 2015-08-28:
@jpfr: but integrate current api brach into master should be done before event, right?

Stasik0, 2015-08-28:
Cuz currently i see master=close to next release i.e. Users should take it.

After the release it could get more instable und should not be used by users by default...

Stasik0, 2015-08-28:
Because currently i see master to be very close to next 0.2 and very far from 0.1, so it should be tested by end-users and be stabilized (banan ripe approach).

After the release, master may get instable again, so i'd discourage isers to take it (dev code approach).

So it actually depends on the current state of the release cycle, imho.

ichrispa, 2015-08-29:
@jpfr: I created the hl-api because out of expirience the current master branch is very hard to use; users who really use the stack in their applications will eventually end up building similar features into the stack.

jpfr, 2015-08-29:
Jup. Your branch should be merged as soon as possible.
I was making a more general point about how we use the master branch in the future.

> in that case I would merge the current master into the hl-api, because I am already missing some bugfixes there that are crucial.

In any case, it is advantageous to regularly merge master into dev branches. That makes the evventual PR review+merge much easier, as the branches do not diverge so much.

ichrispa, 2015-08-31:
master merged in hl-abstractioin in 38f8641.

markusgraube, 2015-09-14:
> Many endusers take master for their development instead of a release (and we currently encourage that). The question is how we see the role of master.
>
> master -> currently best available release [...]

+1
I really like the suggested branching model from http://nvie.com/posts/a-successful-git-branching-model/.
This would mean to have both a really stable *master* branch as well as a cutting edge *develop* branch.

jpfr, 2015-12-10:
Are we now ready for 0.2 feature-wise?

I propose we close/fix the following issues/PR and then freeze the API with a release candidate.
- #479 Adding context to a monitored item handling function
- #478 Consistent cmake option naming
- #313 Authenticate Clients with username/password

Then we have some more blocking issues, but they do not affect the userland API. For example
- #417 Subscription Sequence Number
- #480 processRepeatedJobs, again non-RTC-problems, but in the future

Some more issues will surely come up if we test with the new conformances testing tool of the OPC Foundation.

Stasik0, 2015-12-10:
okay, good for me. What is the status of the subsciprtions as mentioned by @ngoossens? this was #417, can it become a blocker for conformance test?

jpfr, 2016-06-06:
All done in here.

closed 2016-06-06 by jpfr

HolgerJeromin, 2017-05-16:
There are no open bugs with 0.2 label and more bugs coming in with bugs from 0.2 rc2
Any plans for releasing? :)