open62541/open62541 #338: Error by deleting client

tpiat, 2015-08-19:
I'm sorry... but I have SIGSEGV at line 10284
function UA_Client_delete,
UA_UserTokenPolicy_deleteMembers(&client->token);

in new version. In version 1.0 this signal was happend without this string.
Probably, in this version (1.1.0) this function call no longer needed.
(tested under windows and under linux, signal SIGSEGV under Linux)

This line was added (issue #317) in version 1.0, but version 1.1 doesn't work whith this line under linux.

I have problems whith this function, pardon my disordered writing

Stasik0, 2015-08-20:
@tpiat is there a way i can reproduce the error? checking out the 0.1.1 and running server and client i cannot get it

tpiat, 2015-08-20:
void core_Logger(UA_LogLevel level, UA_LogCategory category, const char *msg, ...){}
struct OPCUA_Nodes { 	UA_Client *client; };
int main(int argc, char* argv[])
{
    // init
   mylibfun.init(argc, argv); // Any lib function, very big and memory contains new/delete calls.
    OPCUA_Nodes n;
    n.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
    UA_Client_delete(n.client);
    return 0;
}

When g_AppCore.init(argc, argv) uncomment, SIGSEGV
When g_AppCore.init(argc, argv) comment, SIGSEGV doesn't appear

I use open64521.* version.

I see in debug, that field token unitialized. 
There are random values, and after call mylibfun this random values in UA_free/free functions made memory fault.

When I change 
from
UA_Client * UA_Client_new(UA_ClientConfig config, UA_Logger logger) {
    UA_Client *client = <B>UA_malloc(sizeof(UA_Client));</B>
to 
UA_Client * UA_Client_new(UA_ClientConfig config, UA_Logger logger) {
    UA_Client *client = <B>UA_calloc(sizeof(UA_Client),1);</B>
 
SIGSEGV disappears in all cases.

Stasik0, 2015-08-20:
tpiat: this is already fixed, but has not been updated in 0.1.1, could you try this one? http://open62541.org/releases/905d6c8be1.zip

tpiat, 2015-08-20:
Do not crashed.
Thank you!

Stasik0, 2015-08-20:
@tpiat okay, you can access the last builds here: http://open62541.org/releases/ we will probably release a 0.1.2 after some more bugs have been fixed

closed 2015-08-20 by Stasik0