open62541/open62541 #339: QNX version

tpiat, 2015-08-20:
open62541.c works in QNX 6.5.0 after two changes:

file endian.h is absent in QNX
Instead 
#  include <endian.h>

write
#ifndef __QNX__
#  include <endian.h>
#endif

Also add next:
#ifdef __QNX__
#include <sys/socket.h>
#endif

I add it after
# include <arpa/inet.h>

I.e.
# include <arpa/inet.h>
#ifdef __QNX__
#include <sys/socket.h>
#endif

Stasik0, 2015-08-20:
@tpiat: great news, do you want us to integrate it, or could you fork the repositroy and propose pull-request to get on the list of contributors :) ?

tpiat, 2015-08-20:
It added very little to create a separate branch. And for everyone who doesn't works with QNX they are invisible. But I would like them to appear in the basic version, not to add one every time.

HolgerJeromin, 2015-08-20:
We are able to integrate this into our code. But if you would fork the Repo and do a pull request it would also land into our code but your username would show up on:
https://github.com/acplt/open62541/graphs/contributors

Stasik0, 2015-08-20:
@tpiat no i mean if you do a fork, add these changes and propose a pull-request back to us - you will be conserved as the contributor of these lines

tpiat, 2015-08-20:
I'll be glad if this project will contain a couple of lines to support QNX. Who can add it?

Stasik0, 2015-08-20:
You could do it if you do a fork.

tpiat, 2015-08-20:
If I don't make it, QNX will not appear?

Stasik0, 2015-08-20:
No, I can do it aswell, but in this case you will not appear as contributor.

tpiat, 2015-08-20:
Make it, please.

Stasik0, 2015-08-20:
@tpiat the current master including QNX entries can be found here: http://open62541.org/releases/db4a39cbab.zip

Stasik0, 2015-08-24:
@tpiat did you get it working? can we close this one?

tpiat, 2015-08-24:
Version db4a39cbab works,
Version 24565aa4bf works too.
Thanks!!!

closed 2015-08-24 by Stasik0