open62541/open62541 #340: Memory leak connect/new client

tpiat, 2015-08-21:
I use db4a39c version form  #339
ClientNetworkLayerTCP_connect creates connection->handle
ClientNetworkLayerClose must to free it, but if server closed (external server, I write a client), connection.handle is not free, but next ClientNetworkLayerTCP_connect creates new handle.

Probably, this bug is known and fixed. How I can download latest version of open62541.c and open62541.h files?

Stasik0, 2015-08-21:
Latest distributions are here: http://open62541.org/releases/

Stasik0, 2015-08-21:
@tpiat yes, you are using the latest version, could you try to fix it?

tpiat, 2015-08-21:
I'm using latest verion.
I dont't know, how receive event "connection lost".

I inserted variable z, change all free to UA_free, malloc to UA_malloc & etc.
After each cycle <b>server; kill server</b> variable z increaces by one and UA_malloc calls two timesmore than UA_free.

```
static int z=0;

static void ClientNetworkLayerClose(UA_Connection *connection) {
    if(connection->state == UA_CONNECTION_CLOSED)
        return;
    connection->state = UA_CONNECTION_CLOSED;
    socket_close(connection);
#ifndef UA_MULTITHREADING
    UA_ByteString_delete(connection->handle);
	connection->handle=UA_NULL; //tpiat
	z--;
#endif
}

/* we have no networklayer. instead, attach the reusable buffer to the handle */
UA_Connection ClientNetworkLayerTCP_connect(UA_ConnectionConfig localConf, char *endpointUrl,
                                            UA_Logger *logger) {
    UA_Connection connection;
    UA_Connection_init(&connection);
    connection.localConf = localConf;

#ifndef UA_MULTITHREADING
    connection.handle = UA_ByteString_new();
    UA_ByteString_newMembers(connection.handle, localConf.maxMessageSize);
	z++;
#endif
```
I think that new connection created, but old one don't deleted.

It is all my results.

tpiat, 2015-08-24:
```
#include "open62541.h"
#include <string>

using namespace std;

struct OPCUA_Nodes
{
	string ip;
	UA_Client *client;
	int client_connected;
	UA_ReadRequest *rReq;
};

OPCUA_Nodes node;

void core_Logger(UA_LogLevel level, UA_LogCategory category, const char *msg, ...)
{	
}

int localget()
{
	string uri("opc.tcp://");
	uri+=node.ip;

	if(!node.client_connected)
	{
		UA_StatusCode retval = UA_Client_connect(node.client, ClientNetworkLayerTCP_connect,(char *)uri.c_str());
		if(retval != UA_STATUSCODE_GOOD)
		{
			UA_Client_delete(node.client);
			node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
			node.client_connected=0;
			return retval;
		}
		else node.client_connected=1;
	}

	UA_ReadResponse rResp = UA_Client_read(node.client, node.rReq);
	if(rResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD)node.client_connected=0;
	UA_ReadResponse_deleteMembers(&rResp);

	return 0;
}

void mainLoop()
{
	while (1)
	{
		localget();
//		sleepMicro(1000000); // Sleep 1s
	}
}

int main(int argc, char* argv[])
{
	node.ip="127.0.0.1:16664";
	node.client_connected=0;
	node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
	node.rReq = new UA_ReadRequest;
	UA_ReadRequest_init(node.rReq);
	node.rReq->nodesToRead = (UA_ReadValueId *)UA_Array_new(&UA_TYPES[UA_TYPES_READVALUEID], 1);
	node.rReq->nodesToReadSize = 1;
	node.rReq->nodesToRead[0].nodeId = UA_NODEID_STRING_ALLOC(1, "51035"); // assume this node exists

	mainLoop();
	return 0;
}
```

Minimal sample with memory leak,
9226ecb4fd version.

If server do not run. memory leak present.
Windows 7, Qt, gcc.

Stasik0, 2015-08-24:
@tpiat oh no, i have forgotten a line for windows :( commit comes in a second

tpiat, 2015-08-24:
Server -> Kill server -> server -> kill server 
give memory leak with newest verion.

Stasik0, 2015-08-24:
But the one error is gone?

I cannot reproduce this server, kill server sequence here on linux.... what exactly is happening? To you have the same problem with the value of "z"? Can you step-through with a debugger to see what exactly is happening in the network layer e.g. where does the error come from??

tpiat, 2015-08-24:
z increased by two after ecah "server; kill server"
Version 24565aa4bf

Stasik0, 2015-08-24:
@tpiat try the new one please, works here

tpiat, 2015-08-24:
It works.

closed 2015-08-24 by FlorianPalm