open62541/open62541 #341: Build in Cygwin environment

mustafacamurli, 2015-08-22:
Hello all,

I try to build the project on Cygwin. 

I have done exact same things which are written in /doc/building.rst file.

mkdir build
cd build
cmake ..

I couldn't get any makefile out of it.

Stasik0, 2015-08-22:
What error message do you see? Is cmake installed?

mustafacamurli, 2015-08-22:
I was missing couple of background errors. I got my makefile create right now.

Now, I'm dealing with this error:

[ 89%] Building C object CMakeFiles/open62541-object.dir/examples/networklayer_tcp.c.o
In file included from /home/Mustafa/devel/open62541/examples/networklayer_tcp.c:21:0:
/usr/include/netinet/tcp.h:53:2: error: unknown type name ‘u_short’
  u_short th_sport;               /* source port */
  ^
/usr/include/netinet/tcp.h:54:2: error: unknown type name ‘u_short’
  u_short th_dport;               /* destination port */
  ^
/usr/include/netinet/tcp.h:58:2: error: unknown type name ‘u_int’
  u_int   th_x2:4,                /* (unused) */
  ^
/usr/include/netinet/tcp.h:65:2: error: unknown type name ‘u_char’
  u_char  th_flags;
  ^
/usr/include/netinet/tcp.h:74:2: error: unknown type name ‘u_short’
  u_short th_win;                 /* window */
  ^
/usr/include/netinet/tcp.h:75:2: error: unknown type name ‘u_short’
  u_short th_sum;                 /* checksum */
  ^
/usr/include/netinet/tcp.h:76:2: error: unknown type name ‘u_short’
  u_short th_urp;                 /* urgent pointer */
  ^
CMakeFiles/open62541-object.dir/build.make:588: recipe for target 'CMakeFiles/open62541-object.dir/examples/networklayer_tcp.c.o' failed
make[2]: *** [CMakeFiles/open62541-object.dir/examples/networklayer_tcp.c.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Stasik0, 2015-08-22:
From what I read here https://www.cygwin.com/ml/cygwin/2001-07/msg00705.html, you may try to include ```#include <sys/types.h>``` right before ```#include <netinet/tcp.h>```. Just being curious: is there any particular reason to use cygwin?

Stasik0, 2015-08-25:
@mustafacamurli did it work?

ichrispa, 2015-08-25:
@mustafacamurli : You can always attempt to switch to mingw, which is working with open62541,

Stasik0, 2015-08-31:
@mustafacamurli ping?

mustafacamurli, 2015-08-31:
Adding netinet/tcp.h does not solve the problem. I guess  i have problem
with my cygwin version and etc.
I have not found a time to try it with mingw, I will update you as soon as
I try.

the reason of my interest is that im working a pirvate opcua server-client
project in an industrial automation company.
I'm looking for a chance to contribute if i can.
Nowadays strict deadlines matters :)

Thanks for that fast replies. I'll update you.

joyhope, 2015-09-01:
In cygwin environment, disable the problem head file, and then everything is OK.

 //# include <netinet/tcp.h>

Stasik0, 2015-09-01:
@joyhope @mustafacamurli is it it?

Stasik0, 2015-09-01:
can you start it?

mustafacamurli, 2015-09-01:
it worked. Thanks.

closed 2015-09-02 by Stasik0