open62541/open62541 #342: subscriptions: client crashes after calling doPublish() multiple times

martina-felber, 2015-08-25:
Hello!

I tried to create a subscription as in your example client implementation for "the answer". At the server I added a node "int32" and the client subscribes to it. I set the publishInterval to 0 as in your example. 
I thought that the client could send many PublishRequests to the server and the server queues them. If the server detects a change, it generates a notification and after the PublishInterval it dequeues a PublishRequest and sends a response to the client. 

But the client crashes if I call doPublish multiple times and there hasn't been a change.
I'm using the master branch from 20. August. 

server.c main():
```C
int main(int argc, char** argv) {
	signal(SIGINT, stopHandler); /* catches ctrl-c */

    UA_Server *server = UA_Server_new(UA_ServerConfig_standard);
    logger = Logger_Stdout_new();
    UA_Server_setLogger(server, logger);
	UA_Server_addNetworkLayer(server, ServerNetworkLayerTCP_new(UA_ConnectionConfig_standard, 16664));

	// add a static variable node to the address space
	UA_Variant *myIntegerVariant = UA_Variant_new();
	UA_Int32 myInteger = 42;
	UA_Variant_setScalarCopy(myIntegerVariant, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
	const UA_QualifiedName myIntegerName = UA_QUALIFIEDNAME(1, "int32");
	const UA_NodeId myIntegerNodeId = UA_NODEID_STRING(1, "int32");
	UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	UA_Server_addVariableNode(server, myIntegerVariant, myIntegerName,
			myIntegerNodeId, parentNodeId, parentReferenceNodeId, NULL);

	//start server
	UA_StatusCode retval = UA_Server_run(server, 1, &running); //blocks until running=false

	//ctrl-c received -> clean up
	UA_Server_delete(server);

	return retval;
}
```

client:
```C
void handler_int32_changed(UA_UInt32 handle, UA_DataValue *value)
{
	if(value != 0) {
		UA_Int32* data = (UA_Int32*) value->value.data;
		printf("--------------\nThe Int32 variable has changed!\n");
		printf("The new data is: %d\n", *data);
		printf("--------------\n");
	}
}

#define DELAY 1000000000

int main(int argc, char *argv[]) {

    UA_Client *client = UA_Client_new(UA_ClientConfig_standard, Logger_Stdout_new());
    UA_StatusCode retval = UA_Client_connect(client, ClientNetworkLayerTCP_connect,
                                             "opc.tcp://localhost:16664");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return retval;
    }
    char ch = 'a';

    printf("S U B S C R I P T I O N S\n\n");

    UA_ReadRequest rReq;		// Requests and Responses
    UA_ReadResponse rResp;		//
    UA_WriteRequest wReq;		//
    UA_WriteResponse wResp;		//

    UA_Int32 value = 0;
    UA_NodeId monitorThis;
    UA_UInt32 monId = 0;

    // read the current value of int32 variable
    printf("Reading the value of node (1, \"int32\"):\n");
    UA_ReadRequest_init(&rReq);
    rReq.nodesToRead = UA_ReadValueId_new();
    rReq.nodesToReadSize = 1;
    rReq.nodesToRead[0].nodeId = UA_NODEID_STRING_ALLOC(1, "int32");
    rReq.nodesToRead[0].attributeId = UA_ATTRIBUTEID_VALUE;

	rResp = UA_Client_read(client, &rReq);
    if(rResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD &&
       rResp.resultsSize > 0 && rResp.results[0].hasValue &&
       UA_Variant_isScalar(&rResp.results[0].value) &&
       rResp.results[0].value.type == &UA_TYPES[UA_TYPES_INT32]) {
        value = *(UA_Int32*)rResp.results[0].value.data;
        printf("The int32 value is: %i\n\n", value);
    }

    UA_ReadRequest_deleteMembers(&rReq);
    UA_ReadResponse_deleteMembers(&rResp);

	////////////////////////////////////////////////////////////////////////////////////////////////////////////////
    // Create a subscription on the int32 variable with interval 0 (immediate)
    UA_UInt32 MySubscriptionID = UA_Client_newSubscription(client, 0);
    if (MySubscriptionID) {
        printf("Created subscription with interval 0, id: %u\n", MySubscriptionID);
    } else {
    	printf("Create subscription failed!\n");
    }

    // monitor the int32 value
    monitorThis = UA_NODEID_STRING_ALLOC(1, "int32");
    monId = UA_Client_monitorItemChanges(client, MySubscriptionID, monitorThis,
    		UA_ATTRIBUTEID_VALUE, &handler_int32_changed);
    if (monId) {
        printf("Monitoring 'int32', id: %u\n\n", monId);
    }
    UA_NodeId_deleteMembers(&monitorThis);

    // First Publish always generates a notification and invokes the handler.
    printf("Publish Request() - First publish should always generate a notification!\n");
    UA_Client_doPublish(client);

    // these requests should be queued
    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client);
    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client);
    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client);

    // delay
    // for(int i = 0; i < DELAY; i++);

    // change the value of int32 variable
    value = 0;
    printf("Writing the value of node (1, \"int32\"): %i\n", value);
    UA_WriteRequest_init(&wReq);
    wReq.nodesToWrite = UA_WriteValue_new();
    wReq.nodesToWriteSize = 1;
    wReq.nodesToWrite[0].nodeId = UA_NODEID_STRING_ALLOC(1, "int32");
    wReq.nodesToWrite[0].attributeId = UA_ATTRIBUTEID_VALUE;
    wReq.nodesToWrite[0].value.hasValue = UA_TRUE;
    wReq.nodesToWrite[0].value.value.type = &UA_TYPES[UA_TYPES_INT32];
    wReq.nodesToWrite[0].value.value.storageType = UA_VARIANT_DATA_NODELETE;
    wReq.nodesToWrite[0].value.value.data = &value;

    wResp = UA_Client_write(client, &wReq);
    if(wResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD) {
    	printf("error: write request failed!\n\n");
    }

    UA_WriteRequest_deleteMembers(&wReq);
    UA_WriteResponse_deleteMembers(&wResp);

    // handler should be called automatically (queued Publish Request) but it isn't :(
    printf("handler should be called automatically\n");

    // these requests should be queued
    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client); // this request calls the handler
    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client);

    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client);
    printf("Publish Request() - should not call the handler - no change!\n");
    UA_Client_doPublish(client);
    // Delete subscription (which also unmonitors all items)
    printf("Delete the subscription\n");
    if(!UA_Client_removeSubscription(client, MySubscriptionID)) {
        printf("Subscription removed\n");
    }

    printf("\n\nClient finished!\n");
    ch = getchar();
    UA_Client_disconnect(client);
    UA_Client_delete(client);
    return UA_STATUSCODE_GOOD;
}
```

The crash point depends on the timing (changes if I use  different publishIntervals or use some delays), but it's always at  UA_Client_doPublish() - request.subscriptionAcknowledgements[index].sequenceNumber = ack->subAck.sequenceNumber;

I thought that the PublishRequests of the Client are queued at the server and that the notification message is returned at the PublishInterval when a change has been detected. And I thought that the client could send a lot of PublishRequests, but if I do so the client crashes. Maybe I misunderstood the user of subscriptions? Am I not allowed to call doPublish multiple times, if there's no change?

Many thanks for your help
Best regards
Martina

ichrispa, 2015-08-25:
Hello @martina-felber,

of course you can call doPublish as many times as you want to and naturally the client should not be crashing. 

> I thought that the PublishRequests of the Client are queued at the server and that the notification message is returned at the PublishInterval when a change has been detected.
> Since subscriptions are my piece of work, I have been trying to reproduce your problem anyway I could think off to get rid of any errors that might be lurking in there...

In general terms you are correct. However the server may also reply with empty responses or keepalives. Note that the part of code you quoted tracks acknowledgements for received notifications  on the client side and not the publish requests.

I have tried the following:
  * Linux/OpenSUSE native x86_64 and x86 using gcc
  * Linux/OpenSUSE native x86_64 and x86 using clang
  * mingw32 for windows 7 32 Bit and 64 Bit

In all cases, I get the following, no matter how frequently I run the client:

```
S U B S C R I P T I O N S

Reading the value of node (1, "int32"):
The int32 value is: 0

Created subscription with interval 0, id: 627429331
Monitoring 'int32', id: 627429332

Publish Request() - First publish should always generate a notification!
--------------
The Int32 variable has changed!
The new data is: 0
--------------
Publish Request() - should not call the handler - no change!
Publish Request() - should not call the handler - no change!
Publish Request() - should not call the handler - no change!
Writing the value of node (1, "int32"): 0
handler should be called automatically
Publish Request() - should not call the handler - no change!
Publish Request() - should not call the handler - no change!
Publish Request() - should not call the handler - no change!
Publish Request() - should not call the handler - no change!
Delete the subscription
Subscription removed

Client finished!
```

Please help me to get my subscriptions implementation to crash so I can find out what's going on. As a general rule, I never extensively tested anything but numeric nodeIds. But I do not see how that would provoke an invalid ACK tracking.

Best regards,

ichrispa

martina-felber, 2015-08-26:
Hi @ichrispa !

Thank you very much for your help. I'm sorry that I cause you so much effort. 
Here is the complete code I am using (I had to split the open62541.c file)

[open62541.h](http://pastebin.com/yPEVS3c4)
[open62541.c part 1](http://pastebin.com/HybbKWTT)
[open62541.c part 2](http://pastebin.com/b8XeuSv8)
[server.c](http://pastebin.com/iS5XTuLr)
[client.c](http://pastebin.com/7Lke9HKk)

I tested on Windows 7 with Eclipse and MinGW and on openSUSE 13.1 with linux gcc

[windows client output](http://pastebin.com/mCzY6cUr)
[linux client output](http://pastebin.com/BFKjSrSJ)
unfortunately windows and linux client outputs are not equal

[output from valgrind - client](http://pastebin.com/eiFcTV19)
valgrind states that there are invalid writes at doPublish() - open62541.c:12068
12068: request.subscriptionAcknowledgements[index].sequenceNumber = ack->subAck.sequenceNumber;
there are no memory errors at the server

doPublish() crash variables (Windows MinGW):
```C 
void UA_Client_doPublish(UA_Client *client) {
    UA_PublishRequest request;
    UA_PublishResponse response;
    UA_Client_NotificationsAckNumber *ack;
    UA_Boolean moreNotifications = UA_TRUE;
    int index = 0 ;

    do {
        UA_PublishRequest_init(&request);
        UA_PublishResponse_init(&response);

        request.subscriptionAcknowledgementsSize = 0;
        LIST_FOREACH(ack, &(client->pendingNotificationsAcks), listEntry) {
            request.subscriptionAcknowledgementsSize++;
        }
        request.subscriptionAcknowledgements = (UA_SubscriptionAcknowledgement *) UA_malloc(sizeof(UA_SubscriptionAcknowledgement)*request.subscriptionAcknowledgementsSize);

        index = 0;
        LIST_FOREACH(ack, &(client->pendingNotificationsAcks), listEntry) {
            ack = client->pendingNotificationsAcks.lh_first;
            // CRASH
            // index = 256
            // request.subscriptionAcknowledgementsSize = 2
            // request.subscriptionsAcknowledgements
            //			.subscriptionId = 2055178849
            //			.sequenceNumber = 1

            // ack->subAck.subscriptionId = 2055178849
            // ack->subAck.sequenceNumber = 1
            request.subscriptionAcknowledgements[index].sequenceNumber = ack->subAck.sequenceNumber; // CRASH SEGMENTATION FAULT
            request.subscriptionAcknowledgements[index].subscriptionId = ack->subAck.subscriptionId;
            index++;
        }

        response = UA_Client_publish(client, &request);
        if (response.responseHeader.serviceResult == UA_STATUSCODE_GOOD)
            moreNotifications = UA_Client_processPublishRx(client, response);
        else
            moreNotifications = UA_FALSE;

        UA_PublishResponse_deleteMembers(&response);
        UA_PublishRequest_deleteMembers(&request);
    }  while(moreNotifications == UA_TRUE);
    return;
}

```

Thank you very much for your help! I hope you can reproduce the crash. I am sorry that I didn't state enough code/information at my first request.
Many thanks
Best regards
Martina

ichrispa, 2015-09-01:
Hi @martina-felber,

I'm a bit swamped right now and don't have a lot of time to tend to issues; I haven't forgotten this ticket though.

Regards,

ichrispa

ichrispa, 2015-09-06:
Hi @martina-felber,

I still cannot replicate the error, but a student of mine reported the same problem when using MSVS. Apparently the index used in the code segment you described completely overruns the list end because a the list entry is not correctly initialized...

I did push a speculative fix to the master branch (0f103c08534799729912e137f63268ce6651edd2). Would you mind giving it a try and see if the error persists?

Regards,

Chris

martina-felber, 2015-09-06:
Hi Chris!

Many thanks!
I'm very sorry, I haven't been able to build the current master branch. My internship has ended and I don't have access to my code anymore. I've just tried to build the master branch with my home computer but I have a problem with cmake and another cross compiler and the next three weeks I have no access to a computer.
Regards
Martina

ichrispa, 2015-09-09:
Hi @martina-felber ,

I am sorry I couldn't provide a fix in time. We are of course available for help in case you do not get cmake running.

For the record, I have eliminated another two bugs in the client subscription code:
  * Sequence numbers that are ack'd with BAD_SEQUENCE_NUMBER_INVALID where not deleted from the client queue.
  * Only the first sequence number in the queue was acknowledged, overwriting the for-loop-index in the process. The send ACK was propably meant to be deleted in that loop, but past-ichrispa removed that part. It caused an endless-loop when more then 1 seq.no. where queued, eventually segfaulting the client.

Changes have been committed to master in edfe828af0d3d09732bbe1d11aab1d12c47c556b

Regards,

ichrispa

closed 2015-09-09 by ichrispa