open62541/open62541 #344: encodeBinary consults UA_TYPES by default, even if typeIndex is in another types array

ichrispa, 2015-08-25:
I am still trying to get the namespace compiler completely userspace compatible. After removing most of the problematic functions an placing addMonoDirectionalReference into userspace, this is what remains:

```
dogset_namespace.c:(.text+0xc14): undefined reference to `UA_malloc'
dogset_namespace.c:(.text+0xceb): undefined reference to `UA_String_encodeBinary'
dogset_namespace.c:(.text+0xd15): undefined reference to `UA_NodeId_encodeBinary'
dogset_namespace.c:(.text+0xd3f): undefined reference to `UA_Int32_encodeBinary'
dogset_namespace.c:(.text+0xd65): undefined reference to `UA_UInt32_encodeBinary'
dogset_namespace.c:(.text+0xd8f): undefined reference to `UA_LocalizedText_encodeBinary'
dogset_namespace.c:(.text+0xdb3): undefined reference to `UA_malloc'
dogset_namespace.c:(.text+0xe00): undefined reference to `UA_free'
dogset_namespace.c:(.text+0xe74): undefined reference to `UA_free'
```

The free's are not a problem, the rest is. The encodeBinaries are used to create ExtensionObjects, which basically boiles down to encoding variable contents that cannot be represented by a UA_struct with a dataType. I tought the namespace compiler to do exactly what generate_dataType does and it successfully generates any dataType and _new/_delete/etc. function for them. But here's where I'm stuck:

If the user defines a new dataType for which of course he will want the default type handler functions, a UA_DataType needs to be created. The member contain *the index of the member in UA_TYPES*, which is always consulted by encodeBinary. But if nested types are used (a user type contain another user type), that index is referring to the wrong array. EncodeBinary looks in UA_TYPES and not UA_<USERNAMESPACE>_TYPES.

I see two ways to resolve this: Either we stick with extensionObjects and make encodeBinaries available in userspace, or we need to insert a pointer to the TYPE array along with the member's type index.

This ticket does not require immediate action. I would like to defer it until the hl-branch is merged back into master. But this will have to be resolved if at some point the user should be able to create his own dataTypes.

Regards,

ichrispa

jpfr, 2015-08-27:
> If the user defines a new dataType for which of course he will want the default type handler functions, a UA_DataType needs to be created. The member contain the index of the member in UA_TYPES, which is always consulted by encodeBinary. But if nested types are used (a user type contain another user type), that index is referring to the wrong array. EncodeBinary looks in UA_TYPES and not UA__TYPES.

Hey @ichrispa,

the generic type handling functions were written with your use case in mind. The members are tagged with a boolean to be either in UA_TYPES **or** in the same array as the father type. In the latter case, the type description is found by taking a simple offset from the father type.

ichrispa, 2015-08-27:
Perfect. So I only need to set the ``.namespaceZero = UA_TRUE`` to FALSE and it should work :) Thank you @jpfr

closed 2015-08-27 by ichrispa