open62541/open62541 #345: Node deletion

jpfr, 2015-08-25:
@ichrispa
You wrote the service for nod deletion in the hl abstraction branch. That's looking good in general.
In addition to the existing code, we should about necessary conditions for deletion and intermediate states of the information model.

- Nodes can only be deleted if there are no hierarchical references downwards from them. This rule is not written in a he standard. But it seems to cover most inconsistencies I could think of.
- We don't have real transactions for the information model. So somebody could add a reference at the same time we try to remove the node. However, we can control

ichrispa, 2015-08-25:
hey @jpfr,

yes, I agree on both counts. I would however plead for not having any nodestore userspace functions changing semantics or structure.

I found a comment at some point that questioned if hierachical deletions are an acceptable solution. Should something like that even be considered?

Best regads,

ichrispa

Pro, 2018-02-01:
@jpfr isn't this already implemented in current master?
Ok to close?

Pro, 2018-11-06:
Related/duplicate of #1812 and #1730

Pro, 2019-09-03:
The node store API was changed a lot since this issue was opened and is most likely fixed. If not, please open a new issue.

closed 2019-09-03 by Pro