open62541/open62541 #346: Method call context

Stasik0, 2015-08-26:
Hi guys,

currently the implemented method gets a nodeId besides the arguments. I think we may need some more information e.g. the information from the session like clientURI, or the time of the call.

Does it make sense to add another parameter like a "context". The problem is that I am not yet sure about all the context we need. So we need to make this "context" extendible somehow. Maybe a list of key-value pairs, anyone an idea?

ichrispa, 2015-08-26:
The hl-branch has the ability to pass an arbitrary handle to method calls; you can pass any kind of struct through that. 

In my experience there is no use in hardcoding this; every application I wrote so far needed different parameters. I think it's best to leave the handles content to the user.

jpfr, 2015-08-26:
I agree with @ichrispa.

Maybe, once we have a "pluggable" user login and rights management (like the pluggable network layer), we can add a void pointer to a session context that comes from user space...

We can expect the user-space method to understand the user-space context. Giving access to the session itself is a bit too much, imho.

ichrispa, 2015-08-26:
Pushed/Backported the handle in methods from my application; sry, I though I had already done this ages ago...

See 198597c520e51d1e69ccbad6c8daf4d9bb5a1fab

Stasik0, 2015-08-26:
@ichrispa @jpfr yea but i still do not see how I should access some stuff that is in the session e.g. *client identity*. The problem is that "session" is defined to be "application context" which imho is the userland part of the application. 

I do not claim to provide the whole session to the method callback, but please tell me how I should get the client URI from a method (without workarounding it via a separate method parameter).

jpfr, 2015-08-26:
@stasik0

You are right that we should add another parameter to the method callback: user-defined data that is attached to the session.

We can forward the entire UA_CreateSessionRequest to the userland rights mgmt plug. Then, the client uri can be grabbed trivially.

ichrispa, 2015-08-26:
@Stasik0: Imho tracking connected clients would be part of the userspace application. A client can call a register function and be either mapped to the object nodeset or an internal struct/list. That can then be passed as a parameter. At least that is how I did that.

Is there anything in the spec that a methodCall needs to be limited to a session context? I cannot recall reading anything beyond the objectId, InArgs, OutArgs stuff.

FlorianPalm, 2015-08-26:
@ichrispa what do you mean with limited? The spec says that most of the services need to have an established session before they can be used, (all not stack related services). An authentication Token is passed in every request within the requestHeader. 
Within Service_Call we have full access to the session. But as the user should not edit this function, imho we should pass something down to the actual user implementation

ichrispa, 2015-08-26:
With limited I meant if a method must be executed in the context of a session. If the spec suggests for example that a method must distinguish between the session type or the clients security mechanism and say "shall return false if XY is not met". But as far as I understood (and I may be wrong here) the method being called only gets the arguments and the node (as in a OO paradigm), it does not know anything about a client. Or did I misunderstand that?

The Tokens and Sessions requirements are checked by Service_Call; so an sessionless client should never be able to call a method.

As far as I can see this is purely a question of taste; I do not have a principal problem with passing a session-relevant identifier to the method callback, as long as it is not complicated. Say a session handle or the NodeId of the authentication token? And some hl-function can then getClientByHandle() or something?

@jpfr: What's your take? I haven't ever needed to ID the client implicitly, it always passed me a handle as an argument if it was necessary, but @Stasik0s idea might be usefull in some usecases.

syvaren, 2017-05-16:
@jpfr @ichrispa @Stasik0 @FlorianPalm 

Hi guys, if it helps I have a usecase for you on this.

My specific problem is that I have to deal with an N to 1 problem. In my case I have a process that is operating and clients can submit setpoints for the process to operate at.  There can be many clients, which is to say many clients can read data from the process, but it is only safe to allow one specific client write access.  If I do not limit the write access to one client, then the various clients can (and do) fight over the setpoint the process is operating at (setpoint flipflops around).

I do not own a OPC spec, but it appears that built in permissions only extend to a user and an admin level.  This is not granular enough in my case since I wish my process clients to be admins with full write access and the various petitioning clients to be users.  This means when I give users write access to a node then they all get it.

If I have access to session ids in the method node callback, (also maybe a before read callback in a regular variable node), then I can force my user clients to register themselves after they establish a session and filter them in the method node call before executing.

Please let me know if you need more information or clarification.

jpfr, 2017-05-16:
Hello @syvaren,

the access roles are not limited to admin and anonymous. These are just the default roles the SDK always assumes to be present.

On the master branch, we define a user-definable plugin that handles access control.
https://github.com/open62541/open62541/blob/master/plugins/ua_accesscontrol_default.c

There, you can write custom code that allows and disallows access. Every read/write/... is checked with the access control plugin before execution.

In the 0.2 branch, you can get something similar by using a datasource that handles the variable value. The datasource is custom code that can handle the access rights checking internally.
But it's not as nice as the access control plugin.

By the way, you can just download the spec from the OPC Foundation homepage.
You only need to register with a valid email.

syvaren, 2017-05-16:
@jpfr Thanks, I'll take a look at the user access module as a possible solution.  

One thing I didn't mention is that the client with write access can change based on a choice by the system operator.  I know it is possible to revoke the session from the server side, but can the server then initiate a session with a different client?

I am assuming using the ua_accesscontrol plugin would require a design where each client held multiple sessions.  All clients would have a readonly access, then based on the system operator's choice he would initiate a write session with one of the clients.

Thanks for mentioning the datasource node, I had forgotten about that, but I am planning to use method nodes for this anyway.

I will get myself a spec then, I thought there was a fee.

Pro, 2018-02-01:
Method callbacks now have all the requested data, i.e.:

```
static UA_StatusCode
helloWorldMethodCallback(UA_Server *server,
                         const UA_NodeId *sessionId, void *sessionHandle,
                         const UA_NodeId *methodId, void *methodContext,
                         const UA_NodeId *objectId, void *objectContext,
                         size_t inputSize, const UA_Variant *input,
                         size_t outputSize, UA_Variant *output) {
```

closed 2018-02-01 by Pro