open62541/open62541 #348: Memory leak connect/new client or SIGSEGV

tpiat, 2015-08-27:
I have next code
If I use 
			UA_Client_disconnect(node.client);
			UA_Client_delete(node.client);
I receive SIGSEGV, if I'll comment <B>UA_Client_disconnect(node.client);</B>, I'll get memory leak.

Even if I don't use UA_Client_delete/UA_Client_new, I get memory leak.

Windows 7, GCC.

Probably I wrote incorrect program. 
Code is below.

```
#include "open62541.h"
#include <string>

using namespace std;

struct OPCUA_Nodes
{
	string ip;
	UA_Client *client;
	int client_connected;
	UA_ReadRequest *rReq;
};

OPCUA_Nodes node;

void core_Logger(UA_LogLevel level, UA_LogCategory category, const char *msg, ...)
{	
}

int localget()
{
	string uri("opc.tcp://");
	uri+=node.ip;

	if(!node.client_connected)
	{		
		UA_StatusCode retval = UA_Client_connect(node.client, ClientNetworkLayerTCP_connect,(char *)uri.c_str());
		if(retval != UA_STATUSCODE_GOOD)
		{
			UA_Client_disconnect(node.client);
			UA_Client_delete(node.client);
			node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
			node.client_connected=0;
			return retval;
		}
		else node.client_connected=1;
	}

	UA_ReadResponse rResp = UA_Client_read(node.client, node.rReq);
	if(rResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
	{
		node.client_connected=0;		
	}
	UA_ReadResponse_deleteMembers(&rResp);

	return 0;
}

void mainLoop()
{
	while (1)
	{
		localget();
//		sleepMicro(1000000); // Sleep 1s
	}
}

int main(int argc, char* argv[])
{
	node.ip="127.0.0.1:16664";
	node.client_connected=0;
	node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
	node.rReq = new UA_ReadRequest;
	UA_ReadRequest_init(node.rReq);
	node.rReq->nodesToRead = (UA_ReadValueId *)UA_Array_new(&UA_TYPES[UA_TYPES_READVALUEID], 1);
	node.rReq->nodesToReadSize = 1;
	node.rReq->nodesToRead[0].nodeId = UA_NODEID_STRING_ALLOC(3, "DEMO.statepath.alarm.SourceName"); // assume this node exists

	mainLoop();
	return 0;
}
```

Stasik0, 2015-08-27:
the latest commit seems to work with this code:
```
#include "open62541.h"
#include <string>

using namespace std;

struct OPCUA_Nodes
{
    string ip;
    UA_Client *client;
    int client_connected;
    UA_ReadRequest *rReq;
};

OPCUA_Nodes node;

void core_Logger(UA_LogLevel level, UA_LogCategory category, const char *msg, ...)
{   
}

int localget()
{
    string uri("opc.tcp://");
    uri+=node.ip;

    if(!node.client_connected)
    {       
        UA_StatusCode retval = UA_Client_connect(node.client, ClientNetworkLayerTCP_connect,(char *)uri.c_str());
        if(retval != UA_STATUSCODE_GOOD)
        {
            UA_Client_disconnect(node.client);
            node.client_connected=0;
            return retval;
        }
        else node.client_connected=1;
    }

    UA_ReadResponse rResp = UA_Client_read(node.client, node.rReq);
    if(rResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
    {
    	//cleanups connection information 
	    UA_Client_disconnect(node.client);
        node.client_connected=0;        
    }
    UA_ReadResponse_deleteMembers(&rResp);

    return 0;
}

void mainLoop()
{
    while (1)
    {
        localget();
//      sleepMicro(1000000); // Sleep 1s
    }
}

int main(int argc, char* argv[])
{
    node.ip="127.0.0.1:16664";
    node.client_connected=0;
    node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
    node.rReq = new UA_ReadRequest;
    UA_ReadRequest_init(node.rReq);
    node.rReq->nodesToRead = (UA_ReadValueId *)UA_Array_new(&UA_TYPES[UA_TYPES_READVALUEID], 1);
    node.rReq->nodesToReadSize = 1;
    node.rReq->nodesToRead[0].nodeId = UA_NODEID_STRING_ALLOC(3, "DEMO.statepath.alarm.SourceName"); // assume this node exists

    mainLoop();
    return 0;
}
```
you always have to call disconnect before calling connect or there is a memleak. new and delete have to be called only once in the livecycle of a client instance.

tpiat, 2015-08-27:
I.e I can't delete and create cient?
(In this case will be SIGSEGV)
Why?

tpiat, 2015-08-27:
void UA_String_deleteMembers(UA_String *p) {
    <B>if(!p)return;</B>
    UA_free(p->data);
    p->data = UA_NULL;
}

jpfr, 2015-08-27:
Currently, we assume that the user does not call functions with null pointers, if this is not explicitly permitted.

If we include this line, we have to include it in a 1000 other places as well for consistency.

Imho we should leave the function as it is. The calling context needs to react to the missing string anyway. That's normally due to an memory allocation problem.

tpiat, 2015-08-27:
I'm afraid null pointer is possible.
```
       UA_StatusCode retval = UA_Client_connect(node.client, ClientNetworkLayerTCP_connect,(char *)uri.c_str());
        if(retval != UA_STATUSCODE_GOOD)
        {
            UA_Client_disconnect(node.client);
            UA_Client_delete(node.client);
```
This sequence calls UA_String_deleteMembers with NULL pointer.

I understand that it call is result of another error.
In any case, thank you for your attention and reply

Stasik0, 2015-08-27:
@tpiat can you tell me where the problem occurs? is it UA_String_deleteMembers(&client->endpointUrl);? A stacktrace with function names would be helpful. Or how can i reproduce the error?

Stasik0, 2015-08-27:
@tpiat i can not reproduce it with http://open62541.org/releases/be91c9dea2.zip

Stasik0, 2015-08-27:
@tpiat we need to go through all possible use-cases, follow #350

Stasik0, 2015-08-28:
by @tpiat 

My sample code (localget, #348)

```
int localget()
{
	string uri("opc.tcp://");
	uri+=node.ip;

	if(!node.client_connected)
	{		
		UA_StatusCode retval = UA_Client_connect(node.client, ClientNetworkLayerTCP_connect,(char *)uri.c_str());
		if(retval != UA_STATUSCODE_GOOD)
		{

			UA_Client_disconnect(node.client);

			UA_Client_delete(node.client);
			node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);

			node.client_connected=0;
			return retval;
		}
		else node.client_connected=1;
	}

	UA_ReadResponse rResp = UA_Client_read(node.client, node.rReq);
	if(rResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
	{
		UA_Client_disconnect(node.client);
		node.client_connected=0;		
	}
	UA_ReadResponse_deleteMembers(&rResp);

	return 0;
}
```
With disconnect or without it disconnect,
with Detete/new pair or witout it 
-- memory leak if server down.

versions 1cfd71af28 or b0e4f46304
For version 1cfd71af28:

allocated memory in line 2257 and nerver freed,
allocated memory in line 14556 and nerver freed.

Stasik0, 2015-08-28:
@tpiat, please write function names, please use pastebin.com

please feel free to test http://open62541.org/releases/0b8d8d566e.zip probably it should also work for your code.

tpiat, 2015-08-28:
No memory leaks in short time

Stasik0, 2015-08-28:
so, can we close it?

closed 2015-08-31 by Stasik0