open62541/open62541 #350: Borked UA_Client lifecycle

Stasik0, 2015-08-27:
Hi, 

The livecycle of the client is borked as shown by problems by @tpiat e.g. #348

Normal client lifecycle is:
* new
* connect
* disconnect
* delete

here, connect allocates memory by calling a callback in the networklayer e.g. buffer 
disconnect needs to call a close function in order to get, connect also allocates some memory in success case e.g. authentication token
* new
* connect
* disconnect
* connect
* disconnect
* delete

In this case disconnect needs to 'restore' the state of new somehow?

then there is the last case:
* new
* connect
--- server drops connection ----
do we need to call disconnect now? Is the connection closed? If yes by whom? The error happens in the networklayer that does not know the client.

Currently everything is somehow hacked an complicated since client-securechannel-connection interleaving. I cannot fix it alone - need some design decisions:
* who frees and initializes client? 
* what is the user expected to do in the error case?
* should disconnect 'invert' the application of connect? (in this case we need to deleteMembers of the client as i currently do)

Stasik0, 2015-08-27:
I'd propose the following 

**(WI1)** _init and _deleteMembers functions for UA_client so it can reside on stack

Rule 1: **You have to call _disconnect if _connect returned success**
=> Implication 1 **(WI2)** _connect has to clean up in case it fails internally e.g. if the server is down
=> Consistent with **LS 1**

Rule 2: **_disconnect reverts the changes of _connect e.g. _disconnect = _deleteMembers + _init since _new does not allocate memory (we are lucky atm)**
=> **(WI3)** testing needed especially in subsequent _disconnect + _delete = _deleteMembers is called twice

tpiat, 2015-08-28:
My sample code (localget, #348)

```
int localget()
{
	string uri("opc.tcp://");
	uri+=node.ip;

	if(!node.client_connected)
	{		
		UA_StatusCode retval = UA_Client_connect(node.client, ClientNetworkLayerTCP_connect,(char *)uri.c_str());
		if(retval != UA_STATUSCODE_GOOD)
		{

			UA_Client_disconnect(node.client);

			UA_Client_delete(node.client);
			node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);

			node.client_connected=0;
			return retval;
		}
		else node.client_connected=1;
	}

	UA_ReadResponse rResp = UA_Client_read(node.client, node.rReq);
	if(rResp.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
	{
		UA_Client_disconnect(node.client);
		node.client_connected=0;		
	}
	UA_ReadResponse_deleteMembers(&rResp);

	return 0;
}
```
With disconnect or without it disconnect,
with Detete/new pair or witout it 
-- memory leak if server down.

versions 1cfd71af28 or b0e4f46304

tpiat, 2015-08-28:
If it's intresting:

I added few log lines, and I see, where occur memory alloc or free
```
#if 0
#ifndef UA_free
# define UA_free(ptr) free(ptr)
#endif
#ifndef UA_malloc
# define UA_malloc(size) malloc(size)
#endif
#ifndef UA_calloc
# define UA_calloc(num, size) calloc(num, size)
#endif
#ifndef UA_realloc
# define UA_realloc(ptr, size) realloc(ptr, size)
#endif

#else
#include <stdio.h>
#define UA_free(ptr) UA_free1(__LINE__,ptr)
#define UA_malloc(size) UA_malloc1(__LINE__,size)
#define UA_realloc(ptr, size) UA_realloc1(__LINE__,ptr, size)
#define UA_calloc(num, size) UA_calloc1(__LINE__,num, size)

int mlc=0,fre=0;
int memdebug=1;

void *UA_malloc1 (int line,size_t size)
{
	void *ptr = malloc(size);
	if(memdebug)
	{
		mlc++;
		FILE *f=fopen("malloc.log","a+t");
		fprintf(f,"%p, line=%6d\n",ptr,line);
		fclose(f);
	}
	return ptr;
}

void *UA_realloc1(int line,void *block, size_t size)
{
	void *ptr = realloc(block,size);
	if(memdebug)
	{
		FILE *f=fopen("realloc.log","a+t");
		fprintf(f,"%p, line=%6d\n",ptr,line);
		fclose(f);
	}
	return ptr;
}

void UA_free1(int line,void *block)
{
	if(!block)return;

	if(memdebug)
	{
		fre++;
		FILE *f=fopen("free.log","a+t");
		fprintf(f,"%p, line=%6d\n",block,line);
		fclose(f);
	}
	free(block);
}

void *UA_calloc1 (int line,size_t num, size_t size)
{
	void *ptr = calloc(num,size);

	if(memdebug)
	{
		mlc++;
		FILE *f=fopen("malloc.log","a+t");
		fprintf(f,"%p, line=%6d\n",ptr,line);
		fclose(f);
	}
	return ptr;
}
#endif
```

Stasik0, 2015-08-28:
@tpiat sorry for deleting your message, i will transfer it back to #348 

following idea, i have add ```client->status``` we get the following graph

[image]

In the new branch https://github.com/acplt/open62541/compare/client_lifecycle I tried to implement it.

I also implemented some "fool-safe" functions for the users e.g. calling conenct multiple times will not cause a memleak, but will call a disconnect or a reset first. Is it okay? Should we habituate users with this?

closed 2015-08-29 by Stasik0