open62541/open62541 #351: No endpoints found even though frame received correctly - intermittent UA_STATUS_BADDECODING_ERROR

akshirasagar, 2015-08-28:
I am trying to communicate with a PLC from a Linux system using open62541. With an executable built from the client.c file (only changing the endpoint URL) in the examples folder, I receive the following error: No suitable endpoint found. However Wireshark shows that a frame with the complete endpoint information was received. On debugging, the status code received  was found to be UA_STATUSCODE_BADDECODINGERROR.
Whenever we run this code with a UAAnsiCServer from Unified Automation, this error is obtained in some cases while it is not in some cases. The behavior is not uniform. 
UaExpert is able to successfully connect to the PLC and browse through all nodes with the same endpoint URL.

Stasik0, 2015-08-28:
* which version are you using
* could you please show us the wireshark dump? 
* could you please tell us from which function the UA_STATUSCODE_BADDECODINGERROR is returned
* which PLC do you use? how can we reproduce it?

akshirasagar, 2015-08-28:
Hello,

Version : 0.1.1

UA_STATUSCODE_BADDECODINGERROR is observed in the synchronousRequest() function within EndpointsHandshake().

For reproducing this error you can download the OPC UA ANSI C Demo Server available at https://www.unified-automation.com/downloads/opc-ua-servers.html. If you are unable to download, please provide me a path where I can upload the exe file and also wireshark trace.

Regards,
Ashwin

Stasik0, 2015-08-28:
@akshirasagar it seems that the connect is working in the current master, do you need 0.1.1 i.e. do we need to look for a change and backport it (Florian tries go give it a quick look).

akshirasagar, 2015-08-28:
@Stasik0 The error does not always occur. At some points the output I get is:
inkshias@inkshias-VirtualBox:~/acplt-open62541-776266e/examples$ ./client
retval getbuffer = 00000000
retval releasebuffer = 00000000
retval secure conversion message header = 80070000
retval symmetric algo security header = 80070000
retval sequence header = 80070000
80070000
endpointsSize = ffffffff
No suitable endpoint found (Error condition)
And in certain cases when it works this is the output:
retval getbuffer = 00000000
retval releasebuffer = 00000000
retval secure conversion message header = 0
retval symmetric algo security header = 0
retval sequence header = 0

Writing a value of node (1, "the.answer"):
retval getbuffer = 00000000
retval releasebuffer = 00000000
retval secure conversion message header = 0
retval symmetric algo security header = 0
retval sequence header = 0
the new value is: 1
retval getbuffer = 00000000
retval releasebuffer = 00000000
retval secure conversion message header = 0
retval symmetric algo security header = 0
retval sequence header = 0
The extra printfs were added to check the status code. 
How is it possible to get an error and there is no indication in wireshark?

Stasik0, 2015-08-28:
is it the master now? or still 0.1.1?

akshirasagar, 2015-08-28:
Yes, this error is for both master and 0.1.1.

FlorianPalm, 2015-08-28:
@

Stasik0, 2015-08-28:
@akshirasagar i also tested it with 1.5.1 UaDemoServer. Client runs on linux, server runs on a windows ps. I changed the ip and the port in client.c and started
```while true; do ./client; done | grep "end"```
no error for about 20 minutes

akshirasagar, 2015-08-28:
Is this tested with localhost or some other IP?

akshirasagar, 2015-08-28:
Yes, I am able to do this using the master now. Any reason why this fails in 0.1.1 though?

Stasik0, 2015-08-28:
@akshirasagar i would think it is some TCP fragentation-reassambling issue (I am not more sure if it is built into 0.1 line or not). We are not able to find and fix the error on the quick. Do you have any specific problems while using master? Master will probably become a new stable 0.2 version towards the end of the year.

closed 2015-09-01 by Stasik0

andycrulez, 2016-02-29:
Hi all, I have notice exactly the same problem. I'm connecting to CODESYS OPC server. 100% reproducible. If you are interested I can provide you more details.

akshirasagar, 2016-02-29:
@andycrulez : I used the master version and it worked without problem, but I faced this problem with release 0.1.1 which at that time was stated as the stable release. You might want to try the master version to see if this works.