open62541/open62541 #355: Memory allocation error when reading a matrix of byte strings

billran, 2015-08-30:
I am using the Eclipse CDT GNU 32-bit tool chain on Windows 7 Pro and Open62541 release 0.1.1.

Reading the Unified Automation's demo UA server node for a matrix of byte strings throws a memory allocation error when the read response is deleted, and the program subsequently crashes.

Here is the read request code with the debugger stopped on the member deletion code on line 317:
[image]

The program crashes with an attempt to free heap beyond the requested size:
GNU gdb (GDB) 7.6.1
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "mingw32".
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
[New Thread 1480.0x5240]
[New Thread 1480.0x5170]
warning: HEAP[CallDll.exe]: 
warning: Heap block at 003664E0 modified at 003664F4 past requested size of c

Here's what that node looks like in the Unified Automations's UA Expert browser:
[image]

jpfr, 2015-08-31:
Hello @billran,

Does the same problem occur when you compile with the current master?

I guess we can copy/paste the current types.h + binary encoding into the 0.1 branch. It improved quite a bit, is heavily unit tested and the interface has not changed.

FlorianPalm, 2015-08-31:
Hi @

billran, 2015-08-31:
Yes, this version fixes the problem!

closed 2015-09-01 by Stasik0