open62541/open62541 #356: Matrix read SIGSEGV

tpiat, 2015-08-31:
I want to read matrix from server.

I dont know what data i have read, and I receive SISEGV. What is it?
Array of array or something else?

And I see null pointer in resp.result.value.arrayDimension. But I was expected real pointed, and values 3 and 3. arrayDimensionSize is equal 2, that is correct.

FlorianPalm, 2015-08-31:
Could you please show the relevant part of your code?

tpiat, 2015-08-31:
```
UA_ReadResponse rResp = UA_Client_read(node.client, node.rReq);	
```

FlorianPalm, 2015-08-31:
That did not really enlighten me... perhaps the part where you fill up node.rReq?

tpiat, 2015-08-31:
I wrote full the code in #348.
I see values in debugger.

Stasik0, 2015-08-31:
we may also have a bug, since there is #355 :/

tpiat, 2015-08-31:
I can't read not only string,

tpiat, 2015-08-31:
Tested. Works in short time.

closed 2015-08-31 by Stasik0