open62541/open62541 #362: Re-review client part of hl_abstractions branch

jpfr, 2015-09-01:
Hey all,

thank you @ichrispa for your hard work on the xml namespace import and the API improvements.

I was a bit surprised to see the PR go through so fast.
Normally, the big PRs improve a lot during code review and should be allowed this opportunity.
Also, the PR could have been split into distinct functionalities that are easier to review.

@ichrispa If it is okay with you, I'd keep the xml generation and server part as-is in master. But I would revert the client API for the time being. With the goal of getting it back in after a proper review.

- Many of the 34 new Coverity bugs are in the client (they could have been found *at home* with clang-analyzer)
- Stasik and I asked for some API changes early on (not exposing the node objects in the client). We never heard back on that.
- There is time pressure to have the xml-generation functionality for the workshop. The client is not so time critical.

Is that approach okay with you?
By the way, looking forward to seeing you next week!

ichrispa, 2015-09-01:
Hi @jpfr,

I was a bit suprised to seeing the hl-api having been merged in with master this morning too. So reverting the change to separate branches is ok with me.

I just fixed many of the 34 coverity bugs and I am hunting down several more; I am fixing them in my local version of the hl-branch as well; if you want to, I can push that branch back up to origin and you can revert master.

The non-exposing of the nodes refers to get|deleteNodeCopy? Sry, I must have skipped over that issue somehow.

Regards,

ichrispa

ichrispa, 2015-09-03:
Before anyone complains about the attribute get|set functions being exported with a void parameter, note that this was explicitlu requested by @Stasik0.

```UA_StatusCode UA_EXPORT
UA_Server_getAttributeValue(UA_Server *server, UA_NodeId nodeId, UA_AttributeId attributeId, void **value);```

Stasik0, 2015-09-03:
But the function should be acompanied by a  huge set of defines, just in the same way as _decode functions, so the user is not supposed to use this one directly

ichrispa, 2015-09-03:
Well, I intended to have the userspace access the void polymorphisms directly, so I do definitely not mind having it in userspace. But @jpfr did at some point complain about the void parameter, so before I get another pointer on coding styles I wanted to point out why this function is exported again :)

Stasik0, 2015-09-03:
Well, macros can be easily replaced back to "real" functions without breaking user experience.

Stasik0, 2015-09-03:
For the means of the workshop we only need to expose 
UA_Server_setAttribute_method and UA_Server_setAttribute_datasource

jpfr, 2015-11-01:
All done

closed 2015-11-01 by jpfr