open62541/open62541 #366: Mini-BUG: SIGSEGV in second call of UA_Client_disconnect

tpiat, 2015-09-01:
I write clinet.

If server crashes, function socket_write calls UA_Client_disconnect,
among others in the function ClientNetworkLayerClose presents line connection->handle = NULL;
If I call UA_Client_disconnect. program crashes.

There are two way: check connection->handle before call of UA_Client_disconnect or check connection->handle in the function ClientNetworkLayerClose.

But I can't get state of connection->handle. because struct UA_Clinent description is hidden in .c file.

Thus I can't check connection, but function UA_Client_disconnect call after disconnect crashes my client.

tpiat, 2015-09-01:
0	UA_String_deleteMembers	open62541.c	2070	0x406280	
1	UA_String_delete	open62541.c	2063	0x406235	
2	ClientNetworkLayerClose	open62541.c	13860	0x421678	
3	UA_SecureChannel_deleteMembersCleanup	open62541.c	5723	0x40bef0	
4	UA_Client_deleteMembers	open62541.c	12253	0x41dff3	
5	UA_Client_reset	open62541.c	12223	0x41dee0	
6	UA_Client_disconnect	open62541.c	12748	0x41f7b8	
7	localget	main.cpp	796	0x4042d0	
8	mainLoop	main.cpp	337	0x402a43	
9	main	main.cpp	536	0x40362c

Stasik0, 2015-09-01:
it is interesting, since disconnect should not crash when closed twice, what is the code you are running? (http://pastebin.com/).. you get a crash when you close the server, right?

tpiat, 2015-09-01:
```
	if(rResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD &&
		rResp.resultsSize > 0 && rResp.results[i].hasValue)
	{
...
	}
	else
	{
		UA_Client_disconnect(node.client);
		node.client_connected=0;
		UA_Client_delete(node.client);
		node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
	}
```
Full code do not ready yet. File contains many additional data.

Stasik0, 2015-09-01:
@tpiat and where is the second disconnect? can i just duplicate the line to reproduce it?

tpiat, 2015-09-01:
First disconnect in socket_write.
Second disconnect -- in localget.

Stasik0, 2015-09-01:
```
	if(rResp.responseHeader.serviceResult == UA_STATUSCODE_GOOD &&
		rResp.resultsSize > 0 && rResp.results[i].hasValue)
	{
...
	}
	else
	{
		UA_Client_disconnect(node.client);		
		UA_Client_disconnect(node.client);
		node.client_connected=0;
		UA_Client_delete(node.client);
		node.client = UA_Client_new(UA_ClientConfig_standard, core_Logger);
	}
```
this breaks as well, right?

tpiat, 2015-09-01:
Without two UA_Client_disconnect(node.client);   breaks.
If disconnected in socket write, first (and only) UA_Client_disconnect crashes the program.

tpiat, 2015-09-01:
http://pastebin.com/g2qNZKUz

Stasik0, 2015-09-01:
@tpiat i cannot reproduce it with 5d085a7595c1d4e59019cc786727ccdd60b0e8fb

Stasik0, 2015-09-01:
@tpiat i pushed checking the handle in ClientNetworkLayerClose, so it may help even if i cannot reproduce it

tpiat, 2015-09-01:
>i cannot reproduce it with
Server ->Kill server couple times.
But probably it happens only in Windows.

Tomorrow I'll test new release.

tpiat, 2015-09-02:
works in short time.
Tested 7881893f4a and 86d8f08b2b
Now testing more detailedю

closed 2015-09-02 by Stasik0