open62541/open62541 #372: UA_free count more then UA_malloc counter

tpiat, 2015-09-02:
I'm inserted code for check free and malloc
http://pastebin.com/1KAErQFX
into verison 86d8f08

After few minutes of work I see
7932 UA_malloc and UA_calloc calls and
15803 UA_free calls

Clinet code 
http://pastebin.com/g2qNZKUz
Unnecessary calls of UA_String_deleteMembers

Stasik0, 2015-09-02:
It is not a problem as long as we free everything and do not free an address twice.

Freeing a null pointer is legal and has no action. Therefore, checking pointers for null for freeing is just dummy code.

Again, I'd suggest using a memory profiler like valgrind since it is far more sophisticated.

tpiat, 2015-09-03:
Too many malloc instead UA_malloc.

Close it please.

closed 2015-09-03 by tpiat