open62541/open62541 #373: project source tree

joyhope, 2015-09-03:
In the CMakelist.txt
```
set(lib_sources ${PROJECT_SOURCE_DIR}/src/ua_types.c
                ...
                ${PROJECT_SOURCE_DIR}/examples/networklayer_tcp.c
                ${PROJECT_SOURCE_DIR}/examples/logger_stdout.c)
                ...
```
The major source code is included partial codes in the examples.    Whether to created the ported directory

```
ports
     \linux
     \windows
     \lwip
     ...
```
This may keep the library clear, and it will be easy for volunteer to add more platform layer in the future.

Stasik0, 2015-09-03:
do you mean separating socket-related code? this is probably the main part which is plattform-dependend. But the differences are not that big, maybe 10-15 lines? Or do you mean something else?

ichrispa, 2015-09-04:
Since the network layer is not part of OPC UA, I would not like it compiled into the stack. I have application instances where this communications layer is ommitted and replaced.

We might consider placing the logger into the src tree. It is being used as a wuite specific instance by the server (e.g. UA_LOG_ERROR in addNetworkLayer).

I would definetely not restructure the source tree. We have a ports directory where patches can be placed.

joyhope, 2015-09-04:
>> do you mean separating socket-related code?
Just keeping the concept clear.  These parts are belong low level part of
OPC UA. It is true not belong to OPC UA, but without it OPC UA could run.
But it is not belong to  user examples.

The project structure:

   - Low level part.  This is possible to be ported in different embedded
   system
   - OPC UA stack:  This is the implement for standard.
   - Examples:  server, client:
   - Tools:

>> Or do you mean something else?
I try to express these parts are low level layer for this open62541.  It is
very possible to porting this project to embedded system.  e.g lwIP might
be a choice.  With the separated directly, it may be clear to understand
this project.

>> I would definetely not restructure the source tree.
I also accept this. My suggestion is not very important now.  These days I
am learning the project, I just feel it is awkward to put the network_xxx.c
in the examples.

Lee

jpfr, 2015-09-04:
The network code is in examples since it is not under the LGPL license.
It is in the public domain and thus even more 'open' than the rest.

The idea is to have a base library with *pluggable* components that are adopted to the different architectures.

If possible, I would like to
- get rid of the ports folder
- pull all architecture specific ifdefs into the ua_config.h

so that the library itself is as architecture agnostic as possible.

closed 2015-10-16 by jpfr