open62541/open62541 #374: Write_Service makes too many copies

Stasik0, 2015-09-03:
Hey lines from server_variable.c

```
UA_Int32* myInteger = UA_Int32_new();
*myInteger = 42;
UA_Variant_setScalar***Copy***(myIntegerVariant, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
[...]
UA_Server_addVariableNode( ..
```

works as expected, the link between myInteger and the value in the node is lost.

Now consider this:

```
UA_Int32* myInteger = UA_Int32_new();
*myInteger = 42;
UA_Variant_setScalar(myIntegerVariant, &myInteger, &UA_TYPES[UA_TYPES_INT32]);
[...]
UA_Server_addVariableNode( ..
```

Now the integer is set in the variant, however the first write on the node fees the pointer and reallocates the memory, so there is no way to get the actual value (except for using datasource callbacks). 

Is it a bug, or do we consider it a normal behaviour and ask users to user callbacks? The write service logic is complex (ranges and stuff) so I cannot fix it on the quick.

ichrispa, 2015-09-03:
imho:
A1: No, this is how the stack works; It creates copies of all data passed, not just variants. I perceive it as 'normal' in the sense that I got used to it. It is in sync with "the user has no direct nodestore access".
A2: Is the variant not copied before it is added to the node? If not, I would vote for that in order to keep things consistent.

It's not just values in my eyes. If you pass a QualifiedName to a function, you don't expect that QualifiedName instance in memory in your node. You presume that the variable will be copied. I would expect the variants to be the same way; it's just a pointer because it is more handy for complex structs.

Stasik0, 2015-09-03:
Well i'd expect that the value here the pointer to the integer is not copied over... this works well for datasource. If the user cannot do it he can select going with UA_Variant_setScalarCopy instead UA_Variant_setScalar. Not a good idea? Here the pointer is kept during reads, but breaks on write...

jpfr, 2015-09-04:
> Now the integer is set in the variant, however the first write on the node fees the pointer and reallocates the memory, so there is no way to get the actual value (except for using datasource callbacks).

UA_Server_getNodeAttribute_value returns you a copy of the value. You can set the value with the same mechanism. What else do you need?

I will discuss the design choices regarding nodes during the workshop's concurrency talk.

closed 2015-09-29 by Stasik0