open62541/open62541 #375: Assymetric addnode API

Stasik0, 2015-09-03:
UA_Server_addDataSourceVariableNode
* UA_Server *server
* UA_DataSource dataSource
* const UA_QualifiedName browseName
* UA_NodeId nodeId
* const UA_NodeId parentNodeId
* const UA_NodeId referenceTypeId
* UA_NodeId *createdNodeId

UA_Server_addVariableNode
* UA_Server *server
* UA_NodeId nodeId
* const UA_QualifiedName browseName, 
* UA_LocalizedText displayName
* UA_LocalizedText description
* const UA_NodeId parentNodeId, 
* const UA_NodeId referenceTypeId
* UA_UInt32 userWriteMask
* UA_UInt32 writeMask, 
* UA_Variant *value
* UA_NodeId *createdNodeId

whats going on here? There much more issues like this. Should we equalize this stuff to equalize the signatures -> are we going to mimic addVariableNode? It seems to have everything so we'd become:

UA_Server_addDataSourceVariableNode
* UA_Server *server
* UA_NodeId nodeId
* const UA_QualifiedName browseName, 
* UA_LocalizedText displayName
* UA_LocalizedText description
* const UA_NodeId parentNodeId, 
* const UA_NodeId referenceTypeId
* UA_UInt32 userWriteMask
* UA_UInt32 writeMask, 
* UA_DataSource dataSource
* UA_NodeId *createdNodeId

FlorianPalm, 2015-09-03:
I agree to changing the functions to end up with a consistent api

ichrispa, 2015-09-03:
That is a slipup on my part. The order of the attributes is always:
  * server
  * BASANODEATTRIBUTS
  * NODETYPEATTRIBUES
  * createdNodeId
I do actually require this node order for the namespace compiler. If you change the orders, please also change the order in the compiler. Basetypes are printed by open62541_macrohelper.py, node specific attributes in their respective node class.

DataSourceVariableNode was never touched by my branch.

@Stasik0 Feel free to fix the inconsistencies.

Stasik0, 2015-09-03:
open problem: valueRank is in UA_Server_addVariableTypeNode, but not in UA_Server_addVariableNode

ichrispa, 2015-09-03:
There are several such inconsistencies. If you check the namespace_generated.c, you will find that the namespace compiler currently flags missing type fields as comments when calling addNode.

I was more interested in getting all addNode functions into a consistent state then to make all attributes accessable; partly because too many attributes may confuse users not that familiar with the OPC UA Specs and mostly because the UA_Server_get|setAttributeValue (or it's respective typed Macros) can easily fix these fields if the user really really wants them set.

ichrispa, 2015-10-15:
Since the hl-api is being reimplemented, I am closing this ticket for now. We'll have to reexamine problems in symmetry once @jpfr is done.

closed 2015-10-15 by ichrispa