open62541/open62541 #377: Adding a nodeid to datasource function

Stasik0, 2015-09-04:
In order to make one callback being able to handle multiple nodes.

jpfr, 2015-09-04:
The datasource already comes with a custom handler pointer that goes into the read and write functions. So you can already reuse the same function for multiple nodes.

```c
typedef struct {
    void *handle; ///> A custom pointer to reuse the same datasource functions for multiple sources
    UA_StatusCode (*read)(void *handle, UA_Boolean includeSourceTimeStamp, const UA_NumericRange *range, UA_DataValue *value);
    UA_StatusCode (*write)(void *handle, const UA_Variant *data, const UA_NumericRange *range);
} UA_DataSource;
```

Stasik0, 2015-09-09:
i am still not convinced, but okay, i will come up with it again, when i get a real use-case

jpfr, 2015-12-04:
This was implemented and merge in 705f03b.

closed 2015-12-04 by jpfr