open62541/open62541 #379: Ship only static binaries

jpfr, 2015-09-04:
Currently there is little use for shared libraries.
I guess everybody wants a binary that just works and can be easily copied.

As requested by @Stasik0, the function-specific examples are now built statically.
Can we make the change for all examples and the binaries we ship in the releases?

Of course, we can still put the .dll / .so into the release-zip, even if the binaries are static...

Stasik0, 2015-09-04:
Well, there are already server_static and client_static

jpfr, 2015-09-05:
Yeah. Eventually, we should break up the big examples into per-feature examples.
So we end up with only the static binaries from the /examples directory.

Then we can just omit the _static postfix.

ichrispa, 2015-09-06:
The dynamic linked libraries are imho important when creating multi-server/multi-client applications which all run on the same stack code base; I did use them extensively in the past to ensure a consistent codebase in these scenarios.

I would also like to keep the big-mess examples, because they are rather good for testing/coverage (i.e. the "has it all" binary can be run for testing). Though I would not necessarily call them "examples" in this case?

closed 2015-09-08 by Stasik0