open62541/open62541 #384: Build of examples fails on AMD64 Linux platform

JNail, 2015-09-08:
Build works well with:
`cmake -DCMAKE_BUILD_TYPE=Debug -DENABLE_AMALGAMATION=OFF -DBUILD_EXAMPLESERVER=ON -DBUILD_EXAMPLECLIENT=ON ..`

Adding option `-DBUILD_EXAMPLES=ON` results in:
```
[ 74%] Building C object CMakeFiles/client_firstSteps.dir/examples/client_firstSteps.c.o
open62541/open62541/examples/client_firstSteps.c: In function ‘main’:
open62541/open62541/examples/client_firstSteps.c:43:12: error: format ‘%llu’ expects argument of type ‘long long unsigned int’, but argument 2 has type ‘UA_DateTime’ [-Werror=format=]
            printf("raw date is: %llu\n", raw_date);
            ^
cc1: all warnings being treated as errors
CMakeFiles/client_firstSteps.dir/build.make:54: recipe for target 'CMakeFiles/client_firstSteps.dir/examples/client_firstSteps.c.o' failed 
make[2]: *** [CMakeFiles/client_firstSteps.dir/examples/client_firstSteps.c.o] Error 1
CMakeFiles/Makefile2:95: recipe for target 'CMakeFiles/client_firstSteps.dir/all' failed
make[1]: *** [CMakeFiles/client_firstSteps.dir/all] Error 2
Makefile:76: recipe for target 'all' failed
make: *** [all] Error 2
```

Change of line 43 of ../examples/client_firstSteps.c to:
`printf("raw date is: %lu\n", raw_date);`
solved the issue as a quick hack. Thanks @Stasik0 

Problem occured with gcc-4.8 on Machine: Linux 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt9-3~deb8u1 (2015-04-24) x86_64 GNU/Linux

Stasik0, 2015-09-08:
@JNail I am trying to solve it properly...

HolgerJeromin, 2015-09-08:
perhaps we should use the PRId64 macros for that. acplt/ov has a few defines for that

HolgerJeromin, 2015-09-08:
fixed with https://github.com/acplt/open62541/commit/76bbe178f70bd96a596c51bfc350a921be2c5825

closed 2015-09-08 by HolgerJeromin