open62541/open62541 #385: Subscriptions: compile error "declaration does not declare anything"

LEvertz, 2015-09-08:
when compiling with ENABLE_SUBSCRIPTIONS I get the error
C:\....\open62541.c: In function 'SubscriptionManager_init':
C:\....\open62541.c:15234:10: error: declaration does not declare anything
compiler is minGW
gcc version 4.5.2 (tdm-1) 
Target: mingw32
Configured with: ../../src/gcc-4.5.2/configure --build=mingw32 --enable-languages=c,c++,ada,fortran,objc,obj-c++ --enable-threads=win32 --enable-libgomp --enable-lto --enable-fully-dynamic-string --enable-libstdcxx-debug --enable-version-specific-runtime-libs --with-gnu-ld --disable-nls --disable-win32-registry --disable-symvers --disable-werror --prefix=/mingw32tdm --with-local-prefix=/mingw32tdm --enable-cxx-flags='-fno-function-sections -fno-data-sections' --with-pkgversion=tdm-1 --enable-sjlj-exceptions --with-bugurl=http://tdm-gcc.tdragon.net/bugs

according to http://www.microchip.com/forums/m500025.aspx C99 does not support anonymous structs
what is going wron here?

Stasik0, 2015-09-08:
seems to be problem with these lines
```
    union {
        struct {
            UA_UInt32 ui32h;
            UA_UInt32 ui32l;
        };
        UA_UInt64 ui64;
    } guidInitH;
    guidInitH.ui64 = (UA_UInt64) UA_DateTime_now();
    manager->lastJobGuid.data1 = guidInitH.ui32h;
    manager->lastJobGuid.data2 = (UA_UInt16) (guidInitH.ui32l >> 16);
    manager->lastJobGuid.data3 = (UA_UInt16) (guidInitH.ui32l);
    union {
        struct {
            UA_UInt32 ui32h;
            UA_UInt32 ui32l;
        };
        UA_UInt64 ui64;
    } guidInitL;
```
probably some 4.5.2 problems

Stasik0, 2015-09-08:
@LEvertz did it help?

ichrispa, 2015-09-22:
Just fyi: The UInt32 structs where added because on a 64 bit system, a datetime is longer then on a 32 bit one. I wanted the entire field as a semi-random initializer. The ticket is somewhere in the closed bin.

ichrispa, 2015-11-19:
In absence of further feedback I will simply suppose that it worked and the issue is closed.

closed 2015-11-19 by ichrispa