open62541/open62541 #388: Unit tests not building

markusgraube, 2015-09-08:
On the current master branch I am not able to build the unit tests. The error is somehow related with linking the math lib. However, I was not able to link it correctly.

My system is running a 64 bit Kubuntu 15.04

I have a recent GCC Version:
    $ gcc  --version
    gcc (Ubuntu 4.9.2-10ubuntu13) 4.9.2
    Copyright (C) 2014 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

There is also a recent version of check:
    $ aptitude show check
    Paket: check                                    
    Zustand: Installiert
    Automatisch installiert: nein
    Multi-Arch: same
    Version: 0.10.0-2
    Priorität: optional
    Bereich: devel
    Verwalter: Robert Lemmen <robertle@semistable.com>
    Architektur: amd64
    ...

However, the build fails
    $ rm -r *; cmake -DBUILD_UNIT_TESTS=True ..; make 
    -- The C compiler identification is GNU 4.9.2
    -- Check for working C compiler: /usr/bin/cc
    -- Check for working C compiler: /usr/bin/cc -- works
    -- Detecting C compiler ABI info
    -- Detecting C compiler ABI info - done
    -- Found PythonInterp: /usr/bin/python (found version "2.7.9") 
    -- Found Git: /usr/bin/git (found version "2.1.4") 
    -- Git version: v0.1.0-RC4-543-gc2ff8cf-dirty
    -- CMAKE_BUILD_TYPE not given; setting to 'RelWithDebInfo'.
    -- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
    -- checking for one of the modules 'check'
    -- Looking for include file pthread.h
    -- Looking for include file pthread.h - found
    -- Looking for pthread_create
    -- Looking for pthread_create - not found
    -- Looking for pthread_create in pthreads
    -- Looking for pthread_create in pthreads - not found
    -- Looking for pthread_create in pthread
    -- Looking for pthread_create in pthread - found
    -- Found Threads: TRUE  
    -- Configuring done
    -- Generating done
    -- Build files have been written to: /home/mgraube/Dokumente/workingCopies/open62541/build
    [  3%] Generating src_generated/ua_nodeids.h
    [  6%] Generating src_generated/ua_types_generated.c, src_generated/ua_types_generated.h
    [  9%] Generating src_generated/ua_transport_generated.c, src_generated/ua_transport_generated.h
    Scanning dependencies of target open62541-object
    [ 12%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
    [ 15%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o                                                                                             
    [ 18%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o                                                                                         
    [ 21%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o                                                                                     
    [ 24%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o                                                                                                        
    [ 27%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o                                                                                                     
    [ 30%] Building C object CMakeFiles/open62541-object.dir/src/ua_session.c.o                                                                                                           
    [ 33%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server.c.o                                                                                                     
    [ 36%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_addressspace.c.o                                                                                        
    [ 39%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_binary.c.o                                                                                              
    [ 42%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_nodes.c.o                                                                                                      
    [ 45%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_server_worker.c.o                                                                                              
    [ 48%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_securechannel_manager.c.o                                                                                      
    [ 51%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session_manager.c.o                                                                                            
    [ 54%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_discovery.c.o                                                                                         
    [ 57%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_securechannel.c.o                                                                                     
    [ 60%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_services_session.c.o                                                                                           
[... 17 lines cut ...]
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check.o): In Funktion `tcase_create':
    (.text+0x1a4): Nicht definierter Verweis auf `floor'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check.o): In Funktion `tcase_create':
    (.text+0x1b8): Nicht definierter Verweis auf `floor'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check.o): In Funktion `tcase_set_timeout':
    (.text+0x465): Nicht definierter Verweis auf `floor'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check.o): In Funktion `tcase_set_timeout':
    (.text+0x479): Nicht definierter Verweis auf `floor'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check_log.o): In Funktion `subunit_lfun':
    (.text+0x584): Nicht definierter Verweis auf `subunit_test_start'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check_log.o): In Funktion `subunit_lfun':
    (.text+0x63f): Nicht definierter Verweis auf `subunit_test_fail'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check_log.o): In Funktion `subunit_lfun':
    (.text+0x654): Nicht definierter Verweis auf `subunit_test_pass'
    /usr/lib/gcc/x86_64-linux-gnu/4.9/../../../x86_64-linux-gnu/libcheck.a(check_log.o): In Funktion `subunit_lfun':
    (.text+0x66f): Nicht definierter Verweis auf `subunit_test_error'
    collect2: error: ld returned 1 exit status
    tests/CMakeFiles/check_builtin.dir/build.make:133: recipe for target 'tests/check_builtin' failed
    make[2]: *** [tests/check_builtin] Error 1
    CMakeFiles/Makefile2:190: recipe for target 'tests/CMakeFiles/check_builtin.dir/all' failed
    make[1]: *** [tests/CMakeFiles/check_builtin.dir/all] Error 2
    Makefile:86: recipe for target 'all' failed
    make: *** [all] Error 2

Any ideas?

Stasik0, 2015-09-08:
@markusgraube Seems to be a link-configuration problem. I have added some explicit linking dependencies to check. Could you try the check_error branch?

markusgraube, 2015-09-08:
Great!
The check_error branch is working for me

commit 8f311b0f (2015-09-08): adding some explicit linking dependencies to unit tests, fixes #388
closed 2015-09-08 by Stasik0