open62541/open62541 #396: CMake build fails always on any project generation

reza-ebrahimi, 2015-09-10:
Here is the log:

The C compiler identification is MSVC 17.0.50727.1
Check for working C compiler using: Visual Studio 11
Check for working C compiler using: Visual Studio 11 -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Found PythonInterp: C:/Python27/python.exe (found version "2.7.2") 
Found Git: C:/Program Files/Git/cmd/git.exe (found version "1.9.5.msysgit.0") 
Git failed (not a repo, or no tags). Build will not contain git revision info.
Git version: unknown--git-commit-id-unknown
CMAKE_BUILD_TYPE not given; setting to 'RelWithDebInfo'.
CMake Error at CMakeLists.txt:277 (target_compile_definitions):
  Unknown CMake command "target_compile_definitions".

Configuring incomplete, errors occurred!

Stasik0, 2015-09-10:
@reza-ebrahimi it seems to be a problem with CMake, be sure to install the latest version (greater than 3.2)

Stasik0, 2015-09-10:
And you will probably fail to compile, since visual studio 15 is needed to support C99 :(

reza-ebrahimi, 2015-09-11:
@Stasik0 Ook, but there is a big problem here, Visual Studio 2015 only support windows 8 and 10, and many industrial projects are written mostly in windows 7, currently we have project that their libraries compiled with visual studio 2012 and can not move them to greater versions, this is a big problem in using open62541.

reza-ebrahimi, 2015-09-11:
@Stasik0 I'm understand that microsoft does not support C99 standard in its compilers, but must be a workaround here to solve the problem, if any project want to use open62541, must migrate all of the projects and libraries to visual studio 2015, this is impossible for big projects at the moment!

Stasik0, 2015-09-11:
@reza-ebrahimi 
* as you noticed, the problem is in the c99 support by microsoft :/.
* I am sure that vs15 runs on windows7, since we have only windows7 here: https://www.visualstudio.com/visual-studio-2015-compatibility-vs. (we tied the community edition).
* A possible solution would be to use use .dll / static library in your older projects? We get the compiled with vs15 automatically, e.g. here https://ci.appveyor.com/project/Stasik0/open62541/build/465/artifacts

reza-ebrahimi, 2015-09-11:
@Stasik0 I have accomplished following tasks in yesterday:
1- Install Windows 7 Service Pack 1 (64Bit)
2- Install Visual Studio 2015 Enterprise
3- Install Visual Studio 2012
4- Install Visual Studio 2010
5- Install CMake's latest version
6- Install Python 2.7, Active Perl and Ruby
.
.
.
I got the following error in CMake:

CMake Error at CMakeLists.txt:13 (message):
  Python-lxml is not installed.

Configuring incomplete, errors occurred!
See also "D:/Projects/PlantSiS/materials/OPC/open62541/open62541-master/bin/CMakeFiles/CMakeOutput.log".

The lxml library is not correctly installed on WInodws 7 64bit machine, this is a big problem that stopped me in building process, this is very boring process...

Stasik0, 2015-09-11:
I am sorry to hear that, as you can see from our appveyor build script https://github.com/acplt/open62541/blob/master/appveyor.yml,

lxml is installed as
```
python -m pip install lxml
```

reza-ebrahimi, 2015-09-11:
@Stasik0 
There is problem with that command, lxml has dependencies that not correctly installing on machine, I don't know why?!! 
Please share your repository with generated projects for me, This is very critical for me to use this project as my OPC API.

reza-ebrahimi, 2015-09-11:
@Stasik0 My another question is why project use python libs for CMake building process? could be repalce them with another alternatives?

reza-ebrahimi, 2015-09-11:
@Stasik0 I think there is some problems in CMake building, I was install Windows by today, so the platform is pure to install the project, should give me some hints why you use python lxml or another python extensions in CMake? and another tihng is how to replace them with regular stuff in CMake?

HolgerJeromin, 2015-09-11:
python xml stuff ist used for the XML to C code transformation. 
The build process itself does not need this. 

Perhaps the Single-file distribution is usable for you. One one .h and one .c file. 
http://open62541.org/

Stasik0, 2015-09-11:
@reza-ebrahimi pyhton is critical to generate loads of our encoding/decoding functions. 
you can get VS projects here: 
* https://ci.appveyor.com/project/Stasik0/open62541/history (click under build->artifacts)
* or use single-source releases here http://open62541.org/releases/

reza-ebrahimi, 2015-09-11:
@Stasik0 I can't find any build link in given link, so I'm also compiled single header and source file, I have compile errors:

Two CRT and WINSOCK warning that I'm disable them by using their macros, but after that I have the following error:

error C4703: potentially uninitialized local pointer variable 'reftypes' used
in Line 12642 in open62541.c file.

Stasik0, 2015-09-11:
Well you get the files here (from build 468): https://ci.appveyor.com/project/Stasik0/open62541/build/468/artifacts

which files are you using? Try this one http://open62541.org/releases/a9c7eed786.zip

Well, all I can say is that the builds are running here: https://ci.appveyor.com/project/Stasik0/open62541/build/469 (the build server is a windows machine with msvs)

reza-ebrahimi, 2015-09-11:
@Stasik0  same error in same line!

I have replaced given zip files to my newly created win32 console application in Visual Studio 2015 and have compile error,

The running on a machine doesn't throw any causes! my system is purly installed machine by today, Please help me solve the problem.

Stasik0, 2015-09-11:
well, seems to be a trivial fix:

replace "UA_NodeId *reftypes;" by "UA_NodeId *reftypes = UA_NULL;"

reza-ebrahimi, 2015-09-11:
Ok, 
After taking two actions, it gets compiled:

1- first removing the SDL check by setting it to NO in compiler options (but this is not the way)
2- add WinSock version 2 library to linker (Ws2_32.lib)

for item 1 the corrent way is setting variable to NULL, thank you!

reza-ebrahimi, 2015-09-11:
I'm planning to contribute on the project for next days, be aware that my previous problems in building repository still remains, so please take a plan to resolve these problems, 

Thanks a lot...

Stasik0, 2015-09-11:
well, did you try both ways of getting mxml? did you try the manual one https://github.com/acplt/open62541/issues/396#issuecomment-139644832? 

i am sorry, but i cannot help you with pyhon (i do not even have windows/msvs installed here).

reza-ebrahimi, 2015-09-11:
@Stasik0 The problem fixed!

I have install lxml from https://pypi.python.org/packages/3.2/l/lxml/lxml-3.3.5.win32-py3.2.exe#md5=65f2643dc8de8e7c8e2c4c37bb585fc7

closed 2015-09-12 by reza-ebrahimi