open62541/open62541 #399: More callbacks needed

Stasik0, 2015-09-11:
hey gang, 

we need a callback for a normal variable (not a datasource one) that is called on write, so the user application can get updated. So we will have 3 options for a variable:
* value stored in nodestore
* value stored in nodestore + callbacks ***new one***
* datasource callbacks

everyone okay with that?

Stasik0, 2015-09-13:
I'd propose just to add another datasource-similar struct (codename UA_CallBack with onRead and onWrite functions) to variable node, so we do have Variant+UA_CallBack or UA_DataSource in the union, kay?

jpfr, 2015-09-14:
We might need very similar callbacks for subscriptions.
So we don't have to poll the data to find changes.

Stasik0, 2015-09-14:
yes, this will probably be a part of the "event-like" solution that we need for proper events, anyway.

closed 2015-09-17 by Stasik0