open62541/open62541 #403: Bug in processRepeatedJobs()  UA_Server_cleanup() may destroy list of repeatedJobs

joerg555, 2015-09-15:
Please look for this commit:
https://github.com/joerg555/open62541/commit/8330f077d21f3df57b3c6fdf4217b10c41926775
i work with VC2013 / 2015
- start Uaexpert
- make subscription "current time" and "thats it"
- make brakepoint in the first message
- wait about 20s
- continue and server-sample will crash

it was hard to find, but I can to resolve the issue in the usual manner

Please think about to do a save way to prozess jobs.

joerg555, 2015-09-15:
Sorry, you don't need changes in " include/ua_types.h"

jpfr, 2015-09-16:
UA_Server_cleanup does only two things:

     UA_SessionManager_cleanupTimedOut(&server->sessionManager, server, now);
     UA_SecureChannelManager_cleanupTimedOut(&server->secureChannelManager, now);

Imho we should find the culprit in here. I would be interested to see, what code directly writes into the work queue. That's baaad.

We can work around for now by using the LIST_FOREACH_SAFE macro from queue.h.
Instead of the home-grown solution with LIST_NEXT and a while loop.
That allows members to be added, deleted, etc. while we loop over the list.

joerg555, 2015-09-16:
That's no good solution. processRepeatedJobs() has to clean the list thelf after he has done his jobs.
A work arround will be to call "next = LIST_NEXT(tw, pointers);" allways after processJobs() but that's silly too.
So i think we make a new function:
- cleanupJobs(server, ...) 
- and take the next pointer "next = LIST_NEXT(tw, pointers);"  after processRepeatedJobs() if you think it's possible.

jpfr, 2015-09-16:
processRepeatedJobs does not clean the list. The entries are being reordered.

- the list is ordered according to the next timestamp of execution
- we iterate over the elements until we find a repeated job who is repeated in the future
- repeated jobs are reinserted into the list according to their next execution timestamp

To save time for many repeated job, we might use a priority queue (heap) instead of an ordered list.

joerg555, 2015-09-17:
No i have a commit for this problem.
https://github.com/joerg555/open62541/commit/7f8a1ba76ebd8e24ab3065c29e7b8eac8835f294
same was in:
- Subscription_deleteUnpublishedNotification() ...

Stasik0, 2015-09-17:
looks good as far as i can see, picking the next element after reordering is pragmatic

Stasik0, 2015-09-17:
#410

jpfr, 2015-12-12:
This issue is resolved, imho.

closed 2015-12-12 by jpfr