open62541/open62541 #407: Cache or Device read API

reza-ebrahimi, 2015-09-16:
how can perform a cache or device read in open62541 API?

Stasik0, 2015-09-17:
cache or device?

reza-ebrahimi, 2015-09-17:
Yes, The OPC API provides cache or device mechanism that OPC client can tell OPC server read data from its cache or get data from its original source,

The cache can take less time than device read, because device read goes to fetch data from its source (for example from DCS) and send it to client, but in cache read that was already fetched and stays in memory and OPC server read it from its memory and send to OPC client.

Stasik0, 2015-09-17:
OPC API is probably not standardized, isn't it? In OPC UA clients can set a maximal age in their request. However these fields are not checked from the server.

reza-ebrahimi, 2015-09-17:
@Stasik0 Please give me an example.

Stasik0, 2015-09-17:
Our server currently ignores it. Here is a filed maxAge (https://github.com/acplt/open62541/blob/1a03d0286df8600ddbef372b122e2b17f1b8f8cd/tools/schema/Opc.Ua.Types.bsd#L1419), please look up in the specification of OPC UA what it does mean.

reza-ebrahimi, 2015-09-17:
@Stasik0 You were thrown a question in my mind, open62541 client can connect with a industrial OPC server such as Yokogawa opc server and interact with it?

HolgerJeromin, 2015-09-17:
We have successfully connect from the open62541 client to a siemens opc ua server (not DCOM based OPC DA of course).

reza-ebrahimi, 2015-09-17:
@HolgerJeromin That's good news.

Stasik0, 2015-09-17:
probably you will have to set the maxAge manually in the ReadRequest of the client

reza-ebrahimi, 2015-09-17:
@Stasik0 What is the range of maxAge?

Stasik0, 2015-09-17:
unfortunately, you will have to lookup in the opc ua specification. You can get it from opcfoundation for free, you just need to register

reza-ebrahimi, 2015-09-17:
@Stasik0 which features in OPC UA standard are not implemented in open62541? Have you a list?

Stasik0, 2015-09-17:
https://github.com/acplt/open62541/wiki/Feature-list-of-open62541

reza-ebrahimi, 2015-09-17:
Have any plan for Discovering and Address space browsing functionality in future?

Stasik0, 2015-09-17:
Browsing is built in of course! didn't you see our demo client

reza-ebrahimi, 2015-09-17:
@Stasik0 Thanks, but I'm investigating OPC API for my project, so I would like to use open62541, another question is implementation plan for Events in future?

FlorianPalm, 2015-09-18:
We have in mind to support events. But there is no fixed plan at the moment.

closed 2015-09-20 by reza-ebrahimi