open62541/open62541 #408: logger_stdout.h doesn't compile in C++

tpiat, 2015-09-17:
No <B>extern "C"</B> present.

Must be
```
/*
 * This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information.
 */

#ifndef LOGGER_STDOUT_H_
#define LOGGER_STDOUT_H_

#include "ua_types.h"
#include "ua_log.h"

/** Initialises the logger for the current thread. */
#ifdef __cplusplus
extern "C" {
#endif
UA_EXPORT UA_Logger Logger_Stdout_new(void);
#ifdef __cplusplus
}
#endif

#endif /* LOGGER_STDOUT_H_ */
```

joerg555, 2015-09-17:
I have the same, but you can do:
extern "C" {
# include "../examples/logger_stdout.h"
}

Stasik0, 2015-09-17:
probably the first solution is the more elegant one isn't it?

joerg555, 2015-09-18:
Yes of course.
My version is bad and is only to be in sync with your version.

commit 7139e0ad (2015-09-20): integrating some minor fixes proposed by @tpiat, fixes #418, fixes #408
closed 2015-09-20 by Stasik0