open62541/open62541 #417: Subscription Sequence Number

ngoossens, 2015-09-18:
When using UaExpert with stack trace enabled it picks up a issue where the subscription sequence number is invalid.

[image]

Here is an excerpt from the log:
```
17:17:41.296 | General            |                                | Error: Subscription acknowledgement (SubscriptionId=277128017,SequenceNumber=1) returned status BadSequenceNumberInvalid
17:17:41.296 | General            |                                | Error: Subscription acknowledgement (SubscriptionId=277128017,SequenceNumber=1) returned status BadSequenceNumberInvalid
17:17:41.296 | General            |                                | Error: UaSessionPrivate::publishCallback - Notification returned unexpected SequenceNumber (expected=2, received=1)
17:17:41.296 | General            |                                | Error: UaSessionPrivate::publishCallback - Notification returned unexpected SequenceNumber (expected=2, received=1)
17:17:39.296 | General            |                                | Error: Subscription acknowledgement (SubscriptionId=277128017,SequenceNumber=0) returned status BadSequenceNumberInvalid
17:17:39.296 | General            |                                | Error: Subscription acknowledgement (SubscriptionId=277128017,SequenceNumber=0) returned status BadSequenceNumberInvalid
17:17:39.296 | General            |                                | Error: UaSessionPrivate::publishCallback - Notification returned unexpected SequenceNumber (expected=1, received=0)
17:17:39.296 | General            |                                | Error: UaSessionPrivate::publishCallback - Notification returned unexpected SequenceNumber (expected=1, received=0)
17:17:39.296 | General            |                                | Error: UaSessionPrivate::publishCallback - Notification returned unexpected SequenceNumber (expected=1, received=0)
17:17:37.296 | DA Plugin          | open62541                      | Item [NS1|Numeric|96] succeeded : RevisedSamplingInterval=250, RevisedQueueSize=1, MonitoredItemId=277128018 [ret = Good]
17:17:37.296 | DA Plugin          | open62541                      | CreateMonitoredItems succeeded [ret = Good]
```

This seems to be due to two slight variations from the specification (Part 4 - 5.13.1.1).

Firstly
*The value 0 is never used for the sequence number. The first NotificationMessage sent on a Subscription has a sequence number of 1.*
Should the sequenceNumber be initialised ([ua_subscription.c line 15](https://github.com/acplt/open62541/blob/master/src/server/ua_subscription.c#L15)) to 1?

Secondly
*Sessions maintain a retransmission queue of sent NotificationMessages. NotificationMessages are retained in this queue until they are acknowledged.*
Currently there is no such queue and NotificationMessages are kept in the unpublishedNotifications queue until they are acknowledged. This seems to lead to multiple transmissions of the same NotificationMessages which causes the errors noted above. Should a second publishedNotifications queue be created to which we move notifications when they are published ([ua_services_subscription.c line 182](https://github.com/acplt/open62541/blob/master/src/server/ua_services_subscription.c#L182)) and then delete from this queue when they are acknowledged ([ua_services_subscription.c line 151](https://github.com/acplt/open62541/blob/master/src/server/ua_services_subscription.c#L151))?

ichrispa, 2015-09-18:
Firstly: UpdateNotification preincrements the seq. number, so the first number being used is always one.
Secondly: A client needs to acknowledge the received notifications in its publish (no matter what queue mechanism is used). The server removes all acknowledged notifications prior to anything else. If a client send multiple publish acknowledgements without waiting for the result, then the error message is valid. A second queue would not help, as the message may actually be lost and then becomes unretrievable.

A second queue is an utterly dirty workaround. Propper asynchronous messaging will be available soon to remedy the missing queue feature.

ngoossens, 2015-09-19:
Hi @ichrispa,

Thanks for the details and the correction! I look forward to the coming async version. 

My understanding regarding missed messages that they were meant to be retrieved using republish service, which I understand is not implemented (https://github.com/acplt/open62541/issues/370). Based on my investigations with UaExpert the messages were being acknowledged but there was a time delay resulting in the messages being sent multiple times.

ichrispa, 2015-09-22:
Hi @ngoossens,

I am having trouble reproducing your errors. I am using UAExpert 1.2.2, linux/suse x86_64 and the example server from our current master. Here's the options:

[image]

Could you post the config/steps you used to produce the given errors and kindly provide infos on you os/versions/stack version?

Regards,

ichrispa

Stasik0, 2015-09-22:
@ichrispa jfyi, current eaexpert is 1.3.1

ngoossens, 2015-09-22:
Hi @ichrispa,

The tests were done with UaExpert 1.3.1 Build 206 on Windows 8.1, Windows XP and Ubuntu 14.04 (32bit). The current master of open62541 server was tested running on Linux (32bit and ARM) and OS X. The ENABLE_SUBSCRIPTIONS, ENABLE_METHODCALLS, GENERATE_SELFSIGNED, BUILD_EXAMPLESERVER and ENABLE_ADDNODES options were all turned on for the build.

To reproduce the issue I connect to the server and drag in the current time variable.

The same configuration as yours was used Trace.StackTraceEnabled: true, Trace.StackTraceLevel: DEBUG.
[image]

Here is a screenshot of the issue on Ubuntu:
[image]

Please let me know if there is any additional information I can provide to help.

Regards,
Nick

ichrispa, 2015-09-22:
@Stasik0: 1.3.1 crashed on OpenSUSE 13.2 (segfaults right when accessing it's QT libs).

Stasik0, 2015-09-22:
Ubuntu >= OpenSUSE in this case ;))

ichrispa, 2015-09-22:
@Stasik0: really, ubuntu? are you really really sure you want to start this discussion? Opensuse is a real linux, not the boiled down, amazon funded, novice-user replacement for windows where the solution to a problem is typing forum entries one does not understand into a shell... *breeaaath*

Anyway, back to something more productive then linux-distro-discordance.

 So 1.2.2 does not complain one bit. 1.3.1 compains a lot. So, I ran a bit of a wireshark trace and printf debugging:

[image]

To explain this problem, here's what happens: Every once in a while, the server will not send a publishResponse, because UAExpert apparently requests data a bit faster than the publishInterval permits. So the server is inclined to send a keepalive instead of a response, as he should.

From the spec:
    Subscriptions have a keep-alive counter that counts the number of consecutive publishing
    cycles in which there have been no Notifications to report to the Client. When the maximum
    keep-alive count is reached, a Publish request is dequeued and used to return a keep-alive
    Message. This keep-alive Message informs the Client that the Subscription is still active. Each
    keep-alive Message is a response to a Publish request in which the notificationMessage
    parameter does not contain any Notifications and that contains the sequence number of the next
    NotificationMessage that is to be sent. 

Granted, we do not use the state machine indicated in the spec, as this is a literal quote from the foundation API. But we do use the keepalive counter and because we do not yet have async capabilities, keepalives are the method of choice for saying "no data here". The stack gets a keepalive message (no notification, incremented sequence number, no indices).

And here is what I believe is wrong with 1.3.1: 
When the server does have data after a bit, he send the message with the propper seqNo, which is one less then what UAExpert saw last in the keepalive. The response message is fine though. But UAExpert complains about the seqNo mismatch, even though it should detect that the messages before where keepalives, not notifications.

@ngoossens: I did however fix the preincrement issue, so seqNo start with 1 now. This will not fix this issue.

Stasik0, 2015-09-22:
;))

Cant we rise this issue on ua's forum?

ichrispa, 2015-09-22:
Yes, I think we can. I would be interested in why UAExpert thinks it can (a) spam the server with requests outside the interval (which is not forbidden, just unreasonable) and (b) does not recognize keepalives as what they are.

We may have a wrong take on subscriptions, since the whole deal is quite complex. But I think I can make a stand ;)

Stasik0, 2015-09-22:
Are you going to open it?

jpfr, 2015-09-22:
We can also raise the issue in the opc foundation bug tracker.
They are very active in improving the standard where it needs clarification.

Stasik0, 2015-09-22:
I think you are the only one with access?

ichrispa, 2015-09-23:
@Stasik0: How do I open a ticket? I have neither a developer mail or customer ID required for operning tickets.

Stasik0, 2015-09-23:
@ichrispa http://forum.unified-automation.com/forum22.html I have opened a ticket there (search for Stasik0) after a registration with some disposable mail

ngoossens, 2015-09-23:
Thanks everyone for digging into this issue. I look forward to hear what UA has to say on the matter.

@ichrispa your explanation makes sense from what I was seeing too, so thanks for that. Also thanks for fixing the seqNo start number.

ichrispa, 2015-09-23:
@ngoossens: my pleasure :)

Stasik0, 2015-09-24:
@ichrispa @ngoossens well, i am sorry if you misunderstood me, but i did not open any ticket regarding this issue on ua-forum, so someone of you two has to do it

ngoossens, 2015-09-29:
@Stasik0, @ichrispa I have posted a question to the UaExpert forum here: http://forum.unified-automation.com/topic1624.html

If you have anything additional to add to that post please do.

Stasik0, 2015-09-29:
@ichrispa they answered, they blame us for not having async responses, hmmm

ngoossens, 2015-10-01:
It seems that they are sending the additional publish request to ensure the server's queue always has publish requests so alarms, etc can be sent immediately.

So until the async implementation this will remain an issue?

jpfr, 2015-10-01:
First, we **do have async** in the server. And we always had.
In theory, we can send a message over an open UA_Connection any time. But we don't, since OPC UA does not work that way.

The server can only respond to requests. But it can choose to respond to requests out-of-order.
That's why requests have **RequestIds** so the response can be matched.
In the case of subscriptions, the server receives some publish requests and deliverately puts them on hold (does not send a response). The response (to a publish request) follows when it gets triggered by an internal event.

The **SequenceNumber comes from the SecureChannel**. All sent messages (chunks) get an increasing sequence number. That has nothing to do with subscriptions!
The existence of a function `Subscription_getAvailableSequenceNumbers` indicates that something is wrong in our implementation.

Second, having a second queue where notifications are stored until an ack is received is exactly right. @ichrispa you seem opposed. Do you have another design in mind, where notifications can be re-requested when they were lost in a broken connection?

jpfr, 2015-10-01:
Update: Sorry. My bad. The term sequence number is overloaded. The subscriptions themselves also have sequence numbers on top of those from the secure channel.

ichrispa, 2015-10-05:
@jpfr: 
For me, the desired async capability is getting a request but not sending a reply, which is currently manditory in ua_server_binary. So in essence, there should be a way of saying "no return data" when exiting a service call. Afaik we cannot do that right now.

I am opposed to the second queue because of the management overhead. I would suggest we add a boolean flag to the notifications to mark them as sent, which should serve the same purpose. Searching for unsent subscriptions then simply modifies the loop/iterator over lists. Much easier and a lower code footprint then a second queue.

jpfr, 2015-10-05:
We don't have to answer in ua_server_binary. Here's a snippet from the big switch that selects services. We can just as well not call UA_SecureChannel_sendBinaryMessage.

```c
    case UA_NS0ID_GETENDPOINTSREQUEST: {
        UA_GetEndpointsRequest p;
        UA_GetEndpointsResponse r;
        if(UA_GetEndpointsRequest_decodeBinary(msg, pos, &p))
            return;
        UA_GetEndpointsResponse_init(&r);
        init_response_header(&p.requestHeader, &r.responseHeader);
        Service_GetEndpoints(server, &p, &r);
        UA_GetEndpointsRequest_deleteMembers(&p);
        UA_SecureChannel_sendBinaryMessage(clientChannel, sequenceHeader.requestId, &r,
                                           &UA_TYPES[UA_TYPES_GETENDPOINTSRESPONSE]);
        UA_GetEndpointsResponse_deleteMembers(&r);
        break;
    }
```

Ok. So you want to flag them as "sent" in the queue and remove them when they were acknowledged?

ichrispa, 2015-10-06:
Hi @jpfr,
I was actually trying to avoid touching the ua_server_binary, but since that appears to be the only way, I'll do that. I will be returning NULL as Response from service publish to indicate that the response whould be suppressed.

    Ok. So you want to flag them as "sent" in the queue and remove them when they were acknowledged?

yes.

ngoossens, 2015-10-06:
Hi @ichrispa,

I am glad there seems to be a reasonable solution to this in the end.

Using a sent flag would be the least overhead in terms of queue management but I am curious as to how this would impact:
1. The error condition e.g. when some messages were missed/not acknowledged. Do we now loop over the missed messages for every future transmission?
2. Management of the retransmission queue size? As the flagged messages effectively become the retransmission queue for use by the  republish service (not yet implemented). From the spec: *The Session shall maintain a retransmission queue size of at least two times the number of Publish requests per Session the Server supports.*

Thanks in advance for the assistance just trying to clarify things in my own mind.

ichrispa, 2015-10-07:
Hi @ngoossens,

The problem you describe in 1 is maintained with a queue. If the client simply doesn't acknowledge messages that where sent, we end up with a big transmitted queue that won't empty. In essence, the boolean flag can be seen as a list qualifier: Sent=False is the new notifications queue, Sent=True is the transmitted queue.

2 is equally solvable. If a message is being flagged as "sent" and the count of "sent" notifications is larger then 2*publishPool, then the oldest sent notification will be removed. Right now we do not have a limitation in terms of max publish req. per session. The publish queue described in the spec thereby becomes a simple counter in the clients subscriptionManager.

Best regards,

ichrispa

ngoossens, 2015-10-08:
@ichrispa, thanks for the details.

ichrispa, 2015-11-25:
flagging is now present with efae154

Stasik0, 2015-11-27:
@ngoossens is it solved? is the error gone?

ichrispa, 2015-11-27:
@Stasik0: The publishedOnce-Notifications-Queue won't solve ngoosens problem in itself; we still need to go a bit until we reach the out-of-band reply spec part of subscriptions. We're getting there though :)

jpfr, 2015-12-12:
Commit 360f03b moved the call to `UA_SecureChannel_sendBinaryMessage` into Service_Publish itself. But that just moved the functionality to a new place without touching the underlying behaviour.

Now it should be doable to move the response logic into a separate function that is called asynchronously with a timed job.

ichrispa, 2015-12-12:
Hi @jpfr,

I'd like to do it a bit differently. The publish service is strictly not responsible for sending that message. Also, the PublishResponse is never really send "asynchronously" but is also request-driven. What happens is that when the client sends any message and the is a notification pending, that notification gets send out of order in addition to the clients request.

I want to put the out-of-sequence logic in front of the processBinaryMessage service case section. I will send out-of-order notifications from there, while the publish service will update the queue. If the publish does not return a message, it will free the response and set its pointer to null to indicate the absence of a publish response.

Also, please note that there are Jobs being created for any monitoredItem with a samplingInterval. So as long as the interval is not "0", we are already sampling asynchronously.

Regards,

ichrispa

jpfr, 2015-12-12:
Hey @ichrispa 

> The publish service is strictly not responsible for sending that message.

Exactly. With the latest change, the Publish service can choose not to send a response.
Before it had to.

> I want to put the out-of-sequence logic in front of the processBinaryMessage service case section. I will send out-of-order notifications from there, while the publish service will update the queue.

Branching of publish response before processBinaryMessage means that we will not send updates when there are no requests. Part 4 of the standard is very detailed on the subscription mechanism. And it wants a cyclic, i.e. timed, response behavior.

There are two **separate cyclic mechanisms** at play.
- Checking attributes for updates and creating notifications (MonitoredItem)
- Sending out currently queued notifications (Subscriptions)

> 5.13.1 Subscription model
[...]
Subscriptions have a publishing interval. The publishing interval of a Subscription defines the
cyclic rate at which the Subscription executes. Each time it executes, it attempts to send a
NotificationMessage to the Client. NotificationMessages contain Notifications that have not yet
been reported to Client.

> NotificationMessages are sent to the Client in response to Publish requests. Publish requests
are normally queued to the Session as they are received, and one is de-queued and processed
by a subscription related to this Session for each publishing cycle, if there are Notifications to
report. When there are not, the Publish request is not de-queued from the Session, and the
Server waits until the next cycle and checks again for Notifications.

> Also, please note that there are Jobs being created for any monitoredItem with a samplingInterval. So as long as the interval is not "0", we are already sampling asynchronously.

That's great. :-)
If MonitoredItems are already cyclic, we have a working precedent for Subscriptions as the second cyclic mechanism.

ngoossens, 2015-12-13:
@Stasik0, sorry for the delayed reply, I have been traveling and AFK.

It seems @ichrispa is hard at work solving this one, thanks! Interestingly the error message no longer appears in UaExpert on the master branch.

Stasik0, 2015-12-14:
can i close the issue then?

closed 2016-01-20 by ichrispa