open62541/open62541 #420: open62541 client can't connect to MatrikonOPC UA Wrapper for COM Servers

reza-ebrahimi, 2015-09-20:
open62541 client can't connect to MatrikonOPC UA Wrapper for COM Servers, ```UA_Client_connect()``` returns ```UA_STATUSCODE_BADNONCEINVALID```.

```
	UA_Client *client = UA_Client_new(UA_ClientConfig_standard, Logger_Stdout_new());
	UA_StatusCode retval = UA_Client_connect(client, ClientNetworkLayerTCP_connect,
		"opc.tcp://localhost:21381/MatrikonOpcUaWrapper");
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return retval;
	}
```

Stasik0, 2015-09-20:
@reza-ebrahimi try this one https://github.com/acplt/open62541/tree/nonce please, otherwise we will need to read spec and watch wireshark logs for working clients

Stasik0, 2015-09-22:
@reza-ebrahimi did it work out?

reza-ebrahimi, 2015-09-22:
@Stasik0 There is a problem, How can I generate two header and source (open62541.h and open62541.c) single files? I used two single files in my project not repository project, Please tell me and I'll test it by today.

FlorianPalm, 2015-09-22:
You have to change the CMake configuration -> activate ENABLE_AMALGAMATION

reza-ebrahimi, 2015-09-22:
@FlorianPalm Thanks a lot...

reza-ebrahimi, 2015-09-27:
@Stasik0 Still I get ```UA_STATUSCODE_BADNONCEINVALID``` in ```UA_Client_connect()``` function with https://ci.appveyor.com/project/Stasik0/open62541/build/560/artifacts single header and source files.

Stasik0, 2015-09-27:
Is there a demo version of the wrapper i can debug?

reza-ebrahimi, 2015-09-27:
MatrikonOPC Wrapper could be downloaded from here https://www.matrikonopc.com/opc-ua/products/ua-wrapper-com-opc-servers.aspx
unfortunately that was an executable file and should be installed on system, after that wrapper wraps OPC DA servers installed on system and give you a Opc Ua server address, simply I have changed the address path in ```UA_Client_connect()``` function.

Stasik0, 2015-09-27:
which OPC DA server are you using?

reza-ebrahimi, 2015-09-27:
Matrikon OPC Server for Simulation

Stasik0, 2015-09-27:
okay, without tested it yet, could you try that one http://open62541.org/releases/94c0250da6.zip

reza-ebrahimi, 2015-09-27:
Wow, That's great, ```UA_Client_connect()``` returns ```UA_STATUSCODE_GOOD```,
Thank you, but what happens?

Stasik0, 2015-09-27:
i have set nonce to be complete empty, it seems that is required by the server (and probably this is standard-conform way to do it). I will reintegrate fix into the master.

Stasik0, 2015-09-27:
@reza-ebrahimi i reopened it, will have to test tommorow

reza-ebrahimi, 2015-09-27:
@Stasik0 Ook, Thank you, can I have your email address?

Stasik0, 2015-09-27:
if i post it here - a spam wave will arrive here, have a look into the chat: http://webchat.freenode.net/?channels=%23open62541

commit 67323475 (2015-09-27): Merge pull request #424 from acplt/nonce

sending a 0-initalized nonce, fixes #420
commit 32c50500 (2015-09-27): Merge pull request #425 from acplt/client_fixes

making client more conformant, fixes #420
closed 2015-09-27 by Stasik0