open62541/open62541 #421: C++ Compilation  for QNX or MinGW CGG 4.4.0

tpiat, 2015-09-21:
Macros 
UA_LocalizedText
UA_QualifiedName
UA_ExpandedNodeId
do not work. 

Insted using it I wrote in <B>.c</B> file
```
void createQualifiedName(UA_QualifiedName *qn , UA_UInt16 nsid, const char *string)
{
	*qn = UA_QUALIFIEDNAME(nsid, string);
}

void createExpandedNodeId(UA_ExpandedNodeId *eni, UA_UInt16 nsid, UA_UInt32 numericid)
{
	*eni = UA_EXPANDEDNODEID_NUMERIC(nsid, numericid);
}

void createLocalizedText(UA_LocalizedText *lt, const char* locale, const char *text)
{
	*lt = UA_LOCALIZEDTEXT(locale,text);
}
```
Next, I create varibles, init my variables by this fuctions and use this variables in 
UA_Server_addObjectNode & etc.

Stasik0, 2015-09-21:
compiling with g++ works for https://github.com/acplt/open62541/blob/master/examples/server.cpp
can these be inline functions? Question for all: can we replace these macros with inline functions?

tpiat, 2015-09-21:
I was trying to make inline functions but was memory leaks.
I.e.
x =f() -- memory leaks,
f(&x) -- works.

Stasik0, 2015-09-21:
locate this line
```
//#define UA_QUALIFIEDNAME(NSID, CHARS) (const UA_QualifiedName) { .namespaceIndex = NSID, .name = UA_STRING(CHARS) }
```
replace it by
```
static UA_INLINE UA_QualifiedName UA_QUALIFIEDNAME(UA_UInt16 nsid, char *string){
return (const UA_QualifiedName) { .namespaceIndex = nsid, .name = UA_STRING(string) };
}
```
does it compile/work/leak/?

jpfr, 2015-09-21:
> Question for all: can we replace these macros with inline functions?

Yes. I'm working on that branch since a week. Will land soon.

tpiat, 2015-09-21:
At home, gcc version 4.6.2
Do not compile.
open62541.h: In function <UA_QualifiedName UA_QUALIFIEDNAME(UA_UInt16, char*)>:
open62541.h:758:9: error: expected primary-expression before <const>
open62541.h:758:9: error: expected <)> before <const>

Other results (if possible) will be tomorrow.

Stasik0, 2015-09-22:
'''
static UA_INLINE UA_QualifiedName UA_QUALIFIEDNAME(UA_UInt16 nsid, char *string){
return (UA_QualifiedName){ .namespaceIndex = nsid, .name = UA_STRING(string) };
}
'''?

tpiat, 2015-09-22:
In C file -- works.
In C++ file -- not compiled.
```
In file included from ../opcuaServer.usercallback/ua_server.h:24,
                 from ../opcuaServer.usercallback/main.cpp:9:
../opcuaServer.usercallback/ua_types.h: In function 'UA_QualifiedName UA_QUALIFIEDNAME(UA_UInt16, char*)':
../opcuaServer.usercallback/ua_types.h:431: error: expected primary-expression before ')' token
../opcuaServer.usercallback/ua_types.h:431: error: expected ';' before '{' token
../opcuaServer.usercallback/ua_types.h:431: error: expected primary-expression before '.' token
../opcuaServer.usercallback/ua_types.h:431: error: expected primary-expression before '.' token
../opcuaServer.usercallback/ua_types.h:431: error: expected ';' before '}' token
```

Stasik0, 2015-09-22:
well.. it tried to play with it.. no idea what's the problem.... you can probably use your workarounds... and we stay with 4.8. Can you write them down please so we can log the issues in FAQ?

tpiat, 2015-09-22:
I'm forced to use 4.4.0 for QNX.
I wrote solution in the top message.
If you are using the gcc compiler version is less than 4.7, in the "C" file need to place the function of the first message.

```
	UA_QualifiedName qn;
	createQualifiedName(&qn,1, "Name");
	UA_ExpandedNodeId eni;
	createExpandedNodeId(&eni,0, UA_NS0ID_FOLDERTYPE);
	UA_LocalizedText lt;
	createLocalizedText(&lt,"en_US","Text");

	UA_Server_addObjectNode(server,
						  UA_NODEID_STRING(1, (char *)"Node"),
						  qn,
						  lt,
						  lt,
						  0, 0, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
						  UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
						  eni,
						  NULL);
```

Stasik0, 2015-09-22:
added to FAQ in Wiki

closed 2015-09-22 by Stasik0