open62541/open62541 #422: False positive or not?

Stasik0, 2015-09-22:
Coverity triggered this https://scan8.coverity.com/reports.htm#v22486/p10303/fileInstanceId=8532366&defectInstanceId=3003848&mergedDefectId=130697&eventId=3003848-7

```
1686  arrayOfNodeIds componentRefs = (arrayOfNodeIds) {
1687    .size = 2,
1688    .ids  = (UA_NodeId[]) { UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY)}
1689  };
```

and then

```
1707  UA_Server_addInstanceOf_instatiateChildNode(server, &subtypeRefs, &componentRefs, &typedefRefs,
1708                                              objectRoot, callback, (UA_ObjectTypeNode *) typeDefNode, 
1709                                              UA_TRUE, &instantiatedTypes, handle);
```

apperently Coverity thiks that `typedefRefs.ids` went out of scope since it thinks that the definition is limiting the scope? Does it think the initializer is just a scope block `{ ... }`?

Stasik0, 2015-09-22:
http://git.greensocs.com/fkonrad/mttcg/commit/b0af844007841609cc11fab58f838bd105cbe144 same problem appeared here, so they used explicit memcpy to copy the content over... Do I understand C wrong?

ichrispa, 2015-09-22:
The idea here is the following: To create an instance, there are three types of references of interested: typeDefinitions, subTypes and hasComponent. Each includes all subtypes of each toplevel reference. These are not collected yet, but provisions where taken to make it possible.

the addInstanceOf function creates an arrayOfNodeIds for each of these three and their subtypes.

addInstanceOf is a toplevel function. Separate (similar) functions are called for components and supertypes, but non construct the arrayOfNodeIds - the arrays are always passed as a pointer. So these arrays only stop existing after the toplevel function terminates. I see no way these arrays could possibly vanish from the stack. They are not memcpy'ied.

The macros (inArray) does not return or modify anything. It simply scans the array using NodeId_equal to determine if a nodeid exists in the array. An external, local boolean is used as an indicator. So not even the counting index is reused.

In short: I consider this message to be a false positive by coverity.

What is memcpy'ied is the list of visited nodes (actually realloced), but coverity does not complain about that.

I would propose to not do anything about this. From our IRC discussion I understood that @jpfr has  opted to reimplement large parts of the HL-Branch. Since he appears to have the time, I suggest we let him reimplement the addInstanceOf. Undoubtedly, he will fix this issue too.

Stasik0, 2015-09-23:
yea, the commit fixed it looks like
```
1686  arrayOfNodeIds componentRefs = (arrayOfNodeIds) {
1687    .size = 2,
1688    .ids  = (UA_NodeId[]) { UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY)}
1689  };
```
here the array inializer is considered a local variable and is garbage after line 1689, this remedies it:
```
  UA_NodeId* tempArrayComponentRefs = (UA_NodeId[]) { UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), UA_NODEID_NUMERIC(0, UA_NS0ID_HASPROPERTY)};
  arrayOfNodeIds componentRefs = (arrayOfNodeIds) {
    .size = 2,
    .ids  = tempArrayComponentRefs
  };
```

closed 2015-09-23 by Stasik0