open62541/open62541 #423: Project compilation fails with error "cmd.exe" exited with code 1. in Windows

reza-ebrahimi, 2015-09-26:
Project compilation fails with error "cmd.exe" exited with code 1. in Windows

Stasik0, 2015-09-26:
Yea, we need more info.

Did the matrikon stuff work?

reza-ebrahimi, 2015-09-26:
@Stasik0 I can't test it in these days, this is my first try after posting previous issue, I want to make the project with ```ENABLE_AMALGAMATION``` to use in my project, but there building process fails as posted above.

Stasik0, 2015-09-27:
@jpfr is it a problem with spaces inside of the path?
@reza-ebrahimi cant you take these ones https://ci.appveyor.com/project/Stasik0/open62541/build/560/artifacts?

reza-ebrahimi, 2015-09-27:
@Stasik0 Still I get ```UA_STATUSCODE_BADNONCEINVALID``` in ```UA_Client_connect()``` function with https://ci.appveyor.com/project/Stasik0/open62541/build/560/artifacts single header and source files.

Stasik0, 2015-09-27:
@reza-ebrahimi okay, lets get back to original issue, can you please vary your path

* not to contain space
* make sure you can write to it

reza-ebrahimi, 2015-09-27:
@Stasik0 path has no spaces: ```D:\Projects\open62541\open62541\bin``` and I can write on it, I have no problem with that,

but you have to close my another issue very very soon, so I have one question about that, in botttom lines of code :
```if (resp.responseHeader.serviceResult == UA_STATUSCODE_GOOD &&```, ```serviceResult``` is ```UA_STATUSCODE_BADINVALIDTIMESTAMP``` and can't read value.

Stasik0, 2015-09-27:
which pyhon are you using?

reza-ebrahimi, 2015-09-27:
Python 3.2.3

closed 2015-10-19 by reza-ebrahimi