open62541/open62541 #426: Object and Variable Instantiation

jpfr, 2015-09-28:
Hey @ichrispa,

ua_types.h defines
`typedef UA_StatusCode (*UA_InstantiationCallback)(UA_NodeId objectId, UA_NodeId definitionId,
void *handle);`

I'd like to move instantiation into the AddNodes service.
The UA_AddNodesItem structure has a special field `typeDefinition` that points to the ObjectType / VariableType that shall be instantiated.

Can we **add the instantiationcallback to all UA_ObjectTypeNodes and UA_VariableTypesNodes**?
The callback would normally be null, but can be set internally like an attribute.

ichrispa, 2015-10-02:
Since you are rewriting the hl-api, move it anywhere you please. But it is definetely required for Instantiation.

I do not really see the benefit of the callback for all types. The addXYNode functions return the nodeID of the created node, so it is always possible to identify and modify them. Instances create multiple nodes, so that mechanism is not possible there.

ichrispa, 2015-10-15:
I would also like to point out that there is a bunch of client-side functions waiting to be turned into a symmetric api in the hl-branch. And there is a set of tutorials there that will need to get retested and slightly adapted to the imposed API changes.

closed 2015-11-19 by ichrispa