open62541/open62541 #427: VS 2015 Simple Gui Example C++ MFC

joerg555, 2015-09-30:
Now i habe build a new network-Layer in c++ with MFC CSocket interface and like to publisch it.

There are VS2015 Projekt-Files without need of cmake, you only need python to be installed 
in "C:\Python27" or with path from registry.
all source-generation is done with "xxxx.cmd" in the projekt.
I only need a path for the example to be compatible with the master.

./example-mfc of ./simplemsvc ?

Stasik0, 2015-10-01:
Hi joerg, which dependencies do you have? Can I see the files somewhere? Do you depend on open62541.h and open62541.c or on something more?

jpfr, 2015-10-01:
Hey @joerg555,
we currently have a cross platform networklayer (with some ifdefs).
What are the advantages of replacing it?

Also, depending on C++ in the main repository requires a pretty fundamental decision.

jpfr, 2015-10-01:
By the way, the title of the issue is "Simple Gui Example".
What does the network layer have to do with graphical user interfaces?

joerg555, 2015-10-01:
Hi @Stasik0 ,
i have no special dependencies.
I am in "./open62541/SimpleMSVC" and use:
../deps, ../examples, ../include, ../src, directory.
My Project generate to
./open62541/SimpleMSVC/src_generated
I have my own network-layer because i nedd async network access.
I like to have no **#ifdef** in basic source code.
Sorry, but i can never use "ua_server_worker".
I am intended to use malloc() and free() only in one or two screenpages of code.
The last changes in processBinaryMessage() are verry good, no malloc and free i use it now.
I will reset my fork and make a branch with my projekt.

Stasik0, 2015-10-01:
@joerg555 well please not, I'd like to have a look on the deleteAll flags of the subscriptions

joerg555, 2015-10-01:
Hi @jpfr 
I use MFC with CSocket, that's allways Gui.
They work with Windows message-loop in single thread.
I will post my sample soon see above

joerg555, 2015-10-01:
Hi @Stasik0,
the only thing of deleteAll is to delete all in one loop, and have no problem with "verschachtelte Listen".
The parameter is not very nice, but it makes the diffrence to find all who use it.

joerg555, 2015-10-04:
Hi @jpfr 
we don't need to put my C++ code in main repo, but themain code can be C++ clean.
I will try to make my C++ samples in the branch.
It will be nice, when you check my pull requests to give me the needed interface so that i can work with your main repro without changes.
I dont know if it's good to make all pull request in one or have discussion in 2 or 3 pull requests.

https://github.com/acplt/open62541/pull/428

regards joerg555

joerg555, 2015-10-09:
I have updated my sample and its working a little bit now.
Next step is to integrate own variables and values from the dialog, may be in 2 weeks.

Stasik0, 2015-10-09:
Can we see a demo/binary?

joerg555, 2015-10-10:
Sorry, i am on holyday now :).

jpfr, 2015-12-12:
Closing the issue, as it is no immediate work item.
@joerg555 Feel free to still post a demo in this thread! :+1:

closed 2015-12-12 by jpfr