open62541/open62541 #429: need more documentation for functions

HolgerJeromin, 2015-10-03:
For example all public ua_client functions needs a doxygen comment...
especially these from the examples.

I still have no idea which is the correct way to free a 
`UA_STRING 	foo = UA_String_fromChars("bar");`

This code crashes:
`UA_String_delete(&foo);`

relates to #332

Stasik0, 2015-10-04:
This is simple

*_delete methods free content of the struct + struct
*_deleteMembers methods free content of the struct only

Since the string struct itself is in the stack - take deleteMembers.

HolgerJeromin, 2015-10-04:
Thanks. Should go at least in an comment.

jpfr, 2015-10-04:
The documentation needs a serious upgrade.
Our codebase is nice but completely unusable without.

Handling types is described here: http://open62541.org/doc/sphinx/datatypes.html

HolgerJeromin, 2015-10-04:
I will try to add some comments in the client code I understand.

tpiat, 2015-10-11:
```
UA_STRING foo = UA_String_fromChars("bar");
UA_String_delete(&foo);
```
In c++ gives memory leak if not crashed.

Stasik0, 2015-10-11:
_delete tries to free the pointer itself which is on the stack, try _deleteMembers, it frees only content, no pointer itself

HolgerJeromin, 2015-11-04:
Do we want a huge amount of repeated comment like
```
/* free content of the struct and the struct itself */
void UA_EXPORT UA_Float_delete(UA_Float *p);
/* free content of the struct only */
static UA_INLINE void UA_Float_deleteMembers(UA_Float *p) { }
```
for all datatypes? I added this in in all the C files and in the python script. 
Is this useful for help in the IDE or is this not wanted?

closed 2016-02-09 by Stasik0