open62541/open62541 #434: BadEncodingLimitsExceeded

osaiz, 2015-10-08:
I am writing a OPC-UA server based on open62541.
If I publish a limited number of variables (let's say 80), it goes OK.

But, if I try to publish a larger number of variables (let's say 200) it does not.
When I connect my server with UaExpert, the variables are not displayed, as I get "BadEncodingLimitsExceeded" error.

What could I do?

Pd.- I'm using trunk version of 2015/09/28

Stasik0, 2015-10-08:
what do you mean by publish? Do you mean a subscription?

osaiz, 2015-10-08:
By "publish" I mean adding variables to my server, via
UA_Server_addDataSourceVariableNode(...).

Stasik0, 2015-10-08:
Is the browse request failing? Did you try to tweak the buffers?

osaiz, 2015-10-08:
First I get a “Browse succeeded”, but inmediatly after I get “Browse failed”.

Here it is the whose sequence of messages displayed by the client:

10:33:15.808 | Server Node        | open62541 - None - None (ua... | Connection status of server 'open62541 - None - None (uatcp-uasc-uabinary)' changed to 'Connected'.
10:33:10.640 | AddressSpaceModel  | open62541 - None - None (ua... | Browse failed with error 'BadConnectionClosed'.
10:33:10.633 | Server Node        | open62541 - None - None (ua... | Connection status of server 'open62541 - None - None (uatcp-uasc-uabinary)' changed to 'ConnectionErrorApiReconnect'.
10:33:10.628 | AddressSpaceModel  | open62541 - None - None (ua... | Browse failed with error 'BadEncodingLimitsExceeded'.
10:33:10.617 | AddressSpaceModel  | open62541 - None - None (ua... | Browse succeeded.
10:33:10.614 | Server Node        | open62541 - None - None (ua... | Revised values: SessionTimeout=10000, SecureChannelLifetime=600000
10:33:10.614 | Server Node        | open62541 - None - None (ua... | Successfully connected UA server.
10:33:10.613 | Server Node        | open62541 - None - None (ua... | Connection status of server 'open62541 - None - None (uatcp-uasc-uabinary)' changed to 'Connected'.
10:33:10.613 | AddressSpaceModel  | open62541 - None - None (ua... | Registered for ModelChangeEvents
10:33:10.585 | Server Node        | open62541 - None - None (ua... | The server returned no certificate, all certificate checks will be skipped.
10:33:10.584 | Server Node        | open62541 - None - None (ua... | ApplicationUri: 'urn:unconfigured:open62541:open62541Server'
10:33:10.584 | Server Node        | open62541 - None - None (ua... | Found security policy 'http://opcfoundation.org/UA/SecurityPolicy#None'
10:33:10.584 | Server Node        | open62541 - None - None (ua... | Found endpoint 'opc.tcp://linuxnew:16664'
10:33:06.737 | DiscoveryWidget    |                                | Discarding Server Advosol UA Discovery Server with URL http://localhost:52601/UADiscovery/discovery
10:33:06.737 | DiscoveryWidget    |                                | Discarding Server Advosol UA Discovery Server with URL https://localhost:4843/UADiscovery
10:33:06.737 | DiscoveryWidget    |                                | Discarding Server Advosol UA Discovery Server with URL opc.tcp://localhost:4840/UADiscovery

The server also gets a message:
[10/08/2015 08:33:11.927.821] info/communication        Unknown request: NodeId(ns=0, i=832)
[10/08/2015 08:33:11.928.328] info/communication        The message was not entirely processed, skipping to the end

I understand that is because  REPUBLISHREQUEST is not yet implemented (issue 370). Am I right?

On the other hand, about tweaking the buffers, I’ve tried this:
const UA_ConnectionConfig UA_ConnectionConfig_standard =
    {.protocolVersion = 0, .sendBufferSize = 6553600, .recvBufferSize  = 6553600,
     .maxMessageSize = 6553600, .maxChunkCount   = 1};

Making buffers 100 times larger, but I get the same result.

Enviado el: jueves, 08 de octubre de 2015 10:25
Para: acplt/open62541
CC: Saiz, Óscar
Asunto: Re: [open62541] BadEncodingLimitsExceeded (#434)

Stasik0, 2015-10-08:
also this line?
https://github.com/acplt/open62541/blob/0332cb20410a01e1b068ac47aee28ef74f4b7985/src/server/ua_server_binary.c#L11

Stasik0, 2015-10-08:
No, in my opinion RepublishRequest is connected to subscriptions that you are not using. I bet, that you hit the maximal buffer size, and the connection drops since we do not have chunking implemented yet.

osaiz, 2015-10-08:
Thanks. Now I also changed the line you suggested:
#define MAX_STACK_MESSAGE 6553600

But I get the same error. 
By the way, I don't see any place where MAX_STACK_MESSAGE is used.

Any idea?

Stasik0, 2015-10-08:
can you use wireshark to see if the response is actually coming?

osaiz, 2015-10-08:
Hi Sten.
Which response?

With fewer variables (80) everything works ok.
Wit more variables (100) I get this error.

Enviado el: jueves, 08 de octubre de 2015 12:35

Stasik0, 2015-10-08:
Well, hook up wireshark https://github.com/acplt/open62541/wiki/Capturing-traffic-with-Wireshark to see if the browse response is actually coming from the server. My theory is that we try to send something that is bigger than UaExpert announced it can accept.

osaiz, 2015-10-13:
Hello. I've used wireshark, and the browse response is actually coming from the server.
[image]

Anyway, my problem is partially solved. Why?

Before all the variables where published at the same level (all of them in a single flat level).
Now I've structured them, by creating ObjectNodes.
After doing that, I don't get any error, and everything works ok.

Anyway the problema is still there. If your ask for too many variables in a "level", UaExpert asks for a Republish, and as this is not implemented yet in 62541, It is not able to browse the variables.

Stasik0, 2015-10-13:
No, I think you misunderstood it, the problem is the ***[Malformed Packet]***, having no publish request is okay, since you do not need to implement it.

Stasik0, 2015-10-13:
@osaiz i can not reproduce your error :( i have added 200000 nodes, and it is browsing, which uaexpert are you using?

osaiz, 2015-10-14:
I’m using version 1.02.334.206 for Windows.

I get also a problem with another OPC-UA client: Prosys-OPC-UA-Client.
That’s the reason why I thought it was a problem in the server side, not in the client side…

But I may be wrong…

Enviado el: martes, 13 de octubre de 2015 21:35

Stasik0, 2015-10-14:
Well, with 1.3.1 I cannot reproduce it anymore, but the problem is on our side... i will give it a try in Prosys-OPC-UA if i find it

osaiz, 2015-10-14:
Sorry Sten.
My UaExpert version is 1.3.1 206.
Sorry for the mistake.

I’ll try to make a simple example reproducing the error and sending it to you

Enviado el: miércoles, 14 de octubre de 2015 8:56

osaiz, 2015-10-14:
Please find the example that reproduces the error.
There are 96 calls to OPCUA_CreateVariableNode.
This way  you get the error.

If you delete one call (just make 95 calls), it works ok.
If you rename the variables with shorter names, it also works ok, even with 96 calls, so I guess there is some problem with names of variables.

Stasik0, 2015-10-14:
where can i find the code/binary? you can just paste the code into pastebin

osaiz, 2015-10-14:
I cannot attach .c files.
I tried changing extension to .txt but it didn’t allow either.
I got this error: “Something went really wrong, and we can’t process that file”.
Any alternative?

Enviado el: miércoles, 14 de octubre de 2015 14:20

FlorianPalm, 2015-10-14:
@Stasik0 meant something like this: http://pastebin.com/

osaiz, 2015-10-14:
Thank you.
Here it is
http://pastebin.com/TPPGZiUB

Enviado el: miércoles, 14 de octubre de 2015 14:49

osaiz, 2015-10-14:
Thank you.
Here it is
http://pastebin.com/TPPGZiUB

Stasik0, 2015-10-14:
@osaiz i cannot compile your code with the current head, can you tell me which version are you using? (it is in the 3rd line of open62541.h)

jpfr, 2015-10-15:
Hey @osaiz 
I adapted the code for the current master. It needed just some small changes.
http://paste2.org/yKGUbbgy

Works fine on my box. Was that the version with the error?

osaiz, 2015-10-15:
* Git-Revision: v0.1.0-RC4-614-g32c5050

Stasik0, 2015-10-15:
@osaiz i cannot reproduce it either, I have just modified your code slightly, are you testing over via localhost? Or do you have some networking between the server and the client? Here is the code i test: http://paste2.org/MbDJh1nD, with v0.1.0-RC4-614-g32c5050

osaiz, 2015-10-15:
I'm not testing on localhost. My server is in a Linux machine and the client on a Windows machine.

I am using the amalgamated version.

This is my compilng command:
gcc -std=c99  -DENABLE_SUBSCRIPTIONS -Wl,-rpath,'pwd' -I ./include -L ./ ./open62541.c ./server5.c -o server5

osaiz, 2015-10-15:
The problem seems to be around "browseName" parameter of UA_Server_addDataSourceVariableNode.

With shorter values of strings for this parameter everything works ok.
With longer ones, with many variables, it seems to reach some kind of limit, where the error appears.

Stasik0, 2015-10-15:
is the code you have posted working or not?

osaiz, 2015-10-15:
No, it is not. :-(
I get this error.
[image]

Alternativelly, if I create one variable less, or if the browseNames of the variables are shorter, then it  does work correctly.

Stasik0, 2015-10-15:
you run the same code I posted (http://paste2.org/MbDJh1nD)? Could you please attach a wireshark log? The capture file with the error?

osaiz, 2015-10-15:
Yes. I'm usign your code.

I've just captured the whireshark log.
How can I attach the file?

Stasik0, 2015-10-15:
something like http://www.tinyupload.com/?

osaiz, 2015-10-15:
Here it is.
http://wikisend.com/download/560610/wiresharklog.pcapng

Stasik0, 2015-10-16:
@osaiz okay, if you look into the packet 131, you will see the 97 references that are coming safely, but then, at the end of the message there is a DiagnosticInfo array, which size is set to "ArraySize: 256002305" that is definitely invalid and breaks everything. Since we cannot reproduce it, I'd say it can be a nonititalized memory allocation or somehting similar, can you reproduce the error with the current head? and code of @jpfr http://paste2.org/yKGUbbgy?

Stasik0, 2015-10-16:
@osaiz @jpfr there are more problems, look into these screenshots
@osaiz please try to reproduce it with current code base

The first on is from @oasiz capture.. there the name of the last item is being mixed up with "MSGF", which seems to be a new or old message? How does it get in there?

[image]

The second screenshot is how it has to look like -> my machine
[image]

osaiz, 2015-10-16:
Hello. I did it yesterday (versión 0.1.0RC4-714--g106bf77). Is this the last one? Or should I check with another one?

By the way, this version does not compile with -DENABLE_SUBSCRIPTIONS. It gives compilation errors.
Compiling without -DENABLE_SUBSCRIPTIONS, it shows the same behaviour as I mentioned.  
- With all the ítems it fails.
- With all the ítems minus one, it works OK.

Here you are the wireshark logs of both executions.

http://wikisend.com/download/780370/right.pcapng     Right execution
http://wikisend.com/download/949358/wrong.pcapng  Wrong execution

About the last post by @Stasik0, I have no clue about how that gets corrupted, but please notice that it happens in the last variable (the one that makes things fail).

Stasik0, 2015-10-16:
@osaiz yes, this is already the newer encoding code... the problem is that I can not reproduce it at all :( which compiler and kernel verision are you using? Could you please try to reproduce it via localhost (there is UaExpert for linux too).

osaiz, 2015-10-16:
gcc 4.4.1
Linux kernel   2.6.31.14-0.8-desktop

I'll try to reproduce it in a more modern Linux, and with UaExpert in local

osaiz, 2015-10-16:
I've tested in a newer Linux, and the problem disappeared

gcc 4.8.3
Linux kernel 3.16.6-2-desktop

Stasik0, 2015-10-16:
This is still very creepy....

Stasik0, 2015-10-20:
okay, i will close it

closed 2015-10-20 by Stasik0