open62541/open62541 #438: questions for the demo

joyhope, 2015-10-12:
run the client and server demo.  Got the result

Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME
0         61               FolderType       FolderType
0         2253             Server           Server
1         96               current time     current time
1         the.answer       the answer       the answer
1         50000            Demo             Demo
1         62541            ping             ping
...

In the server.c, there is a node from 50001 ~ 5003, but why these node is not displayed in the client. 
#define SCALARID 50001
  UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, SCALARID), UA_QUALIFIEDNAME(1, "Scalar"), UA_LOCALIZEDTEXT("en_US","Scalar"),
                          UA_LOCALIZEDTEXT("en_US","Scalar"), 0, 0, UA_NODEID_NUMERIC(1, DEMOID), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE), NULL);

#define ARRAYID 50002
  UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, ARRAYID), UA_QUALIFIEDNAME(1, "Array"), UA_LOCALIZEDTEXT("en_US","Array"),
                          UA_LOCALIZEDTEXT("en_US","Array"), 0, 0, UA_NODEID_NUMERIC(1, DEMOID), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE), NULL);

#define MATRIXID 50003
  UA_Server_addObjectNode(server, UA_NODEID_NUMERIC(1, MATRIXID), UA_QUALIFIEDNAME(1, "Matrix"), UA_LOCALIZEDTEXT("en_US","Matrix"),
                          UA_LOCALIZEDTEXT("en_US","Matrix"), 0, 0, UA_NODEID_NUMERIC(1, DEMOID), UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_EXPANDEDNODEID_NUMERIC(0, UA_NS0ID_FOLDERTYPE), NULL);

Stasik0, 2015-10-12:
Yes, it is because, the client does not "deepen" into the folder "Demo", only one hierarchy level is fetched.

If you replace 'UA_NS0ID_OBJECTSFOLDER' with '50000' in this line https://github.com/acplt/open62541/blob/master/examples/client.c#L39, you will get the content of the "Demo" folder, which will contain the three nodes 50001-50003

joyhope, 2015-10-12:
I do this 

    bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, 50000); //browse objects folder

but the result is worse.  

Nothing displayed 

###
$ ./client
Browsing nodes in objects folder:
NAMESPACE NODEID           BROWSE NAME      DISPLAY NAME
Create subscription succeeded, id 1041106539
Monitoring 'the.answer', id 1041106539
The Answer has changed!

Stasik0, 2015-10-12:
yea, because you have picked the wrong namespace, go for UA_NODEID_NUMERIC(1, 50000)

joyhope, 2015-10-13:
sorry for my questions again.

using UA_NODEID_NUMERIC(1, 50000), it did appear the 50001, 50002, ... .  But the original nodeid information is gone. 

How to keep the original nodeid information, and with 50001, 50002?   (I tried do some adjustment by myself, but I did not success. )

Stasik0, 2015-10-13:
yes, you need to browse recursive with two levels deepening... I am not sure how to help you?

joyhope, 2015-10-14:
clear.

closed 2015-10-14 by joyhope