open62541/open62541 #440: ENABLE_EXTERNAL_NAMESPACES will cause compile failure

joyhope, 2015-10-12:
I enable the ENABLE_EXTERNAL_NAMESPACES, but meet 3 error in the following file. 

open62541\src\server\ua_services_view.c

#ifdef UA_EXTERNAL_NAMESPACES
    	if(*isExternal == UA_TRUE){
    		UA_ObjectNode_delete((UA_ObjectNode*)node);
    	} else
#endif

I am unable to make this pass compile if -Werror=cast-qual  is enable.

closed 2015-10-15 by jpfr

jpfr, 2015-10-15:
The compiler does not complain. But the solution ain't pretty.
The external namespace part definitely needs some love...

Imho, it would be a lot cleaner, if we had a level of indirection for nodestores.
So that we would internally select an "external" nodestore based on the namespaceindex.

We then can have various "translation" layers. For nodes in databases, nodes in ACPLT, ...
The services are becoming really intricate and need to be constantly adapted in every external namespace, whereas the nodestore API is stable.

@Stasik0 @FlorianPalm 
What do you think?

FlorianPalm, 2016-02-01:
I am thinking about a new interface level. The current suggestion of having an interface on service level is probably the best when it comes to performance (thinking of writing thousands of nodes at one time). But the drawbacks in maintenance are obvious. So a "per-node" interface (on nodestore level) might fit more to our current applications.
In the next few days I will provide another proposal how external nodes could be accessed using an individual implementation of the nodestore api.

commit 106bf774 (2015-10-15): fix #440