open62541/open62541 #441: ENABLE_GENERATE_NAMESPACE0=ON will cause compile error

joyhope, 2015-10-15:
If enable this, the generated code will cause lots of problem.  The script may need updated.

jpfr, 2015-10-15:
Good find.
Generating ns0 is not covered by unit tests. So the breakage went unnoticed.

Are you using this feature or are you just looking?

ichrispa, 2015-10-15:
The errors in the namespace compiler are fixed. Though I do wonder how fatal errors could have been committed to master since everyone is tripple testing and has excellent code discipline.

Wasn't ENABLE_GENERATE_NAMESPACE0 in travis?

@jpfr: Great work on getting the c printer out of the node classes. Please remove the dead code from any nodes classes (which should be all printSubtypeEarly). It would also be great if you could remove commented lines that are no longer needed.

joyhope, 2015-10-16:
>>Are you using this feature or are you just looking?
I am learning recently. I try to understand the project, so I enable as much as optional to understand the current project feature.

joyhope, 2015-10-16:
The compile time is very very long.  how to make default compile with debug mode, so to shorten compile time.

ichrispa, 2015-10-16:
@joyhope  Switch the build type to Debug. 

Avoid any String optimizations on the generated source, as they can take up to an hour.

jpfr, 2015-10-16:
done

@ichrispa 
I did a small perf test if ns0 generation with https://github.com/evanhempel/python-flamegraph

It looks as if most of the time was spent in printing definitions.
Probably the running time can be reduced with little effort.

[image]

I still find it really cool to see a server with **all** type definitions. :+1:

ichrispa, 2015-10-16:
Hi @jpfr 

that a really cool perf tool. The namespace compiler must be brought back up to speed after your changes - most notably it does not initialize all attributes (there was a comment of yours).

Once the HL-API is closed, generating user types is waaay up in my todo list. I already have the structs printing in a local branch and the DataType generation does work albeit still buggy. But I want to rewrite the compiler to use a c-file class so we can cleanly generate different headers. I also want to only "include includes" when they become necessary.

When both of these are done, I will consider optimizing the compiler. But until then the codebase is a bit too volatile to get any meaningfull optimization going.

joyhope, 2015-10-16:
>>Switch the build type to Debug.
how to set Debug flag. I tried 

export CMAKE_BUILD_TYPE=1 , but not work.

ichrispa, 2015-10-16:
First, run a make clean (or clean project or whatever you build system calls it).

Change ```CMAKE_BUILD_TYPE=RelWithDebInfo``` into ```CMAKE_BUILD_TYPE=Debug```.

joyhope, 2015-10-17:
client output

$ ./client
[10/17/2015 00:54:30.761] error/client  No suitable endpoint found

Nothing display.  

Whether it is current situation,  client is not ready to display the NS0 information.

jpfr, 2015-10-17:
Hey @joyhope,
this should be fixed after b4e103d3cac3e9794ffea3bd6001792cc0c928c1.

closed 2015-10-19 by jpfr