open62541/open62541 #442: sample server.c crash with current master

StalderT, 2015-10-16:
Hello,

The sample server.c crash with current master on :

UA_Server_setNodeAttribute_displayName(server, UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER), objectsName);

Valgrind result : 

==7865== Conditional jump or move depends on uninitialised value(s)
==7865==    at 0x4029FA6: malloc (in /usr/lib/valgrind/vgpreload_memcheck-x86-linux.so)
==7865==    by 0x40459FF: UA_Variant_setScalarCopy (open62541.c:3467)
==7865==    by 0x4053956: UA_Server_setNodeAttribute (open62541.c:7676)
==7865==    by 0x80495CD: UA_Server_setNodeAttribute_displayName (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==    by 0x804B7C6: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865== 
==7865== Conditional jump or move depends on uninitialised value(s)
==7865==    at 0x4044DD1: UA_copy (open62541.c:3617)
==7865==    by 0x4045A1B: UA_Variant_setScalarCopy (open62541.c:3470)
==7865==    by 0x4053956: UA_Server_setNodeAttribute (open62541.c:7676)
==7865==    by 0x80495CD: UA_Server_setNodeAttribute_displayName (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==    by 0x804B7C6: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865== 
==7865== Conditional jump or move depends on uninitialised value(s)
==7865==    at 0x4043DCF: UA_init (open62541.c:3539)
==7865==    by 0x4044EBC: UA_copy (open62541.c:3630)
==7865==    by 0x4045A1B: UA_Variant_setScalarCopy (open62541.c:3470)
==7865==    by 0x4053956: UA_Server_setNodeAttribute (open62541.c:7676)
==7865==    by 0x80495CD: UA_Server_setNodeAttribute_displayName (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==    by 0x804B7C6: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865== 
==7865== Use of uninitialised value of size 4
==7865==    at 0x4043DBA: UA_init (open62541.c:3539)
==7865==    by 0x4043EF8: UA_init (open62541.c:3593)
==7865==    by 0x4044EBC: UA_copy (open62541.c:3630)
==7865==    by 0x4045A1B: UA_Variant_setScalarCopy (open62541.c:3470)
==7865==    by 0x4053956: UA_Server_setNodeAttribute (open62541.c:7676)
==7865==    by 0x80495CD: UA_Server_setNodeAttribute_displayName (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==    by 0x804B7C6: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865== 
==7865== Invalid read of size 1
==7865==    at 0x4043DBA: UA_init (open62541.c:3539)
==7865==    by 0x4043EF8: UA_init (open62541.c:3593)
==7865==    by 0x4044EBC: UA_copy (open62541.c:3630)
==7865==    by 0x4045A1B: UA_Variant_setScalarCopy (open62541.c:3470)
==7865==    by 0x4053956: UA_Server_setNodeAttribute (open62541.c:7676)
==7865==    by 0x80495CD: UA_Server_setNodeAttribute_displayName (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==    by 0x804B7C6: main (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==  Address 0xbe59dbc5 is not stack'd, malloc'd or (recently) free'd
==7865== 
==7865== 
==7865== Process terminating with default action of signal 11 (SIGSEGV)
==7865==  Access not within mapped region at address 0xBE59DBC5
==7865==    at 0x4043DBA: UA_init (open62541.c:3539)
==7865==    by 0x4043EF8: UA_init (open62541.c:3593)
==7865==    by 0x4044EBC: UA_copy (open62541.c:3630)
==7865==    by 0x4045A1B: UA_Variant_setScalarCopy (open62541.c:3470)
==7865==    by 0x4053956: UA_Server_setNodeAttribute (open62541.c:7676)
==7865==    by 0x80495CD: UA_Server_setNodeAttribute_displayName (in /data/linux/open62541/src/open62541-master/examples/a.out)
==7865==    by 0x804B7C6: main (in /data/linux/open62541/src/open62541-master/examples/a.out)

Best regards,

Thomas

jpfr, 2015-10-16:
Hey Thomas,
I cannot reproduce this with master on my box. Can you tell us
- the compiler
- output of "git describe" and
- build options?

StalderT, 2015-10-16:
[] gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/i686-pc-linux-gnu/5.2.0/lto-wrapper
Target: i686-pc-linux-gnu
Configured with: ./configure --prefix=/usr --sysconfdir=/etc --enable-shared --enable-threads=posix --with-system-zlib --enable-languages=c,c++ --enable-stack-protector -disable-libstdcxx-pch --disable-multilib
Thread model: posix
gcc version 5.2.0 (GCC) 

to build open62541 : 
[] cmake ../open62541-master -DENABLE_AMALGAMATION=ON -DENABLE_SUBSCRIPTIONS=ON -DENABLE_METHODCALLS=ON 
[] make

to build server
[] gcc server.c -lopen62541
[] ./a.out
Segmentation fault

I have downloaded : 
https://github.com/acplt/open62541/archive/master.zip

jpfr, 2015-10-16:
`gcc server.c -lopen62541`
That won't work for me. You at least need to specify -std=c99.

Try to compile statically with with open62541.c generated during amalgamation.
`gcc -std=c99 ../examples/server.c open62541.c`

jpfr, 2015-10-16:
Does it still happen when you update the master source?

StalderT, 2015-10-16:
Hello,

I have made : 

[root@devel data]$ git clone https://github.com/acplt/open62541.git
Cloning into 'open62541'...
remote: Counting objects: 16734, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 16734 (delta 0), reused 0 (delta 0), pack-reused 16729
Receiving objects: 100% (16734/16734), 7.96 MiB | 447.00 KiB/s, done.
Resolving deltas: 100% (12709/12709), done.
Checking connectivity... done.

[root@devel data]$ mkdir build
[root@devel data]$ cd build
[root@devel build]$ cmake ../open62541 -DENABLE_AMALGAMATION=ON -DENABLE_SUBSCRIPTIONS=ON -DENABLE_METHODCALLS=ON
-- The C compiler identification is GNU 5.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.10") 
-- Found Git: /usr/bin/git (found version "2.4.6") 
fatal: Not a git repository (or any parent up to mount point /data)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- Git failed (not a repo, or no tags). Build will not contain git revision info.
-- Git version: unknown--git-commit-id-unknown
-- CMAKE_BUILD_TYPE not given; setting to 'RelWithDebInfo'.
-- Configuring done
-- Generating done
-- Build files have been written to: /data/build

[root@devel build]$ make
......

[root@devel build]$ cp ../open62541/examples/server.c .
[root@devel build]$ gcc -std=c99 server.c open62541.c
[root@devel build]$ ./a.out 
Segmentation fault

jpfr, 2015-10-16:
On 64bit Debian, gcc5 works fine:
```
cmake .. -DENABLE_AMALGAMATION=ON -DENABLE_SUBSCRIPTIONS=ON -DENABLE_METHODCALLS=ON -DCMAKE_C_COMPILER=gcc-5
make
cp ../examples/server.c .
gcc-5 -std=c99 server.c open62541.c
./a.out
```

Can you compile with debug information?
```
cmake .. -DENABLE_AMALGAMATION=ON -DENABLE_SUBSCRIPTIONS=ON -DENABLE_METHODCALLS=ON -DCMAKE_C_COMPILER=gcc-5 -DCMAKE_BUILD_TYPE=Debug
gcc-5 -std=c99 -g server.c open62541.c
```

Then you can get a trace with gdb:
```
gdb ./a.out
r
<here the crash happens>
bt
<prints the backtrace>
```

StalderT, 2015-10-16:
[root@devel build]$ gcc  -std=c99 -g -ggdb server.c open62541.c
[root@devel build]$ gdb a.out 
GNU gdb (GDB) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "i686-pc-linux-gnu".

Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from a.out...done.
(gdb) r
Starting program: /data/open62541/build/a.out 

Program received signal SIGSEGV, Segmentation fault.
0x0804f0c5 in UA_init (p=0x808c312, dataType=0xc00295b4) at open62541.c:3539
3539	    switch(dataType->typeIndex + (0x8000 * !dataType->namespaceZero)) {
(gdb)

StalderT, 2015-10-16:
found the bug : 

https://github.com/acplt/open62541/pull/444

jpfr, 2015-10-16:
*ouch*
Don't trust the type system.

Thank you!

commit f1d725c7 (2015-10-16): Merge pull request #444 from StalderT/patch-1

fix setNodeAttribute, fixes #442
closed 2015-10-16 by jpfr