open62541/open62541 #443: "The message was not entirely processed" in demo server.

tpiat, 2015-10-16:
Every 40 s: info/communication        The message was not entirely processed, skipping to the end
I have use only UA_Client_read.
No subscriptions.

What could it be?

jpfr, 2015-10-16:
Hey @tpiat,
probably the client tries to create a subscriptions.
Our server will not recognize the request (since subscriptions are disabled) and discard it.
That's perfectly normal.

tpiat, 2015-10-16:
I wrote this client. Subscriptions generated automatically wihtout my participation?

jpfr, 2015-10-16:
Ok. Then we shall look into it.
- Can you trace what service you call when the error happens?
- Is the 40s interval regular?
- Maybe provide a dump of the network stream with `tcpdump` or wireshark.

tpiat, 2015-10-16:
```
[10/16/2015 12:14:20.509.000]
[10/16/2015 12:15:00.552.000]
[10/16/2015 12:15:40.595.000]
[10/16/2015 12:16:20.639.000]
[10/16/2015 12:17:00.683.000]
[10/16/2015 12:17:40.726.000]
[10/16/2015 12:18:20.769.000]
```
I don't understand how trace it.
Now I'm trying made dump, but server and client on the same computer. I have started demo server on linux but no similar messages.
I will try to get a dump from one computer

tpiat, 2015-10-16:
Probably I know, why...
There are another opc software on my computer. Including SIMATIC. It can send dubscriptions.

Thanks!!!

closed 2015-10-16 by tpiat