open62541/open62541 #447: API changed. Questions.

tpiat, 2015-10-19:
API is changed, and my server can't build now.
Few questions:
1) Instead UA_Server_setAttribute_value I must use UA_Server_setNodeAttribute_value?
2) UA_Server_setAttribute_value has param type, this param do not used in this function and type presents in UA_Variant.

Stasik0, 2015-10-19:
+1 here to remove the type from function's signature, seems to be a copy-paste issue

jpfr, 2015-10-19:
1)
- Master is the bleeding edge
- During development, APIs occasionally break (if we don't want to go the way of the win32 API) and bugs occur
- Stability is provided by the stable relase we test, release and support. Or after a feature freeze or release candidate is announced.

2) Yes. You are right. The param was removed in 54d05425f8246553a16d5e78570e458f450ead94.

Expect some more breakage, as we currently align the naming between server / client / OPC UA service sets... Things should be somewhat stable in a week.

closed 2015-10-19 by tpiat