open62541/open62541 #449: SigSegv

tpiat, 2015-10-19:
Version dc7192ba84

SigSev in line 6277 
```
LIST_REMOVE(cp, pointers);
```
cp = 0xfeeefeee.

```
0	UA_Session_deleteMembersCleanup	open62541.c	6277	0x40ec38	
1	UA_SessionManager_cleanupTimedOut	open62541.c	9397	0x417e9e	
2	UA_Server_cleanup	open62541.c	6558	0x40f49c	
3	processJobs	open62541.c	8635	0x417027	
4	processRepeatedJobs	open62541.c	8910	0x41739f	
5	UA_Server_run_mainloop	open62541.c	9154	0x4176c0	
6	UA_Server_run	open62541.c	9228	0x417864	
7	main	main.cpp	453	0x405c9f	
```

Have any tips?

jpfr, 2015-10-19:
Can you try after later latest commit?
The latest clang-analyzer found an issue in UA_Session_deleteMembersCleanup.

Is the error reproducible?

tpiat, 2015-10-20:
```
0	UA_Session_deleteMembersCleanup	open62541.c	6279	0x40ecb6	
1	UA_SessionManager_cleanupTimedOut	open62541.c	9400	0x417ee8	
2	UA_Server_cleanup	open62541.c	6561	0x40f4d0	
3	processJobs	open62541.c	8638	0x417061	
4	processRepeatedJobs	open62541.c	8913	0x4173d9	
5	UA_Server_run_mainloop	open62541.c	9157	0x4176fa	
6	UA_Server_run	open62541.c	9231	0x41789e	
7	main	main.cpp	448	0x405c76	
```
Line 6279
LIST_FOREACH_SAFE(cp, &session->continuationPoints, pointers, temp)
Release fc24902096

Stasik0, 2015-10-20:
@tpiat is it reproducable?

tpiat, 2015-10-20:
Reproducable with my client.

Stasik0, 2015-10-20:
@tpiat can you send us an example or is it too complicated?

tpiat, 2015-10-20:
I'll try create simple code (wihout additional code)

tpiat, 2015-10-20:
Simple code is ready. Windows 7, Qt, gcc 4.4.0

jpfr, 2015-10-20:
If its uncritical, just paste it here: http://paste2.org/

tpiat, 2015-10-20:
2 projects, 8 files. Probably another way?

tpiat, 2015-10-20:
http://paste2.org/Jwg6phCx
http://paste2.org/eLm3W88Y

tpiat, 2015-10-20:
http://paste2.org/hj5bwjBv

osaiz, 2015-10-20:
Which version are you using?
In a older version I also got a SIGSEV (Segmentation Fault) in UA_Suscription_deleteMembersCleanup

This was the call stack:
  UA_Suscription_deleteMembers()
  SubscriptionManager_deleteMembers()
  UA_Session_deleteMemberCleanup()
  UA_SessionManager_cleanupTimedOut()
  UA_Server_cleanup()
  processJobs()
  processRepeatedJobs()
  ...
  
I think I've got the reason.
UA_Suscription_deleteMembers used to make this:

    LIST_FOREACH_SAFE(not, &subscription->unpublishedNotifications, listEntry, tmp_not) {
        Subscription_deleteUnpublishedNotification(not->notification->sequenceNumber, subscription);

The iterating macro LIST_FOREACH_SAFE saved in "tmp_not" the next element for next iteration.
BUT, Subscription_deleteUnpublishedNotification didn't delete "not", but it deleted all the notifications of the list whose sequenceNumber was
(not->notification->sequenceNumber).

So, what happens if tmp_not has the same sequenceNumber as (not->notification->sequenceNumber)?
In that case, after calling to Subscription_deleteUnpublishedNotification(), we'll be trying to access a recently freed memory (tmp_not). (not = tmp_not)

But this has been corrected in current version.

I hope this gives you any clue.

Regards

tpiat, 2015-10-20:
Releases:
fc24902 
01ca999

FlorianPalm, 2015-10-22:
Could it be that this inlined function

```c
static UA_INLINE UA_LocalizedText UA_LOCALIZEDTEXT(char *locale, char *text) {
    UA_LocalizedText lt;
    lt.locale = UA_STRING(locale);
    lt.text = UA_STRING(text);
    return lt; }
```
is actually not inlined and results in a stack problem. My server does not even start. I get trouble when 
```c
UA_ObjectAttributes_deleteMembers(&object_attr);
```
tries to free object_attr. After I have removed your "createLocalizedText" function the server starts up.
@jpfr what do you think?

jpfr, 2015-10-22:
@FlorianPalm 
We have `UA_LOCALIZEDTEXT` and `UA_LOCALIZEDTEXT_ALLOC`.
The difference is that the former does **not copy the input arrays**. And freeing a char array on the stack will always crash.

The idea is that we can often avoid allocations. For example, when calling a function that takes a const pointer.

tpiat, 2015-10-22:
What should I write instead?
```
		createLocalizedText(&myVar.description,"en_US","A");
		createLocalizedText(&myVar.displayName,"en_US","A");
```

FlorianPalm, 2015-10-22:
Either this
```C
	object_attr.description = UA_LOCALIZEDTEXT("en_US","CrashSample");
	object_attr.displayName= UA_LOCALIZEDTEXT("en_US","CrashSample");
```
Or
```C
createLocalizedText(&object_attr.description,"en_US","A");
createLocalizedText(&object_attr.displayName,"en_US","A");
```
and
```C
void createLocalizedText(UA_LocalizedText *lt, const char* locale, const char *text)
{
	*lt= UA_LOCALIZEDTEXT_ALLOC(locale,text);
}
```

jpfr, 2015-10-22:
No.

Just do
```c
object_attr.description = UA_LOCALIZEDTEXT_ALLOC("en_US","CrashSample");
object_attr.displayName= UA_LOCALIZEDTEXT_ALLOC("en_US","CrashSample");
```

Then you can free the content of the localizedtext.
`UA_LocalizedText_deleteMembers(&attr.description)`

Otherwise, just leave object_attr alone. It will be removed from the stack when you leave the function and no memory is leaked.

closed 2015-10-22 by FlorianPalm