open62541/open62541 #450: Refactor big functions into several smaller ones

jpfr, 2015-10-20:
Another starting point for beginners...

Functions that are overly big and should be refactored into digestible (maintainable) pieces.
For example, the content of a long for-loop could become a separate (static) function.

`nm --print-size --size-sort --radix=d libopen62541.so` gives the list of biggest functions.
Currently over 1k in binary size are:
```
01009 processOPN
01032 __UA_Client_Service
01033 walkBrowsePath
01070 instantiateObjectNode
01081 UA_Connection_completeMessages
01103 ClientNetworkLayerTCP_connect
01160 ServerNetworkLayerTCP_getJobs
01554 SecureChannelHandshake
01806 Service_Read_single
01909 Service_Browse_single
02300 MoveAttributeIntoNode
02672 processMSG
19576 UA_Server_new
```

ichrispa, 2015-11-19:
Is there any code in these functions that can be reused in other parts? Why create small functions if they are only used at one place. Will it reduce the compiled binary size of the server?

In regard to the "refactoring" of the higlevel branch I would also like to point out that the visible interfaces of these function should not change.

Stasik0, 2015-11-19:
small function are basically the issue of readability/maintanability

jpfr, 2015-12-12:
Removing the issue as its not an immediate work item.
We should have a place for good practices.

closed 2015-12-12 by jpfr