open62541/open62541 #453: 449 bis: SIFSEGV

tpiat, 2015-10-23:
I have downloaded last verison open62541m and received SIGSEGV in line

LIST_FOREACH_SAFE(cp, &session->continuationPoints, pointers, temp) 
function UA_Session_deleteMembersCleanup

Stasik0, 2015-10-23:
and the code? which server are you running?

tpiat, 2015-10-23:
Client and server writed by me. Codes placed in #449.
--
Отправлено из Mail.

Stasik0, 2015-10-23:
how to reproduce? 

1) does server crashes on ctrl-c?
2) does client need to be invoked?
....

tpiat, 2015-10-23:
Server works abour minute. Then appears sigsegv.
--
Отправлено из Mail.

tpiat, 2015-10-23:
Client must be running.

Stasik0, 2015-10-23:
is it happening during the periodic cleanup? Can you show the whole stack?

tpiat, 2015-10-23:
I wrote stack in #449.  
Client disconnect and after reconnecting server crashes.

Stasik0, 2015-10-23:
hi, can you try the newest commit? its is not directly the cp pointer, but another bug during cleanup

tpiat, 2015-10-23:
Without DBG it works (or do not crashed). Tomorrow I'll test it with DBG.

tpiat, 2015-10-24:
46c6388cb3
Tested. Close.

closed 2015-10-24 by tpiat